EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #2: Some things wrong when building the CPU-only version.

YTGhost, 2019-10-18:
hey, I got some things wrong when I try to build the CPU-only version.
the configuration session look like this

> You have bazel 0.21.0- (@non-git) installed.
Please specify the location of python. [Default is C:\Users\YT\Anaconda3\envs\tensorflow-build\python.exe]:

Found possible Python library paths:
  C:\Users\YT\Anaconda3\envs\tensorflow-build\lib\site-packages
Please input the desired Python library path to use.  Default is [C:\Users\YT\Anaconda3\envs\tensorflow-build\lib\site-packages]

Do you wish to build TensorFlow with XLA JIT support? [y/N]: n
No XLA JIT support will be enabled for TensorFlow.

Do you wish to build TensorFlow with ROCm support? [y/N]: n
No ROCm support will be enabled for TensorFlow.

Do you wish to build TensorFlow with CUDA support? [y/N]: n
No CUDA support will be enabled for TensorFlow.

Please specify optimization flags to use during compilation when bazel option "--config=opt" is specified [Default is /arch:AVX]:

Would you like to override eigen strong inline for some C++ compilation to reduce the compilation time? [Y/n]: y
Eigen strong inline overridden.

Preconfigured Bazel build configs. You can use any of the below by adding "--config=<>" to your build command. See .bazelrc for more details.
        --config=mkl            # Build with MKL support.
        --config=monolithic     # Config for mostly static monolithic build.
        --config=gdr            # Build with GDR support.
        --config=verbs          # Build with libverbs support.
        --config=ngraph         # Build with Intel nGraph support.
        --config=dynamic_kernels        # (Experimental) Build kernels into separate shared objects.
Preconfigured Bazel build configs to DISABLE default on features:
        --config=noaws          # Disable AWS S3 filesystem support.
        --config=nogcp          # Disable GCP support.
        --config=nohdfs         # Disable HDFS support.
        --config=noignite       # Disable Apacha Ignite support.
        --config=nokafka        # Disable Apache Kafka support.
        --config=nonccl         # Disable NVIDIA NCCL support.

Then it runs very well at first，but it encountered some errors in the middle，just like this:

> ERROR: C:/tensorflow-build/tensorflow/tensorflow/core/kernels/BUILD:3221:1: C++ compilation of rule '//tensorflow/core/kernels:scan_ops' failed (Exit 2): cl.exe failed: error executing command

and this:

> ERROR: C:/tensorflow-build/tensorflow/tensorflow/tools/pip_package/BUILD:241:1 C++ compilation of rule '//tensorflow/core/kernels:batch_matmul_op' failed (Exit 2): cl.exe failed: error executing command

I don't know if this error message is sufficient, if you need more details，just @me.

EdjeElectronics, 2019-10-19:
I remember running in to some cl.exe errors initially, but I can't remember what I did to fix them. Let me know if you figure it out!

YTGhost, 2019-10-19:
> I remember running in to some cl.exe errors initially, but I can't remember what I did to fix them. Let me know if you figure it out!

oh, I see. I will try to fix it. But can you send me the the wheel file to my mailbox(my version of tensorflow is same to you). I can't wait to try how fast it can be than before, it is my email: ytdengliang@gmail.com
if I figure the wrong out, I will let you know as soon as possible.

EdjeElectronics, 2019-10-19:
I wish it was as easy as just giving you the wheel file! TensorFlow has to be built on your machine for the TOCO tools to work. There's something extra about the build process that doesn't happen during a normal wheel file installation. I'll have to ask a software developer why it has to be built on the PC rather than just installed.

YTGhost, 2019-10-19:
> I wish it was as easy as just giving you the wheel file! Unfortunately, TensorFlow HAS to be built on your machine for the TOCO tools to work. There's something extra about the build process that doesn't happen during a normal wheel file installation. I'll have to ask a software developer why it has to be built on the PC rather than just installed.

Hey, I think I have found a solution, but now I have to sleep first. I will give the solution in tomorrow morning.

YTGhost, 2019-10-20:
> I wish it was as easy as just giving you the wheel file! Unfortunately, TensorFlow HAS to be built on your machine for the TOCO tools to work. There's something extra about the build process that [...]

Hey, at the end of the fight against various bugs, I successfully ran the TF Lite model. This is really a great tutorial. But before I sorted out this series of errors, I found that it run in my laptop have only 1.96FPS,  it seem don't use the GPU i have in laptop, what should i do?

YTGhost, 2019-10-20:
> > I wish it was as easy as just giving you the wheel file! Unfortunately, TensorFlow HAS to be built on your machine for the TOCO tools to work. There's something extra about the build process that doesn't happen during a normal wheel file installation. I'll have to ask a software developer why it has to be built on the PC rather than just installed.
> 
> Hey, at the end of the fight against various bugs, I successfully ran the TF Lite model. This is really a great tutorial. But before I sorted out this series of errors, I found that it run in my laptop have only 1.96FPS, it seem don't use the GPU i have in laptop, what should i do? I mean that what should i do to use my GPU in laptop to run the TF lite?

EdjeElectronics, 2019-10-20:
Yeah!! I'm glad you got it working. Can you tell me how you resolved the error related to cl.exe?

It runs very slow on my gaming PC too. The TFLite runtime does NOT use the GPU, so it runs slower than regular TensorFlow does if you have GPU on your PC. The benefit comes when you don't have a GPU: then TFLite runs much faster than regular TensorFlow!

Unfortunately, this means that if you were just hoping to use TFLite to speed up FPS on your laptop, then you'll have to try something else. You could try installing Ubuntu on a separate partition on your laptop, and then using it with the [Coral USB Accelerator](https://coral.withgoogle.com/products/accelerator/). (Like I said previously, it won't work with Windows.) Sorry there isn't an easier way to speed up FPS!

YTGhost, 2019-10-21:
> Yeah!! I'm glad you got it working. Can you tell me how you resolved the error related to cl.exe?
> 
> It runs very slow on my gaming PC too. The TFLite runtime does NOT use the GPU, so it runs slower than regular TensorFlow does if you have GPU on your PC. The benefit comes when you don't have a GPU: then TFLite runs much faster than regular TensorFlow!
> 
> Unfortunately, this means that if you were just hoping to use TFLite to speed up FPS on your laptop, then you'll have to try something else. You could try installing Ubuntu on a separate partition on your laptop, and then using it with the [Coral USB Accelerator](https://coral.withgoogle.com/products/accelerator/). (It won't work with Windows.) You'll be able to use the same instructions as I wrote for [using the Accelerator on the Raspberry Pi](https://github.com/EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi/blob/master/Raspberry_Pi_Guide.md#part-2---run-edge-tpu-object-detection-models-on-the-raspberry-pi-using-the-coral-usb-accelerator) (which I'm still working on). Sorry there isn't an easier way to speed up FPS!

hey, let start it. When I got these problems, I started to check all of these steps before build. And I found that may be due to insufficient memory(my RAM only 8G) and insufficient storage(it use 17.8G in my 
hard disk) at compile time. So I change the building command.

>  bazel build --config=opt //tensorflow/tools/pip_package:build_pip_package --local_resources 2048,.5,1.0

you should use **--local_resources 2048,.5,1.0** instead of **--local_ram_resources=2048**, I think it not use in the bazel which version is 0.21.0.
Also, you need to make sure your hard drive is big enough(it use 17.8G in my case, but the whl file I create is only 50MB!! It was amazing.), you can change the bazel output path just like using this:**--output_user_root=D:/build/tensorflow**.
Just like this:

> bazel --output_user_root=D:/build/tensorflow build --config=opt //tensorflow/tools/pip_package:build_pip_package --local_resources 2048,.5,1.0

And it can build with no problem!!!(But it will last a long time, in my case, it last 3 hours to build)
I hope that these experiences and supplements will make this tutorial even better. And also, I want to ask a question that if i want to have higher detect speed(about 30FPS) in win10 using GPU(such like GTX1080), which model should I use? Could you give me some advise?

EdjeElectronics, 2022-09-07:
Closing due to inactivity.

closed 2022-09-07 by EdjeElectronics