EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #6: interpreter.allocate_tensors() error when trying to test detection_image

romellfudi, 2019-10-28:
When i tried to run "python TFLite_detection_image.py --modeldir=TFLite_model --image=P8120180.jpeg"

throws this error:
Traceback (most recent call last):
  File "TFLite_detection_image.py", line 90, in <module>
    interpreter.allocate_tensors()
  File "/usr/local/lib/python3.6/dist-packages/tensorflow_core/lite/python/interpreter.py", line 244, in allocate_tensors
    return self._interpreter.AllocateTensors()
  File "/usr/local/lib/python3.6/dist-packages/tensorflow_core/lite/python/interpreter_wrapper/tensorflow_wrap_interpreter_wrapper.py", line 106, in AllocateTensors
    return _tensorflow_wrap_interpreter_wrapper.InterpreterWrapper_AllocateTensors(self)
RuntimeError: Encountered unresolved custom op: AddV2.Node number 0 (AddV2) failed to prepare.

my TFLite_model directory:
- detect.tflite
- labelmap.txt
- tflite_graph.pb
- tflite_graph.pbtxt

EdjeElectronics, 2019-10-30:
What type of model are you using? Is it SSD-MobileNet-V2, Quantized-SSD-MobileNet-V2, or something else?

romellfudi, 2019-11-03:
@EdjeElectronics, it  is Quantized-SSD-MobileNet-V2

EdjeElectronics, 2019-11-03:
Oh, I see what's causing the problem. You only worked through Step 1. 

Unfortunately, you have to do everything in Steps 1, 2, and 3 to get a custom model to work with TensorFlow Lite! Just doing the export_tflite_ssd_grpah.py command isn't enough. You have to run the exported file through the "TensorFlow Lite Optimizing Converter" (TOCO) tool, which requires building TensorFlow from source (Step 2). Step 3 shows how to convert the model using TOCO.

Please re-open this issue if you run into any errors while following Step 2 and Step 3.

closed 2019-11-03 by EdjeElectronics

romellfudi, 2019-11-04:
Hi @EdjeElectronics, i think the sentence doesn't complete very well. Now i ran it again 4 hours ago and still running.. : 
bazel build --config=opt //tensorflow/tools/pip_package:build_pip_package 
my question is how many hours take to be completed?

romellfudi, 2019-11-04:
its done. but still show the same message:

i delay to reply because i export to [codelab notebook] that i could replicate it:(https://colab.research.google.com/drive/1zjfE9jHrzZgNKQK-e-VEr6vsIqRZOuxv)