EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #8: TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'

timuritu, 2019-11-05:
Hello, 

I installed the project, followed all the steps and it worked fine, then I changed the webcam and it stopped working. Wiped out everything installed it again, and still got this error. 

select timeout
Traceback (most recent call last):
  File "Object_detection_picamera.py", line 192, in <module>
    feed_dict={image_tensor: frame_expanded})
  File "/home/pi/.local/lib/python3.7/site-packages/tensorflow_core/python/client/session.py", line 941, in run
    run_metadata_ptr)
  File "/home/pi/.local/lib/python3.7/site-packages/tensorflow_core/python/client/session.py", line 1133, in _run
    np_val = np.asarray(subfeed_val, dtype=subfeed_dtype)
  File "/usr/lib/python3/dist-packages/numpy/core/numeric.py", line 538, in asarray
    return array(a, dtype, copy=False, order=order)
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'

timuritu, 2019-11-05:
Changed the webcam back and it works...

EdjeElectronics, 2019-11-05:
Hi @timuritu , the 'NoneType' error means that the program received an empty array from the webcam, which typically means something is wrong with the webcam or the interface to the webcam. Try plugging and re-plugging the webcam in a few times and see if that works. If not, you may need to get a new webcam.

P.S. This will be a good error to add to the "common errors" FAQ. Thanks!

closed 2019-11-05 by EdjeElectronics