EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #9: Error from cv2

HenrikFB, 2019-11-07:
Hi. 

When I run: 
python3 TFLite_detection_webcam.py --modeldir=Sample_TFLite_model

I got the following error:
Traceback (most recent call last):
File "TFLite_detection_webcam.py", line 19, in <module>
import cv2
File "/home/pi/tflite1/tflite1-env/lib/python3.7/site-packages/cv2/__init__.py", line 3, in <module>
from .cv2 import *
ImportError: /home/pi/tflite1/tflite1-env/lib/python3.7/site-packages/cv2/cv2.cpython-37m-arm-linux-gnueabihf.so: undefined symbol: __atomic_fetch_add_8

From your other interesting guide:
https://github.com/EdjeElectronics/TensorFlow-Object-Detection-on-the-Raspberry-Pi

I got the error:
n_picamera.py 
Traceback (most recent call last):
  File "Object_detection_picamera.py", line 23, in <module>
    import cv2
  File "/usr/local/lib/python3.7/dist-packages/cv2/__init__.py", line 3, in <module>
    from .cv2 import *
ImportError: libQtTest.so.4: cannot open shared object file: No such file or directory

Is my problem something about the version with my python, OS, cv2 version? I'm on pi 4 and updated. It is just both of the guide results in an error for me.

Henrik.

EdjeElectronics, 2019-11-07:
Hi @HenrikFB, I think this is a problem with the latest release of OpenCV-Python. Can you try installing an older version by using "pip3 install opencv-python==3.4.6.27" and then re-running the script? Let me know if that works!

HenrikFB, 2019-11-08:
Thanks! It works now for both guides! I appreciate it :)

EdjeElectronics, 2019-11-08:
Awesome! Thanks for updating me that it worked. I changed the requirements file to use "pip3 install opencv-python==3.4.6.27", and will change it back to just "pip3 install opencv-python" once I'm confident it's fixed. 

Gotta love when broken tool versions get released!

closed 2019-11-08 by EdjeElectronics