EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #13: About cv2

Yang-Seong-Eun, 2019-11-15:
Hi I am execute line
python3 TFLite_detection_webcam.py --modeldir = Sample_TFLite_model
but that result is
Traceback (most recent call last):
   File "TFLite_detection_webcam.py", line 19, in <module>
     import cv2
ImportError: No module named 'cv2'

Please help me..

saqibshakeel035, 2019-11-15:
I have also the same issue while executing python3 TFLite_detection_webcam.py --modeldir = Sample_TFLite_model command.
Any  solutions??

EdjeElectronics, 2019-11-15:
This is likely because you don't have the tflite1-env environment activated. Please try closing and re-opening your terminal, then issuing:

```
cd tflite1
source tflite1-env/bin/activate
```

Then try re-running the TFLite_detection_webcam.py script. Let me know if that works!

saqibshakeel035, 2019-11-15:
Thanks but I was running it with tflite1-env in active. I solved the problem of cv2 by using sudo before python3 TFLite_detection_webcam.py --modeldir = Sample_TFLite_model but now I am getting this error
(tflite1-env) pi@raspberrypi:~/tflite1 $ sudo python3 TFLite_detection_webcam.py --modeldir=Sample_TFLite_model
/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/dtypes.py:516: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/dtypes.py:517: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/dtypes.py:518: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/dtypes.py:519: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/dtypes.py:520: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/dtypes.py:525: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: compiletime version 3.4 of module 'tensorflow.python.framework.fast_tensor_util' does not match runtime version 3.5
  return f(*args, **kwds)
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: builtins.type size changed, may indicate binary incompatibility. Expected 432, got 412
  return f(*args, **kwds)
/usr/local/lib/python3.5/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:541: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/usr/local/lib/python3.5/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:542: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/usr/local/lib/python3.5/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:543: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/usr/local/lib/python3.5/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:544: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/usr/local/lib/python3.5/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:545: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/usr/local/lib/python3.5/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:550: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
Traceback (most recent call last):
  File "TFLite_detection_webcam.py", line 119, in <module>
    interpreter = Interpreter(model_path=PATH_TO_CKPT)
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/lite/python/interpreter.py", line 76, in __init__
    _interpreter_wrapper.InterpreterWrapper_CreateWrapperCPPFromFile(
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/python/util/lazy_loader.py", line 62, in __getattr__
    module = self._load()
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/python/util/lazy_loader.py", line 45, in _load
    module = importlib.import_module(self.__name__)
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 986, in _gcd_import
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 673, in exec_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
[... cut ...]
  File "<frozen importlib._bootstrap>", line 666, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 577, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 914, in create_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
ImportError: /usr/local/lib/python3.5/dist-packages/tensorflow/lite/python/interpreter_wrapper/_tensorflow_wrap_interpreter_wrapper.so: undefined symbol: _ZN6tflite12tensor_utils24NeonVectorScalarMultiplyEPKaifPf​

saqibshakeel035, 2019-11-15:
> This is likely because you don't have the tflite1-env environment activated. Please try closing and re-opening your terminal, then issuing:
> 
> ```
> cd tflite1
> source tflite1-env/bin/activate
> ```
> 
> Then try re-running the TFLite_detection_webcam.py script. Let me know if that works!
I solved the cv2 problem by using sudo but now I ran into this problem. Any suggetions  would be appreciated :).
Thanks in advance :)

(tflite1-env) pi@raspberrypi:~/tflite1 $ sudo python3 TFLite_detection_webcam.py --modeldir=Sample_TFLite_model
/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/dtypes.py:516: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/dtypes.py:517: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/dtypes.py:518: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/dtypes.py:519: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/dtypes.py:520: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/dtypes.py:525: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: compiletime version 3.4 of module 'tensorflow.python.framework.fast_tensor_util' does not match runtime version 3.5
  return f(*args, **kwds)
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: builtins.type size changed, may indicate binary incompatibility. Expected 432, got 412
  return f(*args, **kwds)
/usr/local/lib/python3.5/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:541: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/usr/local/lib/python3.5/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:542: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/usr/local/lib/python3.5/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:543: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/usr/local/lib/python3.5/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:544: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/usr/local/lib/python3.5/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:545: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/usr/local/lib/python3.5/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:550: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
Traceback (most recent call last):
  File "TFLite_detection_webcam.py", line 119, in <module>
    interpreter = Interpreter(model_path=PATH_TO_CKPT)
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/lite/python/interpreter.py", line 76, in __init__
    _interpreter_wrapper.InterpreterWrapper_CreateWrapperCPPFromFile(
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/python/util/lazy_loader.py", line 62, in __getattr__
    module = self._load()
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/python/util/lazy_loader.py", line 45, in _load
    module = importlib.import_module(self.__name__)
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 986, in _gcd_import
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
[... cut ...]
  File "<frozen importlib._bootstrap>", line 666, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 577, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 914, in create_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
ImportError: /usr/local/lib/python3.5/dist-packages/tensorflow/lite/python/interpreter_wrapper/_tensorflow_wrap_interpreter_wrapper.so: undefined symbol: _ZN6tflite12tensor_utils24NeonVectorScalarMultiplyEPKaifPf​

EdjeElectronics, 2019-11-15:
That error is a known issue with TensorFlow v1.13, please use a different version (TF v1.14 or higher)! https://github.com/tensorflow/tensorflow/issues/33671

saqibshakeel035, 2019-11-16:
> That error is a known issue with TensorFlow v1.13, please use a different version (TF v1.14 or higher)! [tensorflow/tensorflow#33671](https://github.com/tensorflow/tensorflow/issues/33671)

After updating my tensorflow version to v1.14, I am still getting the same error :(

Yang-Seong-Eun, 2019-11-17:
I solved cv2 problem but this new problem is occur....

ImportError: /home/pi/tflite1/tflite1-env/lib/python3.5/site-packages/tensorflow/lite/python/interpreter_wrapper/_tensorflow_wrap_interpreter_wrapper.so: undefined symbol: _ZN6tflite12tensor_utils24NeonVectorScalarMultiplyEPKaifPf

saqibshakeel035, 2019-11-17:
> I solved cv2 problem but this new problem is occur....
> 
> ImportError: /home/pi/tflite1/tflite1-env/lib/python3.5/site-packages/tensorflow/lite/python/interpreter_wrapper/_tensorflow_wrap_interpreter_wrapper.so: undefined symbol: _ZN6tflite12tensor_utils24NeonVectorScalarMultiplyEPKaifPf

Same Problem :(

EdjeElectronics, 2019-11-17:
Hmm, I know that error occurs because of an issue with older versions of TensorFlow (TF v1.14 may also not be new enough). I was experiencing the same error, even when I was using TF v2.0 inside a virtual environment but still had TF v1.13 installed normally.

The only thing I can recommend at this point is completely erasing your SD card (or getting a new one), and re-installing Rasbpian from stretch. Then, restart this guide from Step 1.

Yang-Seong-Eun, 2019-11-17:
Thank you for your kind reply. I will try it!

saqibshakeel035, 2019-11-17:
After re-installing Raspbianand following your TFlite guide, I am now getting this error at line 29:
from tflite_runtime.interpreter import Interpreter'
ImportError: No module name "tflite_runtime"

EdjeElectronics, 2019-11-17:
@saqibshakeel035 can you try opening Python, importing tensorflow, and checking the version?

Open a Python shell by issuing:
```
python
```

Then, import tensorflow and check the version by using:

```
import tensorflow as tf
tf.__version__
```

And let me know what it prints out as a result.

saqibshakeel035, 2019-11-17:
it's tensorflow-1.14.0

dylankolson, 2019-11-17:
> That error is a known issue with TensorFlow v1.13, please use a different version (TF v1.14 or higher)! [tensorflow/tensorflow#33671](https://github.com/tensorflow/tensorflow/issues/33671)

Having the same issue in 1.14 Python 3.5

EdjeElectronics, 2019-11-17:
@saqibshakeel035 and @dylankolson just want to confirm, did you completely erase your SD card and re-install Rasbpian from scratch? Also, you are working with a Raspberry Pi 3 running the Rasbpian Stretch OS, correct?

I will see if I can reproduce the error on my Pi 3 in a few days.

saqibshakeel035, 2019-11-17:
Yeah I completely erased the card, Re-installed Raspbian stretch using etcher. 
And I am working on Raspberry Pi 3B+.
Figuring it out from last couple of days but failed :(.
I will be looking forward to your output 👍

dylankolson, 2019-11-18:
Tep just tried again with a fresh OS install.

`(tflite1-env) pi@raspberrypi:~/tflite1 $ python3 TFLite_detection_webcam.py --modeldir=Sample_TFLite_model
/home/pi/tflite1/tflite1-env/lib/python3.5/site-packages/tensorflow/python/framework/dtypes.py:516: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/home/pi/tflite1/tflite1-env/lib/python3.5/site-packages/tensorflow/python/framework/dtypes.py:517: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/home/pi/tflite1/tflite1-env/lib/python3.5/site-packages/tensorflow/python/framework/dtypes.py:518: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/home/pi/tflite1/tflite1-env/lib/python3.5/site-packages/tensorflow/python/framework/dtypes.py:519: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/home/pi/tflite1/tflite1-env/lib/python3.5/site-packages/tensorflow/python/framework/dtypes.py:520: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/home/pi/tflite1/tflite1-env/lib/python3.5/site-packages/tensorflow/python/framework/dtypes.py:525: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
/home/pi/tflite1/tflite1-env/lib/python3.5/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:541: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/home/pi/tflite1/tflite1-env/lib/python3.5/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:542: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/home/pi/tflite1/tflite1-env/lib/python3.5/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:543: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/home/pi/tflite1/tflite1-env/lib/python3.5/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:544: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/home/pi/tflite1/tflite1-env/lib/python3.5/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:545: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/home/pi/tflite1/tflite1-env/lib/python3.5/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:550: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
Traceback (most recent call last):
  File "TFLite_detection_webcam.py", line 119, in <module>
    interpreter = Interpreter(model_path=PATH_TO_CKPT)
  File "/home/pi/tflite1/tflite1-env/lib/python3.5/site-packages/tensorflow/lite/python/interpreter.py", line 76, in __init__
    _interpreter_wrapper.InterpreterWrapper_CreateWrapperCPPFromFile(
  File "/home/pi/tflite1/tflite1-env/lib/python3.5/site-packages/tensorflow/python/util/lazy_loader.py", line 62, in __getattr__
    module = self._load()
  File "/home/pi/tflite1/tflite1-env/lib/python3.5/site-packages/tensorflow/python/util/lazy_loader.py", line 45, in _load
    module = importlib.import_module(self.__name__)
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 986, in _gcd_import
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 673, in exec_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
  File "/home/pi/tflite1/tflite1-env/lib/python3.5/site-packages/tensorflow/lite/python/interpreter_wrapper/tensorflow_wrap_interpreter_wrapper.py", line 28, in <module>
    _tensorflow_wrap_interpreter_wrapper = swig_import_helper()
[... cut ...] ifPf
`

`(tflite1-env) pi@raspberrypi:~/tflite1 $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"`

Raspberry Pi 3B+.

`(tflite1-env) pi@raspberrypi:~/tflite1 $ python3 --version
Python 3.5.3
`

>>> tf.__version__
'1.14.0'

Hopefully something in there helps, thanks for looking!

Yang-Seong-Eun, 2019-11-18:
I execute this line bash get_pi_requirements.sh
but error like this -> "THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them."

closed 2019-11-18 by Yang-Seong-Eun

dylankolson, 2019-11-18:
How do I update the hashes?

saqibshakeel035, 2019-11-19:
> @saqibshakeel035 and @dylankolson just want to confirm, did you completely erase your SD card and re-install Rasbpian from scratch? Also, you are working with a Raspberry Pi 3 running the Rasbpian Stretch OS, correct?
> 
> I will see if I can reproduce the error on my Pi 3 in a few days.

Hello friend, 
Hope you are doing good. Can you please tell me if you gave it a try as I am still struggling with the same problem. 
Looking forward :)

saqibshakeel035, 2019-11-19:
> Hmm, I know that error occurs because of an issue with older versions of TensorFlow (TF v1.14 may also not be new enough). I was experiencing the same error, even when I was using TF v2.0 inside a virtual environment but still had TF v1.13 installed normally.
> 
> The only thing I can recommend at this point is completely erasing your SD card (or getting a new one), and re-installing Rasbpian from scratch. Then, restart this guide from Step 1.

I have observed one more thing today that TensorFlow v.2.0 is not available for RPi 3b+ Debrain stretch  as **tensorflow-2.0.0-cp35none-linux_armv7l.whl** is not available. :(
When I try to install tenosrflow 2.0.0 using the following command:
`**sudo pip install tensorflow-2.0.0-cp35-none-linux_armv71.whl**`
It gives me the following error:
`**Requirement tensorflow-2.0.0-cp35-none-linux_armv71.whl looks like a filename, but the file doesnot exist**`

vasylhiruk, 2019-11-20:
I have tried initially with the Raspbian-stretch and failed exactly with the above described by **dylankolson** error. 
Today I have downloaded the OS "Raspbian Buster with desktop" from https://www.raspberrypi.org/downloads/raspbian/, installed it, then followed the step-by-step instruction from 
https://github.com/EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi/blob/master/Raspberry_Pi_Guide.md#part-1---how-to-set-up-and-run-tensorflow-lite-object-detection-models-on-the-raspberry-pi
and run the command
python3 TFLite_detection_webcam.py --modeldir=Sample_TFLite_model
SUCCESSFULLY !!!
Thank you EdjeElectronics !

vasylhiruk, 2019-11-20:
I made it at the Raspberry Pi 3 B+

dylankolson, 2019-11-20:
@vasylhiruk Thanks, I will try this tonight!

dylankolson, 2019-11-21:
@vasylhiruk Thanks this fixed my issue and I can now run it but after a few seconds my PI seems to crash and reboot

vasylhiruk, 2019-11-23:
Actually after running of the **TFLite_detection_webcam.py** at Raspberry Pi 3 B+ I got two issues :
1. endless messages
.....
corrupt JPEG data: 1 extraneous bytes before marker 0xd2
....
which I have hidden by redirecting them to the **/dev/null**
python3 TFLite_detection_webcam.py --modeldir=Sample_TFLite_model > /dev/null 2>&1

2. All bounding boxes were displaced regarding the video objects. I have adjusted them using the factors 0.7 for the Y-coordinates and 0.5 for the X-coordinates :
           # Get bounding box coordinates and draw box
            # Interpreter can return coordinates that are outside of image dimensions, need to force them to be within image using max() and min()
            ymin = int(max(1,(boxes[i][0] * imH * 0.7)))
            xmin = int(max(1,(boxes[i][1] * imW * 0.5)))
            ymax = int(min(imH,(boxes[i][2] * imH * 0.7)))
            xmax = int(min(imW,(boxes[i][3] * imW * 0.5)))

EdjeElectronics, 2019-11-28:
Thank you everyone for your involvement in this thread and getting this dumb NeonVectorScalar error figured out! TensorFlow v1.14 and lower have this error when trying to run TFLite, and unfortunately, there are no newer versions (easily) available for the Pi 3 running Raspbian Stretch.

The workaround is to uninstall regular TensorFlow and then install the [TensorFlow Lite runtime](https://www.tensorflow.org/lite/guide/python) provided by Google instead. Open a new terminal window, and issue the following commands:

```
cd tflite1
source tflite1-env/bin/activate
pip3 uninstall tensorflow
wget https://dl.google.com/coral/python/tflite_runtime-1.14.0-cp35-cp35m-linux_armv7l.whl
pip3 install tflite_runtime-1.14.0-cp35-cp35m-linux_armv7l.whl
```

Then, you should be able to run the TFLite_detection scripts without any issues. 

I will change the get_pi_requirements.txt file so it downloads the TFLite Runtime rather than regular tensorflow if it detects that the user is running Python 3.5 (which typically means the user is on Raspbian Stretch). That way, people working through this guide on a Pi 3 with Raspbian Stretch won't run in to issues.

Hopefully the TensorFlow team will get this error sorted out, or will just upload TF v2.0 wheel files compiled for Python 3.5 and 3.7 to piwheels.org. Once they do, I'll go back in and change get_pi_requirements.txt to just use 'pip3 install tensorflow'.

OJBameyi, 2021-11-23:
Dear EdjeElectronics
I have followed your tutorials and they have been quite helpful. I am having this error when I try to run my own trained model, based on SSD fpn lite.

Please can you help out. I am using a raspberry 3B+ with a raspian buster.

The error message is pated below:

Traceback (most recent call last):
  File "TFLite_detection_webcam.py", line 138, in <module>
    interpreter = Interpreter(model_path=PATH_TO_CKPT)
  File "/home/pi/Projects/Python/tflite/tflite-env/lib/python3.7/site-packages/tflite_runtime/interpreter.py", line 351, in __init__
    experimental_preserve_all_tensors))
ValueError: Didn't find op for builtin opcode 'STRIDED_SLICE' version '6'. An older version of this builtin might be supported. Are you using an old TFLite binary with a newer model?
Registration failed.