EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #14: Cumulative Counting Mode integration?

airqualityanthony, 2019-11-18:
Hi,

Just like to say great work! 

Question: 

How would I go about integrating the Cumulative Counting Model api into the TFLite_detection_webcam.py?

https://github.com/ahmetozlu/tensorflow_object_counting_api

1.2) For detecting, tracking and counting the vehicles with enabled color prediction

Usage of "Cumulative Counting Mode" for the "vehicle counting" case:
```
fps = 24 # change it with your input video fps
width = 640 # change it with your input video width
height = 352 # change it with your input vide height
is_color_recognition_enabled = 0 # set it to 1 for enabling the color prediction for the detected objects
roi = 200 # roi line position
deviation = 3 # the constant that represents the object counting area

object_counting_api.cumulative_object_counting_y_axis(input_video, detection_graph, category_index, is_color_recognition_enabled, fps, width, height, roi, deviation) # counting all the objects
```

armaanpriyadarshan, 2020-04-25:
Hi, since this is a few months old I wanted to ask if you made any progress with the cumulative counting model. I ask cause i'm trying to do the same thing.

m0-n, 2020-08-01:
Did anyone end up finding something here? @armaanpriyadarshan @airqualityanthony  @EdjeElectronics?

armaanpriyadarshan, 2020-08-01:
Yes I found a way to count the number of detections in a frame. I trained a custom model and made a pill counting program. I made a repo here: https://github.com/armaanpriyadarshan/Object-Counting-on-RasPi and a YouTube video here: https://www.youtube.com/watch?v=3jTteM3dYkQ&t=118s

m0-n, 2020-08-01:
Many thanks -- that looks really cool! 

No cumulative counting though? (how many objects are passing through a vertical or horizontal line...)

armaanpriyadarshan, 2020-08-01:
I haven't done so yet, sorry about that

airqualityanthony, 2020-10-18:
I never got anywhere with it, but I'm back at it again now! I'll let you know how it goes

aparico, 2020-12-15:
> I never got anywhere with it, but I'm back at it again now! I'll let you know how it goes

@airqualityanthony Hi. May I know if you have any updates on this?

airqualityanthony, 2020-12-19:
Hey guys, 

Yeah I added some basic count functionality to the script, but it's simply a += to a list for each class you're interested in when it crosses the ROI line. 

See attached. 

[TFLite_detection_webcam_counting_pi_cam.txt](https://github.com/EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi/files/5720170/TFLite_detection_webcam_counting_pi_cam.txt)

Framkenstein, 2021-07-27:
@airqualityanthony  Awesome work with the pi counting!

airqualityanthony, 2021-07-29:
Closing issue with above TFLite_detection_webcam_counting_pi_cam.txt file.

closed 2021-07-29 by airqualityanthony