EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #16: ImportError: /home/pi/tflite1/tflite1-env/lib/python3.5/site-packages/tensorflow/lite/python/interpreter_wrapper/_tensorflow_wrap_interpreter_wrapper.so: undefined symbol: _ZN6tflite12tensor_utils24NeonVectorScalarMultiplyEPKaifPf

saqibshakeel035, 2019-11-19:
I have observed one more thing today that TensorFlow v.2.0 is not available for RPi 3b+ Debrain stretch as tensorflow-2.0.0-cp35none-linux_armv7l.whl is not available. :(
When I try to install tenosrflow 2.0.0 using the following command:
**sudo pip install tensorflow-2.0.0-cp35-none-linux_armv71.whl**
It gives me the following error:
**Requirement tensorflow-2.0.0-cp35-none-linux_armv71.whl looks like a filename, but the file doesnot exist**

Al-Martin, 2019-11-20:
Same error here.

saqibshakeel035, 2019-11-20:
> Same error here.

Can you please tell more details about your operating system and hardware etc?
Also are you running your python script by SSH ?

Al-Martin, 2019-11-21:
Hi,

I'm using Raspberry Pi Model 3B+, with Raspbian (based on Debian Buster)
and a Pi NoIR Camera V2 (Nigth Vision Camera for Raspberry Pi).
I'm using VNC Viewer, and since there using the terminal to write the
commands.

Thank you!

EdjeElectronics, 2019-11-28:
This error occurs when you are using TensorFlow v1.14 or lower to run TensorFlow Lite. It's a [known bug](https://github.com/tensorflow/tensorflow/issues/33671), and hopefully the TensorFlow team will resolve it soon.

To work around the error, we can uninstall regular TensorFlow and then use the tflite_runtime package [provided by Google](https://www.tensorflow.org/lite/guide/python) instead.

Open a new terminal window and issue the following commands:

```
cd tflite1
source tflite1-env/bin/activate
pip3 uninstall tensorflow
wget https://dl.google.com/coral/python/tflite_runtime-1.14.0-cp35-cp35m-linux_armv7l.whl
pip3 install tflite_runtime-1.14.0-cp35-cp35m-linux_armv7l.whl
```

Then, you should be able to run the TFLite_detection scripts. I will update the get_pi_requirements.txt file so future users of this guide don't encounter this problem.

closed 2019-11-28 by EdjeElectronics

saqibshakeel035, 2019-11-28:
An off topic question regarding training of custom data. 
I am following this tutorial for the training of my data (Train images 4 and test images 2) all are car images. Everything was running fine starting from loss value =3 but as the training process proceeded, I am getting very big values. My question here is to know the reason behind this weird behavior and how can I possibly train my model right???

I1123 23:14:24.024599 8896 learning.py:507] global step 5746: loss = 3332910014858788864000.0000 (8.303 sec/step) INFO:tensorflow:global step 5747: loss = 3994775060170829463552.0000 (8.519 sec/step) I1123 23:14:32.546626 8896 learning.py:507] global step 5747: loss = 3994775060170829463552.0000 (8.519 sec/step) INFO:tensorflow:Recording summary at step 5747. I1123 23:14:38.774678 6928 supervisor.py:1050] Recording summary at step 5747. INFO:tensorflow:global step 5748: loss = 1396395248399239086080.0000 (10.488 sec/step) I1123 23:14:43.037031 8896 learning.py:507] global step 5748: loss = 1396395248399239086080.0000 (10.488 sec/step) INFO:tensorflow:global step 5749: loss = 892919831656981856256.0000 (8.293 sec/step) I1123 23:14:51.332884 8896 learning.py:507] global step 5749: loss = 892919831656981856256.0000 (8.293 sec/step) INFO:tensorflow:global step 5750: loss = 893498403471610609664.0000 (8.364 sec/step) I1123 23:14:59.700745 8896 learning.py:507] global step 5750: loss = 893498403471610609664.0000 (8.364 sec/step) INFO:tensorflow:global step 5751: loss = 4019180066551550181376.0000 (8.456 sec/step) I1123 23:15:08.158498 8896 learning.py:507] global step 5751: loss = 4019180066551550181376.0000 (8.456 sec/step) INFO:tensorflow:global step 5752: loss = 4025371108664301060096.0000 (8.323 sec/step) I1123 23:15:16.482330 8896 learning.py:507] global step 5752: loss = 4025371108664301060096.0000 (8.323 sec/step) INFO:tensorflow:global step 5753: loss = 9825390717043526860800.0000 (8.322 sec/step) I1123 23:15:24.806868 8896 learning.py:507] global step 5753: loss = 9825390717043526860800.0000 (8.322 sec/step) INFO:tensorflow:global step 5754: loss = 9830483162322176049152.0000 (8.451 sec/step) I1123 23:15:33.259524 8896 learning.py:507] global step 5754: loss = 9830483162322176049152.0000 (8.451 sec/step) INFO:tensorflow:global step 5755: loss = 2754848514362811875328.0000 (8.394 sec/step) I1123 23:15:41.656417 8896 learning.py:507] global step 5755: loss = 2754848514362811875328.0000 (8.394 sec/step) INFO:tensorflow:global step 5756: loss = 9176984960692859699200.0000 (8.338 sec/step) I1123 23:15:49.997432 8896 learning.py:507] global step 5756: loss = 9176984960692859699200.0000 (8.338 sec/step) INFO:tensorflow:global step 5757: loss = 9177446579654665175040.0000 (8.335 sec/step) I1123 23:15:58.334255 8896 learning.py:507] global step 5757: loss = 9177446579654665175040.0000 (8.335 sec/step)

Al-Martin, 2019-11-29:
Thank you so much!

saqibshakeel035, 2019-11-29:
> Hi, I'm using Raspberry Pi Model 3B+, with Raspbian (based on Debian Buster) and a Pi NoIR Camera V2 (Nigth Vision Camera for Raspberry Pi). I'm using VNC Viewer, and since there using the terminal [...]

Is your system running or you are still having difficulties?

Al-Martin, 2019-11-29:
I'm gonna test it tonight, and I'll answered you. Thanks!

saqibshakeel035, 2019-11-29:
> […](#)

Sure :)

Al-Martin, 2019-12-05:
Thank you so much!
It works perfectly!

[image: Screenshot from 2019-12-05 18-10-47.png]