EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #17: cannot connect to X server

saqibshakeel035, 2019-11-20:
I was able to resolve the issue [ImportError](https://github.com/EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi/issues/16#issue-525329245) by installing Debian Buster on RPi-3B+ but now I am encountering another problem which I guess is not that tricky but I still I am struck :(
I get the following error when I try to run:
`(tflite1-env) pi@raspberrypi:~/tflite1 $ python3 TFLite_detection_webcam.py --modeldir=Sample_TFLite_model`
`INFO: Initialized TensorFlow Lite runtime.`
`: cannot connect to X server`
`Segmentation fault`
@EdjeElectronics  any suggestion?

saqibshakeel035, 2019-11-20:
So I got to know about the issue.
It's a silly thing to run `python3 TFLite_detection_webcam.py --modeldir=Sample_TFLite_model` this command in SSH as SSH doesn't open the camera stream window.  Instead, if you don't have access to monitor. Use VNC.
100% Worked for me .
Thanks @EdjeElectronics for your assistance and this great project. :) 
Cheers

closed 2019-11-20 by saqibshakeel035

kkasha, 2019-12-30:
i also made this silly mistake of running this through SSH. Executed it form terminal in VNC worked perfectly.