EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #19: bazel run command building everytime windows CPU version.

SahilChachere, 2019-11-21:
Every time I am running the command "bazel run --config=opt tensorflow/lite/toco:toco -- --input_file=$OUTPUT_DIR/tflite_graph.pb --output_file=$OUTPUT_DIR/detect.tflite --input_shapes=1,300,300,3 --input_arrays=normalized_input_image_tensor --output_arrays=TFLite_Detection_PostProcess,TFLite_Detection_PostProcess:1,TFLite_Detection_PostProcess:2,TFLite_Detection_PostProcess:3 --inference_type=FLOAT --allow_custom_ops" to convert pb file to tflite, first it builds bazel again like this "INFO: Analysed target //tensorflow/lite/toco:toco (79 packages loaded, 3345 targets configured)."  can yoy provide solution for this? how can I permanantly set some indication that bazel is already build.

EdjeElectronics, 2019-11-28:
I'm pretty sure that's how it's supposed to work. Those messages at the beginning indicate that bazel is initializing and preparing to execute the command.

closed 2019-11-28 by EdjeElectronics

Petros626, 2021-10-05:
> Every time I am running the command "bazel run --config=opt tensorflow/lite/toco:toco -- --input_file=$OUTPUT_DIR/tflite_graph.pb --output_file=$OUTPUT_DIR/detect.tflite --input_shapes=1,300,300,3 [...]

hey are u still active on github or on this project

SahilChachere, 2021-10-05:
> hey are u still active on github or on this project

I am not working on this project.

Petros626, 2021-10-05:
@SahilChachere so you cant help me with the bazel error?