EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #21: import cv2 ModuleNotFoundError: No module named 'cv2'

hathemi, 2019-11-22:
(tensorflow-build) C:\tensorflow-build>python TFLite_detection_webcam.py --modeldir=C:\tensorflow\models\research\object_detection\coco_ssd_mobilenet_v1_1.0_quant_2018_06_29
Traceback (most recent call last):
  File "TFLite_detection_webcam.py", line 19, in <module>
    import cv2
ModuleNotFoundError: No module named 'cv2'

i have installed tensorflow V1.13.2, but i always got this error !!
how can i fix it ?

closed 2019-11-22 by hathemi

redfoo22, 2020-01-20:
I'm getting same error "No module named 'cv2" . I'm in the virtual env and I went to install opencv-python and i already have it installed. perhaps cv2 is the incorrect name for the import statement?

jasalazar, 2020-07-05:
I am having the same error on an AIY Google Vision Kit, and I cannot even install opencv-python on my Raspberry Pi Zero WH.

lilithebowman, 2020-12-15:
To get opencv to install, the name has changed to python3-opencv

lilithebowman, 2020-12-15:
This was my fix for the install script
[image]