EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #28: ERROR with USB webcam

liam-chen, 2019-12-10:
when i run TFLite_detection_webcam.py
I met this error

INFO: Initialized TensorFlow Lite runtime.
VIDEOIO ERROR: V4L2: Pixel format of incoming image is unsupported by OpenCV
VIDEOIO ERROR: V4L: can't open camera by index 0
Traceback (most recent call last):
  File "tflite/TFLite_detection_webcam.py", line 171, in <module>
    frame = frame1.copy()
AttributeError: 'NoneType' object has no attribute 'copy'

I tried two camera, but got same error.
maybe I should buy a new one?

EdjeElectronics, 2019-12-10:
Yes, unfortunately this error is caused by the webcams you're using. Can you please let me know the name/model of the webcams you used? 

See if you can borrow or buy a different webcam to use. Here's the webcam I use (it's a bit expensive): https://www.amazon.com/gp/product/B006JH8T3S

closed 2019-12-10 by EdjeElectronics

liam-chen, 2019-12-10:
thanks a lot. 
camera are "logitech quickcam express" and "generalplus technology camera"

EdjeElectronics, 2019-12-10:
Okay, thank you for letting me know. The Logitech Quickcam Express camera is very old! It came out in 1999. So I don't expect it to work with OpenCV :smile:. I think the Generalplus Technology Camera must also be kind of old.

ramkkreddy14, 2020-02-16:
even i got the same isue but i am using picamera