EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #29: Errors with step 2e

heenee77, 2019-12-11:
I am having errors after executing this step:
`bazel build --config=opt //tensorflow/tools/pip_package:build_pip_package `

Here is the error after entering the build command:
```
(tensorflow-build) D:\tensorflow-build\tensorflow>bazel build --config=opt //tensorflow/tools/pip_package:build_pip_package
WARNING: The following rc files are no longer being read, please transfer their contents or import their path into one of the standard rc files:
d:\tensorflow-build\tensorflow/.bazelrc
Starting local Bazel server and connecting to it...
WARNING: Option 'experimental_shortened_obj_file_path' is deprecated
INFO: Invocation ID: 4508ddc2-a932-49db-aa03-e21190a5425f
ERROR: error loading package '': Encountered error while reading extension file 'closure/defs.bzl': no such package '@io_bazel_rules_closure//closure': The native http_archive rule is deprecated. load("@bazel_tools//tools/build_defs/repo:http.bzl", "http_archive") for a drop-in replacement.
Use --incompatible_remove_native_http_archive=false to temporarily continue using the native rule.
ERROR: error loading package '': Encountered error while reading extension file 'closure/defs.bzl': no such package '@io_bazel_rules_closure//closure': The native http_archive rule is deprecated. load("@bazel_tools//tools/build_defs/repo:http.bzl", "http_archive") for a drop-in replacement.
Use --incompatible_remove_native_http_archive=false to temporarily continue using the native rule.
INFO: Elapsed time: 2.345s
INFO: 0 processes.
FAILED: Build did NOT complete successfully (0 packages loaded)
    Fetching @io_bazel_rules_closure; fetching
```
My session configuration also seems to be slightly different than yours as well. I am using tensorflow v 1.12
The following is my configuration session:
```
You have bazel 0.21.0- (@non-git) installed.
Please specify the location of python. [Default is D:\Anaconda3\envs\tensorflow-build\python.exe]:

Found possible Python library paths:
  D:\Anaconda3\envs\tensorflow-build\lib\site-packages
Please input the desired Python library path to use.  Default is [D:\Anaconda3\envs\tensorflow-build\lib\site-packages]

Do you wish to build TensorFlow with Apache Ignite support? [Y/n]: n
No Apache Ignite support will be enabled for TensorFlow.

Do you wish to build TensorFlow with XLA JIT support? [y/N]: n
No XLA JIT support will be enabled for TensorFlow.

Do you wish to build TensorFlow with ROCm support? [y/N]: n
No ROCm support will be enabled for TensorFlow.

Do you wish to build TensorFlow with CUDA support? [y/N]: n
No CUDA support will be enabled for TensorFlow.

Please specify optimization flags to use during compilation when bazel option "--config=opt" is specified [Default is /arch:AVX]:

Would you like to override eigen strong inline for some C++ compilation to reduce the compilation time? [Y/n]: n
Not overriding eigen strong inline, some compilations could take more than 20 mins.
```

heenee77, 2019-12-11:
Solved by downgrading bazel to 0.18.
Also I realized that for tensorflow v1.12, step 3a's bazel run command wont work.
It has to be runned from C:\tensorflow-build\tensorflow and also toco's path should be changed to //tensorflow/contrib/lite/toco rather than //tensorflow/lite/toco
`bazel run --config=opt tensorflow/contrib/lite/toco:toco -- --input_file=%OUTPUT_DIR%/tflite_graph.pb --output_file=%OUTPUT_DIR%/detect.tflite --input_shapes=1,300,300,3 --input_arrays=normalized_input_image_tensor --output_arrays=TFLite_Detection_PostProcess,TFLite_Detection_PostProcess:1,TFLite_Detection_PostProcess:2,TFLite_Detection_PostProcess:3 --inference_type=QUANTIZED_UINT8 --mean_values=128 --std_values=128 --change_concat_input_ranges=false --allow_custom_ops`
This is only true for tensorflow v1.12 though as the paths changed for versions after that.

closed 2019-12-11 by heenee77

EdjeElectronics, 2019-12-11:
Hi, thank you for updating this issue with your solution! I'm sure it will be helpful for others who run in to the same problem.