EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #30: Same FPS using TFLite on Raspberry PI 3

tomcontr, 2019-12-14:
Hi!
I was able to convert and ssd_mobilenet_v1_coco Model that I trained my self to detect, following this:

`python python export_tflite_ssd_graph.py --input_type image_tensor --pipeline_config_path training/ssd_mobilenet_v1_coco.config --trained_checkpoint_prefix training/model.ckpt-224593 --output_directory output/frozen_tflite/frozen_inference_graph.pb -add_postprocessing_op True --max_detections 1`

and then

`tflite_convert --output_file=tflite/detect.tflite \
--graph_def_file=output/frozen_tflite/frozen_inference_graph.pb/tflite_graph.pb \
--input_arrays=normalized_input_image_tensor \
--output_arrays='TFLite_Detection_PostProcess','TFLite_Detection_PostProcess:1','TFLite_Detection_PostProcess:2','TFLite_Detection_PostProcess:3' \
--input_shape=1,300,300,3 \
--allow_custom_ops`

Then I get a folder "tflite" with this files in it: detect.tflite  labelmap.txt  tflite_graph.pb  tflite_graph.pbtxt

After that I copied all the files to the Raspberry PI, and using your TFLite_detection_webcam.py script I runned the model. It Works, but wont go over 1.3 FPS. 

If I run the original version of this model, which I was running with the Object_detection_picamera.py script, it gives me same FPS. No change at all.

What could I be doing wronge?

Great tutorial by the way!! 
Regards,

EdjeElectronics, 2020-01-20:
Hi @tomcontr , I'm glad you were able to get your model trained and converted to TFLite! It does look like you did everything correctly.

Check out my performance comparison video showing FPS on the Pi 3 with regular TensorFlow vs TensorFlow Lite: https://www.youtube.com/watch?v=TiOKvOrYNII . When running at a 640x360 resolution on the Pi 3, I get 1.27 FPS with regular TensorFlow and 2.19 FPS with TensorFlow Lite. There really isn't much improvement. The big improvement comes when you use a Raspberry Pi 4 4GB model, because the extra RAM helps TFLite models run much faster. 

It's possible that if you're using a higher resolution than 640x360, there is no performance difference between regular TensorFlow and TensorFlow Lite. Unfortunately, you'll have to get a Pi 4 to see the significant difference.

closed 2020-01-20 by EdjeElectronics

vasylhiruk, 2020-02-02:
Hi @tomcontr , using the Colab notebook I have managed to obtain the files
ssd_mobilenet_v2_coco.config
label_map.pbtxt
frozen_inference_graph.pb
However I cannot convert the file .pb to .tflite neither in Colab Notebook nor at the Raspberry Pi 3.
On which platform have you converted ?

In the Colab Notebook I cannot find any legal value for input_arrays and output_arrays. The values specified in your example are not accepted. How to find out which input_arrays and output_arrays are valid ?

Petros626, 2021-10-06:
> Hi @tomcontr , using the Colab notebook I have managed to obtain the files ssd_mobilenet_v2_coco.config label_map.pbtxt frozen_inference_graph.pb However I cannot convert the file .pb to .tflite [...]

Hey I can send you a Google colab Lin where you can upload your pb file and convert it to the tflite file