EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #35: Adding new class while preserving existing ones (SSD MobileNet v2 Quantized)

elektronika-ba, 2020-01-10:
Thanks for your great tutorial. I trained my own custom class and managed to make it work on my PC. I used Google's COLAB to convert to TFLite format.

Question:
As we are using SSD MobileNet v2 model as a base for our custom training, is it possible to keep all classes that exist within it (all 90+1). I would like to detect everything it already supports including my own new class that I am adding during training. In the tutorial if I run the training it only detects my custom class(es) but without training (when downloaded) it detects all classes that COCO has.

Thanks

dangthanhhao, 2020-03-11:
> Thanks for your great tutorial. I trained my own custom class and managed to make it work on my PC. I used Google's COLAB to convert to TFLite format.
>
> Question:
> As we are using SSD MobileNet v2 [...]

Can you show me how to convert to TF lite using colab?

elektronika-ba, 2020-03-12:
You can use my shared Notebook. I described it here:

https://github.com/EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi/issues/40#issuecomment-578021023

closed 2022-09-07 by EdjeElectronics