EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #36: Unable to create model file 'detect.tflite' to use with TensorFlow Lite!

thisishasan, 2020-01-10:
I executing the following command:

> **bazel run --config=opt tensorflow/lite/toco:toco -- --input_file=../../models/research/object_detection/TFLite_model/tflite_graph.pb --output_file=../../models/research/object_detection/TFLite_model/detect.tflite --input_shapes=1,300,300,3 --input_arrays=normalized_input_image_tensor --output_arrays=TFLite_Detection_PostProcess,TFLite_Detection_PostProcess:1,TFLite_Detection_PostProcess:2,TFLite_Detection_PostProcess:3 --inference_type=QUANTIZED_UINT8 --mean_values=128 --std_values=128 --change_concat_input_ranges=false --allow_custom_ops**

Command response:

**> > INFO: Analysed target //tensorflow/lite/toco:toco (0 packages loaded).
> INFO: Found 1 target...
> Target //tensorflow/lite/toco:toco up-to-date:
> bazel-bin/tensorflow/lite/toco/toco
> INFO: Elapsed time: 0.423s, Critical Path: 0.00s
> INFO: 0 processes.
> INFO: Build completed successfully, 1 total action
> INFO: Running command line: bazel-bin/tensorflow/lite/toco/toco '--input_file=../../models/research/object_detection/TFLite_model/tflite_graph.pb' '--output_file=../../models/research/object_detection/TFLite_model/detect.tflite' '--input_shapes=1,300,300,3' '--input_arrays=normalized_input_image_tensor' '--output_arrays=TFLite_Detection_PostProcess,TFLite_Detection_PostProcess:1,TFLite_Detection_PostProcess:2,TFLite_Detection_PostProcess:3' '--inference_type=QUANTIZED_UINT8' '--mean_values=128' '--std_values=128' '--change_concat_input_ranges=false' --alINFO: Build completed successfully, 1 total action
> 2020-01-10 19:27:36.435310: F tensorflow/lite/toco/toco_convert.cc:45] Check failed: port::file::Exists(input_file.value(), port::file::Defaults()).ok() Specified input_file does not exist: ../../models/research/object_detection/TFLite_model/tflite_graph.pb.**

Although file is exists. I have checked it properly. Also I put the file in the same directory despite not working.

ArefAz, 2020-04-19:
Instead of --input-file use --graph_def_file, and --std_dev_values instead of --std_values. Also rename tflite_graph.pb to save_model.pb and tflite_graph.pbtxt to saved_model.pbtxt just in case.

simeon9696, 2020-07-31:
> Instead of --input-file use --graph_def_file, and --std_dev_values instead of --std_values. Also rename tflite_graph.pb to save_model.pb and tflite_graph.pbtxt to saved_model.pbtxt just in case.

Hi @ArefAz if I replace --input-file with --graph_def_file I get this error: 
`2020-07-31 22:07:22.856926: F tensorflow/lite/toco/toco_convert.cc:44] Check failed: input_file.specified() Missing required flag --input_file.`

Any thoughts?

ArefAz, 2020-08-01:
> Hi @ArefAz if I replace --input-file with --graph_def_file I get this error:
> `2020-07-31 22:07:22.856926: F tensorflow/lite/toco/toco_convert.cc:44] Check failed: input_file.specified() Missing [...]

Hi,

I have explained the issue and the solution I found [here](https://stackoverflow.com/questions/61749548/how-to-convert-tflite-graph-pb-to-detect-tflite-properly). I hope it helps you.

Good luck.

closed 2022-09-07 by EdjeElectronics