EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #37: IndexError: list index out of range

christianbaun, 2020-01-11:
Hello. Thanks for this great tutorial. I was able to deploy tensorflow lite on the RasPi3+ and this here works well:

`python3 TFLite_detection_webcam.py --modeldir=Sample_TFLite_model`

But when I try to calculate an own model and use it the same way (with `TFLite_detection_webcam.py`) I get the `IndexError: list index out of range` error.

I created the own model in Debian Linux 9 with TF2.

```
$ pip3 list | grep tensorflow
tensorflow                   2.0.0     
tensorflow-estimator         2.0.1     
tensorflow-hub               0.7.0     
```
I created the model whis way:

```
$ make_image_classifier \
--image_dir ~/tensorflow/images_train/ \
--tfhub_module https://tfhub.dev/google/tf2-preview/mobilenet_v2/feature_vector/4 \
--image_size 224 \
--saved_model_dir /tmp/mynewmodel \
--labels_output_file /tmp/mynewmodel/labelmap.txt \
--tflite_output_file /tmp/mynewmodel/detect.tflite
```
This is the output:

```
$ ls -lh /tmp/mynewmodel/
insgesamt 11M
drwxr-xr-x 2 bnc bnc 4,0K Jan 10 22:55 assets
-rw-r--r-- 1 bnc bnc 8,5M Jan 11 00:02 detect.tflite
-rw-r--r-- 1 bnc bnc   20 Jan 11 00:02 labelmap.txt
-rw-r--r-- 1 bnc bnc 2,0M Jan 11 00:02 saved_model.pb
drwxr-xr-x 2 bnc bnc 4,0K Jan 11 00:02 variables
```

When I try to use the self-generated tensorflow lite model on the RasPi 3+, I get this error message:

```
$ python3 TFLite_detection_webcam.py --modeldir=/home/pi/mynewmodel
INFO: Initialized TensorFlow Lite runtime.
Traceback (most recent call last):
  File "TFLite_detection_webcam.py", line 186, in <module>
    classes = interpreter.get_tensor(output_details[1]['index'])[0] # Class index of detected objects
IndexError: list index out of range
```
Every 1-2 hours, I get one of these lines as additional output:

```
Corrupt JPEG data: premature end of data segment
```

What can I do to fix this? 
Has anyone the same issue?

beige, 2020-02-04:
I ran make_image_classifier on Windows 10 and got the same error when running TFLite_detection_webcam.py on an RPI 4 with the results.

My index file contains:
daisy
dandelion
roses
sunflowers
tulips

This index appears to be correct if you don't add your own pictures and classes.

make_image_classifier --image_dir my_image_tflt dir \ --tfhub_module https://tfhub.dev/google/tf2-preview/mobilenet_v2/feature_vector/4 --image_size 224 --saved_model_dir my_dir_tflt\new_model_tflt --labels_output_file labelmap.txt --tflite_output_file new_mobile_model.tflite

Any help would be appreciated.

lery600, 2020-04-16:
i get "list index is out of range" after running "python3 TFLite_detection_webcam.py --modeldir=TFLite_model". it works only about a few seconds or if it's dark. as soon as it starts seeing any picture it freezes. i followed this guide and i trained my own model with only one label. 
File "TFLite_detection_webcam.py", line 204, in <module>
object_name = labels[int(classes[i])]
IndexError: list index out of range

cobalt67, 2020-06-08:
classes = interpreter.get_tensor(output_details[1]['index'])[0] # Class index of detected objects
IndexError:
list index out of range

I am also trying to run my own model in Rpi. But the same issue is occurring.
Any solution to this issue?

hushki610, 2020-06-17:
i have the same problem.. did anyone came across a solution?

EdjeElectronics, 2020-06-17:
Hi all, the "list index out of range" error is occurring because you are using an "image classification" model rather than an "object detection" model. Image classification models apply a single label to an image, while object detection models locate and label multiple objects in an image. This code is written for object detection models.

If you'd like to see how to use an image classification model on the Raspberry Pi, please see this example:
https://github.com/tensorflow/examples/tree/master/lite/examples/image_classification/raspberry_pi

closed 2020-06-17 by EdjeElectronics

sharmila-polamuri, 2021-03-24:
Hi @EdjeElectronics 
I built an object detection model and converted it to tflite model format. But I got `IndexError: list index out of range` error when I try to predict images/video using tflite formated model.  Can you please help me to resolve this problem?

seung-gu, 2021-04-16:
It's because there are missing classes in "labelmap.txt" file.
Class number is up to 89, which was supposed to be 90 classes for total, but if you count it, it's actually 80 classes (10 classes are removed). Therefore, "object_name = labels[int(classes[i])]" this line doesn't work. What you need to do is to fill in the missing numbers with just garbage class names such as "??", and it will work if you just make the number of column of labelmap.txt file with 90 lines.

PiotrG1996, 2021-05-31:
This should solve the problem. 
https://github.com/EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi/blob/master/Raspberry_Pi_Guide.md#5-indexerror-list-index-out-of-range

1. Download correct detect.tflite file for object detection.
2. Create file labelmap.txt and paste all objects in this way:

???
person
bicycle
car
motorcycle
airplane
bus
train
truck
boat
traffic light
fire hydrant
???
stop sign
parking meter
bench
bird
cat
dog
horse
sheep
cow
elephant
bear
zebra
giraffe
???
backpack
umbrella
???
???
handbag
tie
suitcase
frisbee
skis
snowboard
sports ball
kite
baseball bat
baseball glove
skateboard
surfboard
tennis racket
bottle
???
wine glass
cup
fork
knife
spoon
bowl
banana
apple
sandwich
orange
broccoli
carrot
hot dog
pizza
donut
cake
chair
couch
potted plant
bed
???
dining table
???
???
toilet
???
tv
laptop
mouse
remote
keyboard
cell phone
microwave
oven
toaster
sink
refrigerator
???
book
clock
vase
scissors
teddy bear
hair drier
toothbrush

AlexM1989, 2021-07-25:
Still no solution? I still have the same issue

  File "TFLite_detection_webcam.py", line 186, in <module>
    classes = interpreter.get_tensor(output_details[1]['index'])[0] # Class index of detected objects
IndexError: list index out of range

markfilan, 2022-01-17:
The way Python indexing works is that it starts at 0, so the first number of your list would be [0]. Index Error: List index out of range means you are trying to get an item in a list that doesn't exist. It means that you are referring to n-th element of the [python list](http://net-informations.com/python/ds/lists.htm), while the length of the list is smaller than n. whenever you get this type of error please cross check with items that comes between/middle in range, and insure that their index is not last if you get output then you have made perfect error that mentioned.

An index in Python refers to a position within an ordered list . To retrieve an element of the list, you use the index operator ([]) . Using indexing you can easily get any element by its position.