EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #39: Problems while training ssd_mobilenet_v2_quantized_coco on GPU but same works well for faster_rcnn_inception_v2_coco

saqibshakeel035, 2020-01-17:
I am trying to train a custom model that I will use later on raspberry pi for object detection. The model I want to train using GPU TensorFlow is  **ssd_mobilenet_v2_quantized_coco**  but when I try to run the training, itload all the gpu files successfuly but ran into the error of memory which surprisingly works perfectly well in case of training the **faster_rcnn_inception_v2_coco**. CPU version of TensorFlow works fine with both of the models.
 
My system specifications are:
**Operating System:**       Windows 10
**Graphics Card :**             Nvidia MX250
**Ram              :**               16GB
**Processor:**                      Intel core i7-10th Gen.
**Tensorflow Version:**        Tensorflow GPU 1.15.0
**Cuda :**                              10.0
**CuDNN :**                         7.4.6 (As the enssorflow model-master was compiled using this version)
Any recommendations to that helps me fasten my SSD network training process will be highy appreciated.

elektronika-ba, 2020-01-17:
Try lowering the "batch_size" parameter in your training config file.

saqibshakeel035, 2020-01-17:
> Try lowering the "batch_size" parameter in your training config file.

That is already 6. :(

elektronika-ba, 2020-01-17:
Try lower, maybe your training images are too large?

saqibshakeel035, 2020-01-17:
Okay let me try and I will come back.  If not, I will post the exact error for your better understanding. 

> Try lower, maybe your training images are too large?

elektronika-ba, 2020-01-17:
OK, I just finished my SSD training too, and I am fresh at it.
Mind you, I used Google's COLAB for training, as they have much more poweful GPUs than what I have at home.
I also resized/cropped my training images to be 300x300px and I also had some slightly larger than 300x300 too. My model has only 1 class. If you are interested in using Google's COLAB I am happy to help as well.

saqibshakeel035, 2020-01-17:
> OK, I just finished my SSD training too, and I am fresh at it.
> Mind you, I used Google's COLAB for training, as they have much more poweful GPUs than what I have at home.
> I also resized/cropped my [...]

So here is the error when I have  Batch_Size=6
`Please switch to tf.train.MonitoredTrainingSession
W0117 16:24:52.586001 15696 deprecation.py:323] From C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\contrib\slim\python\slim\learning.py:742: Supervisor.__init__ (from tensorflow.python.training.supervisor) is deprecated and will be removed in a future version.
Instructions for updating:
Please switch to tf.train.MonitoredTrainingSession
2020-01-17 16:24:55.158174: I tensorflow/core/platform/cpu_feature_guard.cc:142] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2
2020-01-17 16:24:55.167479: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library nvcuda.dll
2020-01-17 16:24:55.988934: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1618] Found device 0 with properties:
name: GeForce MX250 major: 6 minor: 1 memoryClockRate(GHz): 1.582
pciBusID: 0000:3b:00.0
2020-01-17 16:24:55.992960: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library cudart64_100.dll
2020-01-17 16:24:58.202813: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library cublas64_100.dll
2020-01-17 16:25:01.267440: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library cufft64_100.dll
2020-01-17 16:25:02.980329: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library curand64_100.dll
2020-01-17 16:25:06.881732: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library cusolver64_100.dll
2020-01-17 16:25:08.663803: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library cusparse64_100.dll
2020-01-17 16:25:22.792813: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library cudnn64_7.dll
2020-01-17 16:25:23.259035: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1746] Adding visible gpu devices: 0
2020-01-17 16:25:25.111022: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1159] Device interconnect StreamExecutor with strength 1 edge matrix:
2020-01-17 16:25:25.113254: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1165]      0
2020-01-17 16:25:25.114749: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1178] 0:   N
2020-01-17 16:25:25.118370: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1304] Created TensorFlow device (/job:localhost/replica:0/task:0/device:GPU:0 with 1356 MB memory) -> physical GPU (device: 0, name: GeForce MX250, pci bus id: 0000:3b:00.0, compute capability: 6.1)
INFO:tensorflow:Restoring parameters from training/model.ckpt-20024
I0117 16:25:25.128645 15696 saver.py:1284] Restoring parameters from training/model.ckpt-20024
WARNING:tensorflow:From C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\training\saver.py:1069: get_checkpoint_mtimes (from tensorflow.python.training.checkpoint_management) is deprecated and will be removed in a future version.
Instructions for updating:
Use standard file utilities to get mtimes.
W0117 16:25:28.945794 15696 deprecation.py:323] From C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\training\saver.py:1069: get_checkpoint_mtimes (from tensorflow.python.training.checkpoint_management) is deprecated and will be removed in a future version.
Instructions for updating:
Use standard file utilities to get mtimes.
INFO:tensorflow:Running local_init_op.
I0117 16:25:28.959755 15696 session_manager.py:500] Running local_init_op.
INFO:tensorflow:Done running local_init_op.
I0117 16:25:30.059591 15696 session_manager.py:502] Done running local_init_op.
INFO:tensorflow:Starting Session.
I0117 16:25:46.880596 15696 learning.py:754] Starting Session.
INFO:tensorflow:Saving checkpoint to path training/model.ckpt
I0117 16:25:47.849779 18632 supervisor.py:1117] Saving checkpoint to path training/model.ckpt
INFO:tensorflow:Starting Queues.
I0117 16:25:47.855558 15696 learning.py:768] Starting Queues.
INFO:tensorflow:global_step/sec: 0
I0117 16:26:01.673384  4716 supervisor.py:1099] global_step/sec: 0
2020-01-17 16:26:05.421051: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library cudnn64_7.dll
2020-01-17 16:26:06.634071: W tensorflow/stream_executor/cuda/redzone_allocator.cc:312] Internal: Invoking ptxas not supported on Windows
Relying on driver to perform ptx compilation. This message will be only logged once.
2020-01-17 16:26:06.717876: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library cublas64_100.dll
INFO:tensorflow:Recording summary at step 20024.
I0117 16:26:09.918403  9180 supervisor.py:1050] Recording summary at step 20024.

2020-01-17 16:27:03.917594: I tensorflow/core/common_runtime/bfc_allocator.cc:917] 2 Chunks of size 14592 totalling 28.5KiB
2020-01-17 16:27:03.940242: I tensorflow/core/common_runtime/bfc_allocator.cc:917] 4 Chunks of size 18432 totalling 72.0KiB
2020-01-17 16:27:03.962538: I tensorflow/core/common_runtime/bfc_allocator.cc:917] 7 Chunks of size 20736 totalling 141.8KiB
[... cut ...]
         [[{{node gradients/FeatureExtractor/MobilenetV2/expanded_conv_2/depthwise/Relu6_grad/Relu6Grad-1-TransposeNHWCToNCHW-LayoutOptimizer}}]]
Hint: If you want to see a list of allocated tensors when OOM happens, add report_tensor_allocations_upon_oom to RunOptions for current allocation info.

         [[FeatureExtractor/MobilenetV2/expanded_conv_6/expand/Relu6/_3087]]
Hint: If you want to see a list of allocated tensors when OOM happens, add report_tensor_allocations_upon_oom to RunOptions for current allocation info.

0 successful operations.
0 derived errors ignored.`

saqibshakeel035, 2020-01-17:
After reducing the batch_size to 1. I get the following outcome;

```Original stack trace for 'CheckNumerics':
  File "train.py", line 185, in <module>
    tf.app.run()
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\platform\app.py", line 40, in run
    _run(main=main, argv=argv, flags_parser=_parse_flags_tolerate_undef)
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\absl\app.py", line 299, in run
    _run_main(main, args)
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\absl\app.py", line 250, in _run_main
    sys.exit(main(argv))
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\util\deprecation.py", line 324, in new_func
    return func(*args, **kwargs)
  File "train.py", line 181, in main
    graph_hook_fn=graph_rewriter_fn)
  File "C:\tensorflow1\models\research\object_detection\legacy\trainer.py", line 323, in train
    total_loss = tf.check_numerics(total_loss, 'LossTensor is inf or nan.')
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\ops\gen_array_ops.py", line 1011, in check_numerics
    "CheckNumerics", tensor=tensor, message=message, name=name)
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\framework\op_def_library.py", line 794, in _apply_op_helper
    op_def=op_def)
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\util\deprecation.py", line 507, in new_func
    return func(*args, **kwargs)
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\framework\ops.py", line 3357, in create_op
    attrs, op_def, compute_device)
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\framework\ops.py", line 3426, in _create_op_internal
    op_def=op_def)
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\framework\ops.py", line 1748, in __init__
    self._traceback = tf_stack.extract_stack()

I0117 16:30:56.307994 16520 coordinator.py:224] Error reported to Coordinator: <class 'tensorflow.python.framework.errors_impl.InvalidArgumentError'>, 2 root error(s) found.
  (0) Invalid argument: LossTensor is inf or nan. : Tensor had NaN values
         [[node CheckNumerics (defined at C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\framework\ops.py:1748) ]]
  (1) Invalid argument: LossTensor is inf or nan. : Tensor had NaN values
         [[node CheckNumerics (defined at C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\framework\ops.py:1748) ]]
         [[gradients/BoxPredictor_0/ClassPredictor/BiasAdd_grad/BiasAddGrad/_4330]]
0 successful operations.
0 derived errors ignored.

Original stack trace for 'CheckNumerics':
  File "train.py", line 185, in <module>
    tf.app.run()
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\platform\app.py", line 40, in run
    _run(main=main, argv=argv, flags_parser=_parse_flags_tolerate_undef)
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\absl\app.py", line 299, in run
    _run_main(main, args)
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\absl\app.py", line 250, in _run_main
    sys.exit(main(argv))
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\util\deprecation.py", line 324, in new_func
    return func(*args, **kwargs)
  File "train.py", line 181, in main
    graph_hook_fn=graph_rewriter_fn)
  File "C:\tensorflow1\models\research\object_detection\legacy\trainer.py", line 323, in train
    total_loss = tf.check_numerics(total_loss, 'LossTensor is inf or nan.')
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\ops\gen_array_ops.py", line 1011, in check_numerics
    "CheckNumerics", tensor=tensor, message=message, name=name)
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\framework\op_def_library.py", line 794, in _apply_op_helper
    op_def=op_def)
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\util\deprecation.py", line 507, in new_func
    return func(*args, **kwargs)
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\framework\ops.py", line 3357, in create_op
    attrs, op_def, compute_device)
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\framework\ops.py", line 3426, in _create_op_internal
    op_def=op_def)
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\framework\ops.py", line 1748, in __init__
    self._traceback = tf_stack.extract_stack()

Traceback (most recent call last):
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\client\session.py", line 1365, in _do_call
    return fn(*args)
[... 54 lines cut ...]
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\platform\app.py", line 40, in run
    _run(main=main, argv=argv, flags_parser=_parse_flags_tolerate_undef)
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\absl\app.py", line 299, in run
    _run_main(main, args)
[... cut ...]
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\framework\ops.py", line 3357, in create_op
    attrs, op_def, compute_device)
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\framework\ops.py", line 3426, in _create_op_internal
    op_def=op_def)
  File "C:\Users\saqib\Anaconda3\envs\tensorflow1\lib\site-packages\tensorflow_core\python\framework\ops.py", line 1748, in __init__
    self._traceback = tf_stack.extract_stack()
```

elektronika-ba, 2020-01-20:
I googled a bit and ended up here: https://github.com/tensorflow/models/issues/6392

> Invalid argument: LossTensor is inf or nan. : Tensor had NaN values

It seems that there might be a problem with your masks/labels on images, or at least on some images. Can you post your label XML file here, of one image? It might be that label is too small after it gets scaled down to 300x300.

I do not know what would happen if you provide a non-rectangle labels in your XML files while training the SSD. Maybe you have non-rectangles in XML files?

saqibshakeel035, 2020-01-20:
> I googled a bit and ended up here: [tensorflow/models#6392](https://github.com/tensorflow/models/issues/6392)
> 
> > Invalid argument: LossTensor is inf or nan. : Tensor had NaN values
> 
> It seems that there might be a problem with your masks/labels on images, or at least on some images. Can you post your label XML file here, of one image? It might be that label is too small after it gets scaled down to 300x300.
> 
> I do not know what would happen if you provide a non-rectangle labels in your XML files while training the SSD. Maybe you have non-rectangles in XML files?

```
<annotation>
	<folder>train</folder>
	<filename>0_2019-12-15_16.14.25_dec.jpg</filename>
	<path>C:\tensorflow1\models\research\object_detection\images\train\0_2019-12-15_16.14.25_dec.jpg</path>
	<source>
		<database>Unknown</database>
	</source>
	<size>
		<width>1040</width>
		<height>808</height>
		<depth>3</depth>
	</size>
	<segmented>0</segmented>
	<object>
		<name>car</name>
		<pose>Unspecified</pose>
		<truncated>0</truncated>
		<difficult>0</difficult>
		<bndbox>
			<xmin>484</xmin>
			<ymin>443</ymin>
			<xmax>1038</xmax>
			<ymax>771</ymax>
		</bndbox>
	</object>
</annotation>

```
This is how my annotation file looks like. Also, I am not converting the images into 300*300 as I want to train my model from high-resolution files.

elektronika-ba, 2020-01-20:
XML looks fine.
You do know that any image larger than 300x300px will be scaled down automatically by the training process? Since your photo is not 1:1 (height:width) it will also get distorted during this process.

Anyway, my objects were smaller so I don't have experience with high-res photos in regards to the SSD training.

Just to narrow down the problem, you could reduce the number of images in /train and /test folders to, for example 10 and 2, and go over each one manually to verify that labels are OK, and then run the training on small number of images. If it works, then you have error in labels on some of your remaining images.

Also, try making experimental 300x300 images and try training on them. See what happens.

saqibshakeel035, 2020-01-20:
Also, since I am using LabelImg so it only works with rectangles.

saqibshakeel035, 2020-01-20:
Okay, that's good info. I have images with different sizes, don't you think to convert them to 300*300 changes their sizes and also I labeled images before resizing. If I resize them, the bounding boxes will be of no use as per my understating. What do you think?
Anyhow, I will try to consider your points when I will train my model in couple of days and will post the outcome.
Thanks for assisting :)

elektronika-ba, 2020-01-20:
> I have images with different sizes, don't you think to convert them to 300*300 changes their sizes and also I labeled images before resizing. If I resize them, the bounding boxes will be of no use as per my understating.

This is true, and there must be scripts online that resize images and resize bounding-boxes accordingly as well.

What I meant for you to do (as a test) is to manually get 12 images, resize them to 300x300 and do the labeling manually. Run training on these 10 images (remaining 2 use for /test folder) just to see if Tensorflow SSD training works with these smaller images.

saqibshakeel035, 2020-01-20:
> > I have images with different sizes, don't you think to convert them to 300*300 changes their sizes and also I labeled images before resizing. If I resize them, the bounding boxes will be of no use as per my understating.
> 
> This is true, and there must be scripts online that resize images and resize bounding-boxes accordingly as well.
> 
> What I meant for you to do (as a test) is to manually get 12 images, resize them to 300x300 and do the labeling manually. Run training on these 10 images (remaining 2 use for /test folder) just to see if Tensorflow SSD training works with these smaller images.
> 
> P.S. I am not an expert here, still learning as I go.

Yeah, I completely understand what you said in your last comment. :)
Will get back to you soon. 
Have a great day!!

closed 2022-09-07 by EdjeElectronics