EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #40: Error while building tensorflow with bazel

AndrejPiecka, 2020-01-20:
Hello, i got this error:
```
ERROR: C:/tensorflow-build/tensorflow/tensorflow/core/BUILD:2515:1: Executing genrule //tensorflow/core:version_info_gen failed (Exit 2): bash.exe failed: error executing command
  cd C:/users/*user*/_bazel_*user*/j7bi4x5j/execroot/org_tensorflow
  SET PATH=C:\msys64\usr\bin;C:\msys64\bin
    SET PYTHON_BIN_PATH=C:/Users/*user*/Anaconda3/envs/tensorflow-build/python.exe
    SET PYTHON_LIB_PATH=C:/tensorflow1/models/research/slim
    SET TF_DOWNLOAD_CLANG=0
    SET TF_NEED_CUDA=0
    SET TF_NEED_OPENCL_SYCL=0
    SET TF_NEED_ROCM=0
  C:/msys64/usr/bin/bash.exe -c source external/bazel_tools/tools/genrule/genrule-setup.sh; bazel-out/x64_windows-opt/bin/tensorflow/tools/git/gen_git_source.exe --generate external/local_config_git/gen/spec.json external/local_config_git/gen/head external/local_config_git/gen/branch_ref "bazel-out/x64_windows-opt/genfiles/tensorflow/core/util/version_info.cc" --git_tag_override=${GIT_TAG_OVERRIDE:-}
Execution platform: @bazel_tools//platforms:host_platform
C:/Users/*user*/Anaconda3/envs/tensorflow-build/python.exe: can't open file 'C:\users\*user*': [Errno 2] No such file or directory
Target //tensorflow/tools/pip_package:build_pip_package failed to build
INFO: Elapsed time: 45,888s, Critical Path: 3,68s
INFO: 7 processes: 7 local.
FAILED: Build did NOT complete successfully
```
Please help.

elektronika-ba, 2020-01-20:
Is your windows username really somehow called: &ast;user&ast;?
Because &ast;user&ast; is an invalid directory name in Windows OS because of asterisk, hence the error:

> C:/Users/&ast;user&ast;/Anaconda3/envs/tensorflow-build/python.exe: can't open file 'C:/users/&ast;user*': [Errno 2] No such file or directory

AndrejPiecka, 2020-01-20:
Hello, thank you for reply.
I changed it to *user* because of personal information (i didn't know it was important).
Here is changed user to "me and you" (I have something similar)
```
ERROR: C:/tensorflow-build/tensorflow/tensorflow/core/BUILD:2515:1: Executing genrule //tensorflow/core:version_info_gen failed (Exit 2): bash.exe failed: error executing command
  cd C:/users/me and you/_bazel_me and you/j7bi4x5j/execroot/org_tensorflow
  SET PATH=C:\msys64\usr\bin;C:\msys64\bin
    SET PYTHON_BIN_PATH=C:/Users/me and you/Anaconda3/envs/tensorflow-build/python.exe
    SET PYTHON_LIB_PATH=C:/tensorflow1/models/research/slim
    SET TF_DOWNLOAD_CLANG=0
    SET TF_NEED_CUDA=0
    SET TF_NEED_OPENCL_SYCL=0
    SET TF_NEED_ROCM=0
  C:/msys64/usr/bin/bash.exe -c source external/bazel_tools/tools/genrule/genrule-setup.sh; bazel-out/x64_windows-opt/bin/tensorflow/tools/git/gen_git_source.exe --generate external/local_config_git/gen/spec.json external/local_config_git/gen/head external/local_config_git/gen/branch_ref "bazel-out/x64_windows-opt/genfiles/tensorflow/core/util/version_info.cc" --git_tag_override=${GIT_TAG_OVERRIDE:-}
Execution platform: @bazel_tools//platforms:host_platform
C:/Users/me and you/Anaconda3/envs/tensorflow-build/python.exe: can't open file 'C:\users\me and': [Errno 2] No such file or directory
Target //tensorflow/tools/pip_package:build_pip_package failed to build
INFO: Elapsed time: 45,888s, Critical Path: 3,68s
INFO: 7 processes: 7 local.
FAILED: Build did NOT complete successfully
```
Note: The "can't open file 'C:\users\**me and**': [Errno 2] No such file or directory" is not a mistake, it gave me output like that.

elektronika-ba, 2020-01-21:
I see. That was the only thing that looked bad to me.

Now, if this is not the problem, I really don't know what is:

> TensorFlow doesn't yet support Bazel 1.x (it's coming, but not yet done), so yes you'll need 0.29.1 until then.

https://stackoverflow.com/questions/58547555/unable-to-compile-tensorflow-from-source-on-an-intel-i7-930-cpu-gts-250-gpu

AndrejPiecka, 2020-01-21:
Is it possible to convert trained model from tensorflow to tensorflow-lite on raspberry?

elektronika-ba, 2020-01-21:
I also gave up on compiling Tensorflow on my PC and went on to Google's COLAB where you have a possibility to convert it there.

Here is my shared notebook for this task. In few steps you will get your model converted to TFLite:
https://colab.research.google.com/drive/1MUL7JZDEFYs3A9-xj39UXfx3uJi4aH-x

AndrejPiecka, 2020-01-21:
I tried it and got this error.

```
2020-01-21 16:06:51.637543: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcuda.so.1
2020-01-21 16:06:51.712363: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:983] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-01-21 16:06:51.712932: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1618] Found device 0 with properties: 
name: Tesla T4 major: 7 minor: 5 memoryClockRate(GHz): 1.59
pciBusID: 0000:00:04.0
2020-01-21 16:06:51.720933: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudart.so.10.1
2020-01-21 16:06:51.972007: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcublas.so.10
2020-01-21 16:06:52.117293: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcufft.so.10
2020-01-21 16:06:52.137871: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcurand.so.10
2020-01-21 16:06:52.400421: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcusolver.so.10
2020-01-21 16:06:52.447223: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcusparse.so.10
2020-01-21 16:06:52.961105: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudnn.so.7
2020-01-21 16:06:52.961348: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:983] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-01-21 16:06:52.962200: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:983] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-01-21 16:06:52.962781: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1746] Adding visible gpu devices: 0
2020-01-21 16:06:53.030206: I tensorflow/core/platform/profile_utils/cpu_utils.cc:94] CPU Frequency: 2300000000 Hz
2020-01-21 16:06:53.030413: I tensorflow/compiler/xla/service/service.cc:168] XLA service 0x562ca0d06d80 initialized for platform Host (this does not guarantee that XLA will be used). Devices:
2020-01-21 16:06:53.030439: I tensorflow/compiler/xla/service/service.cc:176]   StreamExecutor device (0): Host, Default Version
2020-01-21 16:06:53.170649: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:983] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-01-21 16:06:53.171356: I tensorflow/compiler/xla/service/service.cc:168] XLA service 0x562ca0d072c0 initialized for platform CUDA (this does not guarantee that XLA will be used). Devices:
2020-01-21 16:06:53.171386: I tensorflow/compiler/xla/service/service.cc:176]   StreamExecutor device (0): Tesla T4, Compute Capability 7.5
2020-01-21 16:06:53.172149: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:983] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-01-21 16:06:53.172715: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1618] Found device 0 with properties: 
name: Tesla T4 major: 7 minor: 5 memoryClockRate(GHz): 1.59
pciBusID: 0000:00:04.0
2020-01-21 16:06:53.172757: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudart.so.10.1
2020-01-21 16:06:53.172777: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcublas.so.10
2020-01-21 16:06:53.172822: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcufft.so.10
2020-01-21 16:06:53.172837: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcurand.so.10
2020-01-21 16:06:53.172852: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcusolver.so.10
2020-01-21 16:06:53.172865: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcusparse.so.10
2020-01-21 16:06:53.172878: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudnn.so.7
2020-01-21 16:06:53.172941: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:983] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-01-21 16:06:53.173487: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:983] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-01-21 16:06:53.174017: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1746] Adding visible gpu devices: 0
2020-01-21 16:06:53.177044: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudart.so.10.1
2020-01-21 16:06:53.178189: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1159] Device interconnect StreamExecutor with strength 1 edge matrix:
[... cut ...] rmat, or run quantized training with your model from a floating point checkpoint to change the input graph to contain min/max information. If you don't care about accuracy, you can pass --default_ranges_min= and --default_ranges_max= for easy experimentation.
Aborted (core dumped)
```
Do I have to make a quantized model for better accuracy/ speed? Because I was using ssd_mobilenet_v2_coco.

This error is saying, that i should add  --default_ranges_min= and --default_ranges_max=. But what values?

elektronika-ba, 2020-01-22:
Yes, this was for quantized model. Quantized must run faster because it does not do computations with floats. As Edje says in his tutorial:
> You can also use a standard SSD-MobileNet model (V1 or V2), but it will not run quite as fast as the quantized model.

There are probably TOCO parameters for un-quantized model as well, but I have no idea what they might look like.

I also have a notebook for training on Google's COLAB, if you want it let me know.

AndrejPiecka, 2020-01-22:
Thanks I will train it for quantized model. I would like to try that Google's colab.

elektronika-ba, 2020-01-23:
Here is the Notebook on Google Colab:

https://colab.research.google.com/drive/1gSd6GpXp1OgJoAFs_kvBbsJVdMNA45os

I tweaked it a little bit without testing. If you run into problems let me know. Just issue steps one by one.

AndrejPiecka, 2020-01-23:
Error in the # Run the training!!!
```
/usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:526: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:527: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:528: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:529: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:530: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:535: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])

WARNING: The TensorFlow contrib module will not be included in TensorFlow 2.0.
For more information, please see:
  * https://github.com/tensorflow/community/blob/master/rfcs/20180907-contrib-sunset.md
  * https://github.com/tensorflow/addons
If you depend on functionality not listed there, please file an issue.

WARNING:tensorflow:From /usr/local/lib/python3.6/dist-packages/tensorflow/python/platform/app.py:125: main (from __main__) is deprecated and will be removed in a future version.
Instructions for updating:
Use object_detection/model_main.py.
Traceback (most recent call last):
  File "train.py", line 184, in <module>
    tf.app.run()
  File "/usr/local/lib/python3.6/dist-packages/tensorflow/python/platform/app.py", line 125, in run
    _sys.exit(main(argv))
  File "/usr/local/lib/python3.6/dist-packages/tensorflow/python/util/deprecation.py", line 324, in new_func
    return func(*args, **kwargs)
  File "train.py", line 93, in main
    FLAGS.pipeline_config_path)
  File "/content/tensorflow1/models/research/object_detection/utils/config_util.py", line 95, in get_configs_from_pipeline_file
    proto_str = f.read()
  File "/usr/local/lib/python3.6/dist-packages/tensorflow/python/lib/io/file_io.py", line 125, in read
    self._preread_check()
  File "/usr/local/lib/python3.6/dist-packages/tensorflow/python/lib/io/file_io.py", line 85, in _preread_check
    compat.as_bytes(self.__name), 1024 * 512, status)
  File "/usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/errors_impl.py", line 528, in __exit__
    c_api.TF_GetCode(self.status.status))
tensorflow.python.framework.errors_impl.NotFoundError: /content/tensorflow1/models/research/object_detection/training/ssd_mobilenet_v2_quantized_300x300_coco.config; No such file or directory
```
And before I got this:
```
mkdir: cannot create directory ‘./training’: File exists
mv: './training/labelmap.pbtxt' and './training/labelmap.pbtxt' are the same file
mv: './training/ssd_mobilenet_v2_quantized_300x300_coco.config' and './training/ssd_mobilenet_v2_quantized_300x300_coco.config' are the same file
Copied training/
Copied images.zip
```
in the 
# Copy already prepared stuff from our GDrive to object_detection/ where we should currently be step

I don't know if it's related.

I putted the ssd_mobilenet_v2_quantized_300x300_coco.config and labelmap.pbtxt in the SSD_TRAINER/training.

elektronika-ba, 2020-01-24:
I fixed it, sorry.

I just tested it and it works:

```
INFO:tensorflow:global step 1: loss = 13.5873 (32.959 sec/step)
INFO:tensorflow:global step 2: loss = 12.4880 (1.242 sec/step)
INFO:tensorflow:global step 3: loss = 11.5915 (1.218 sec/step)
INFO:tensorflow:global step 4: loss = 10.5284 (1.202 sec/step)
INFO:tensorflow:global step 5: loss = 10.5109 (1.188 sec/step)
INFO:tensorflow:global step 6: loss = 10.6667 (1.212 sec/step)
INFO:tensorflow:global step 7: loss = 10.0286 (1.220 sec/step)
INFO:tensorflow:global step 8: loss = 9.3836 (1.222 sec/step)
INFO:tensorflow:global step 9: loss = 8.7306 (1.196 sec/step)
INFO:tensorflow:global step 10: loss = 8.9945 (1.220 sec/step)
INFO:tensorflow:global step 11: loss = 8.0672 (1.193 sec/step)
INFO:tensorflow:global step 12: loss = 8.6084 (1.196 sec/step)
INFO:tensorflow:global step 13: loss = 7.5563 (1.202 sec/step)
INFO:tensorflow:global step 14: loss = 7.9247 (1.211 sec/step)
INFO:tensorflow:global step 15: loss = 7.8446 (1.247 sec/step)
INFO:tensorflow:global step 16: loss = 8.0108 (1.214 sec/step)
INFO:tensorflow:global step 17: loss = 7.1595 (1.207 sec/step)
INFO:tensorflow:global step 18: loss = 7.4473 (1.218 sec/step)
INFO:tensorflow:global step 19: loss = 7.0707 (1.212 sec/step)
INFO:tensorflow:global step 20: loss = 6.7756 (1.208 sec/step)
INFO:tensorflow:global step 21: loss = 7.2939 (1.204 sec/step)
... and so on ...
```

Here it is again: https://colab.research.google.com/drive/1gSd6GpXp1OgJoAFs_kvBbsJVdMNA45os

And for convenience, you can see my /SSD_TRAINER folder here: https://drive.google.com/drive/folders/1yd-mGxzfj5xduO--y-E85SypocqEOjc5?usp=sharing

AndrejPiecka, 2020-01-24:
I am getting:
```
INFO:tensorflow:Starting Queues.
INFO:tensorflow:global_step/sec: 0
INFO:tensorflow:global_step/sec: 0
INFO:tensorflow:global_step/sec: 0
```
Do I need to make the images 300x300 pixels?

elektronika-ba, 2020-01-27:
I also get that notification, but then after a while it starts working and reporting those steps to me:

> INFO:tensorflow:global step 1: loss = ...

You can try smaller images, or even my prepared images, just to get it running and then you can play with it:
https://drive.google.com/drive/folders/1yd-mGxzfj5xduO--y-E85SypocqEOjc5?usp=sharing

dangthanhhao, 2020-02-28:
> I also get that notification, but then after a while it starts working and reporting those steps to me:
>
>
> You can try smaller images, or even my prepared images, just to get it running and then you [...]
I 've followed your Collab script but where is the label text file for TFlite?
One more question, Output model can detect the car (your train) and other thing (Toco pretrain) or just the car(your train)?

tazelisk, 2020-03-30:
> I gave up on compiling Tensorflow on my PC and went on to Google's COLAB where you have a possibility to convert it there.
> 
> Here is my shared notebook for this task. In few steps you will get your model converted to TFLite:
> https://colab.research.google.com/drive/1MUL7JZDEFYs3A9-xj39UXfx3uJi4aH-x
> 
> EDIT: This is for quantized model.

> I gave up on compiling Tensorflow on my PC and went on to Google's COLAB where you have a possibility to convert it there.
> 
> Here is my shared notebook for this task. In few steps you will get your model converted to TFLite:
> https://colab.research.google.com/drive/1MUL7JZDEFYs3A9-xj39UXfx3uJi4aH-x
> 
> EDIT: This is for quantized model.

Hey, you were really my strike for light, I tried everything for almost 14 days now and I should do a project for my university. 

Unfortunately I get this error using your notebook.

usage: toco [-h] --output_file OUTPUT_FILE
            [--saved_model_dir SAVED_MODEL_DIR | --keras_model_file KERAS_MODEL_FILE]
            [--enable_v1_converter] [--experimental_new_converter]
toco: error: one of the arguments --saved_model_dir --keras_model_file is required

Do you have any solution on that?

I even tried both version (quantized and non-quantized by changing the inference_type to FLOAT...

Thank you (:

elektronika-ba, 2020-04-08:
I confirm that the Notebook does not work **anymore**. Something changed on the Google Colab, probably the version of TOCO that is there. I am going to investigate...

I hate these things.

elektronika-ba, 2020-04-08:
Oops, my bad. It actually still works, it just does not say that it has done with the conversion

> /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:526: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>   _np_qint8 = np.dtype([("qint8", np.int8, 1)])
> /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:527: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>   _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
> /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:528: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>   _np_qint16 = np.dtype([("qint16", np.int16, 1)])
> /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:529: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>   _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
> /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:530: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>   _np_qint32 = np.dtype([("qint32", np.int32, 1)])
> /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:535: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>   np_resource = np.dtype([("resource", np.ubyte, 1)])
> 2020-04-08 13:42:00.517632: I tensorflow/core/platform/cpu_feature_guard.cc:141] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
> 2020-04-08 13:42:00.589598: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:998] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
> 2020-04-08 13:42:00.590477: I tensorflow/compiler/xla/service/service.cc:150] XLA service 0x1c46f20 executing computations on platform CUDA. Devices:
> 2020-04-08 13:42:00.590514: I tensorflow/compiler/xla/service/service.cc:158]   StreamExecutor device (0): Tesla K80, Compute Capability 3.7
> 2020-04-08 13:42:00.592585: I tensorflow/core/platform/profile_utils/cpu_utils.cc:94] CPU Frequency: 2300000000 Hz
> 2020-04-08 13:42:00.592799: I tensorflow/compiler/xla/service/service.cc:150] XLA service 0x1c46dc0 executing computations on platform Host. Devices:
> 2020-04-08 13:42:00.592845: I tensorflow/compiler/xla/service/service.cc:158]   StreamExecutor device (0): <undefined>, <undefined>
> 2020-04-08 13:42:00.593013: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1433] Found device 0 with properties: 
> name: Tesla K80 major: 3 minor: 7 memoryClockRate(GHz): 0.8235
> pciBusID: 0000:00:04.0
> totalMemory: 11.17GiB freeMemory: 11.11GiB
> 2020-04-08 13:42:00.593051: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1512] Adding visible gpu devices: 0
> 2020-04-08 13:42:00.594138: I tensorflow/core/common_runtime/gpu/gpu_device.cc:984] Device interconnect StreamExecutor with strength 1 edge matrix:
> 2020-04-08 13:42:00.594169: I tensorflow/core/common_runtime/gpu/gpu_device.cc:990]      0 
> 2020-04-08 13:42:00.594187: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1003] 0:   N 
> 2020-04-08 13:42:00.594261: W tensorflow/core/common_runtime/gpu/gpu_bfc_allocator.cc:42] Overriding allow_growth setting because the TF_FORCE_GPU_ALLOW_GROWTH environment variable is set. Original config value was 0.
> 2020-04-08 13:42:00.594309: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1115] Created TensorFlow device (/job:localhost/replica:0/task:0/device:GPU:0 with 10805 MB memory) -> physical GPU (device: 0, name: Tesla K80, pci bus id: 0000:00:04.0, compute capability: 3.7)

And then you end up with a `detect.tflite` file in the output directory as specified: `/content/tensorflow1/models/research/object_detection/TFLite_model/`

tazelisk, 2020-04-08:
Okay thank you for your effort. Maybe I get the concept of this totally wrong.

what I did:
I uploaded my file to my googleDrive (my googleDrive isn't connected to my computer) and I just thought this would be enough.

So do I have to connect myDrive directly to my computer? 

Thank you, have a nice day and stay healthy (:

elektronika-ba, 2020-04-08:
I don't have GDrive on my computer neither, I use only (Chrome) browser.

Just open the Notebook link I provided, and follow all steps as it says. Execute instruction by instruction and read what it asks you (when it asks you).

There is also a section telling you to upload certain files and create folders on your Google Drive account. These files will be used by to Notebook to perform all required tasks.

If you already have your training completed, you should be having a frozen inference graph file (*.pb). You use that file when you run the TOCO command and get the *.tfilte file as the output. In this case, you need to skip the "training" part of the Notebook...

If you have the *.pb file for conversion, you can skip many steps in there...

tazelisk, 2020-04-08:
Okay, I really did the same thing as I did before and now it worked.

Anyways, thank you for providing your notebook to all of us, you are really a great help !!

vasylhiruk, 2020-04-08:
Hi, I am also fighting unsuccessfully for three days during quarantine with the command "bazel build" on Windows 10.
Unfortunately when I am trying to reach the notebook 
[https://colab.research.google.com/drive/1MUL7JZDEFYs3A9-xj39UXfx3uJi4aH-x](url)
I am getting :

`Notebook loading error
There was an error loading this notebook. Ensure that the file is accessible and try again.
'TextDecoder' is not defined
https://drive.google.com/drive/?action=locate&id=1MUL7JZDEFYs3A9-xj39UXfx3uJi4aH-x&authuser=0`

`Details`

`TextDecoder is not defined
ReferenceError: 'TextDecoder' is not defined
   at Anonymous function `(https://colab.research.google.com/v2/external/external_polymer_binary.js?vrz=colab-20200406-085600-RC01_305134096:1521:431)
   at Ga (https://colab.research.google.com/v2/external/external_polymer_binary.js?vrz=colab-20200406-085600-RC01_305134096:17:328)
   at Ea.prototype.next_ (https://colab.research.google.com/v2/external/external_polymer_binary.js?vrz=colab-20200406-085600-RC01_305134096:15:467)
   at next (https://colab.research.google.com/v2/external/external_polymer_binary.js?vrz=colab-20200406-085600-RC01_305134096:18:197)
   at b (https://colab.research.google.com/v2/external/external_polymer_binary.js?vrz=colab-20200406-085600-RC01_305134096:26:259)

Any idea how to solve it ?

elektronika-ba, 2020-04-08:
Maybe try different browser? I just opened it using different google account and on my phone, and it worked.

vasylhiruk, 2020-04-08:
Sorry, I have used not appropriate internet browser Microsoft Edje.
Under Google Chrome it is running, but at the 3rd step 

> CONVERT USING TOCO

> !toco --graph_def_file=./tflite_graph.pb --output_file=./detect.tflite --input_arrays=…

I get an error

> usage: toco [-h] --output_file OUTPUT_FILE

            [--saved_model_dir SAVED_MODEL_DIR | --keras_model_file KERAS_MODEL_FILE]
            [--enable_v1_converter] [--experimental_new_converter]

> toco: error: one of the arguments --saved_model_dir --keras_model_file is required

tazelisk, 2020-04-08:
> Sorry, I have used not appropriate internet browser Microsoft Edje.
> Under Google Chrome it is running, but at the 3rd step
>
>
>
> I get an error
>
>
> ```
>         [--saved_model_dir SAVED_MODEL_DIR | [...]

Had the same issue there and I just startet again from the beginning.
You have to read really carefully and check that you don't miss a little thing.

Good luck (:

moinkhan3012, 2020-06-18:
**ERROR: C:/tensorflow/tensorflow/core/BUILD:2750:1: Executing genrule** //tensorflow/core:version_info_gen failed (Exit 2): bash.exe failed: error executing command
  cd C:/users/moin khan/_bazel_moin/xv6zejqw/execroot/org_tensorflow
  SET PATH=C:\msys64\usr\bin;C:\msys64\bin;C:\Users\Moin Khan\tensorflow-build\Scripts;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Program Files\Java\jdk-11.0.4\bin;C:\Users\Moin Khan\test\Scripts;C:\Users\Moin Khan\tensor\Scripts;E:\chromedriver;C:\Program Files\R\R-3.6.1\bin\x64;E:\setup;C:\msys64\usr\bin;C:\Users\Moin Khan\AppData\Local\Programs\Python\Python37\Scripts\;C:\Users\Moin Khan\AppData\Local\Programs\Python\Python37\;C:\Users\Moin Khan\AppData\Local\Microsoft\WindowsApps;E:\PyCharm Community Edition 2019.1.3\bin;C:\Users\Moin Khan\AppData\Local\GitHubDesktop\bin;E:\chromedriver;C:\Users\Moin Khan\AppData\Local\Android\Sdk\platform-tools;C:\Users\Moin Khan\AppData\Local\Microsoft\WindowsApps;C:\Program Files\Docker Toolbox;E:\setup;C:\msys64;C:\Program Files (x86);
    SET PYTHON_BIN_PATH=C:/Users/Moin Khan/tensorflow-build/Scripts/python.exe
    SET PYTHON_LIB_PATH=C:/Users/Moin Khan/tensorflow-build/lib/site-packages
    SET TF_CONFIGURE_IOS=0
    SET TF_DOWNLOAD_CLANG=0
    SET TF_NEED_CUDA=0
    SET TF_NEED_OPENCL_SYCL=0
    SET TF_NEED_ROCM=0
  C:/msys64/usr/bin/bash.exe -c source external/bazel_tools/tools/genrule/genrule-setup.sh; bazel-out/x64_windows-opt/bin/tensorflow/tools/git/gen_git_source.exe --generate external/local_config_git/gen/spec.json external/local_config_git/gen/head external/local_config_git/gen/branch_ref "bazel-out/x64_windows-opt/genfiles/tensorflow/core/util/version_info.cc" --git_tag_override=${GIT_TAG_OVERRIDE:-}
Execution platform: @bazel_tools//platforms:host_platform
**C:\Users\Moin Khan\AppData\Local\Programs\Python\Python37\python.exe: can't open file 'C:\users\moin': [Errno 2] No such file or directory**
Target //tensorflow/tools/pip_package:build_pip_package failed to build
INFO: Elapsed time: 363.531s, Critical Path: 171.02s
INFO: 144 processes: 144 local.
**FAILED: Build did NOT complete successfully******

Please help ....

aflat00n, 2020-06-23:
@elektronika-ba I used your Google COLAB notebook for just converting my quantized frozen infrence graph to tflite but i am getting this error!

`usage: toco [-h] --output_file OUTPUT_FILE
            [--saved_model_dir SAVED_MODEL_DIR | --keras_model_file KERAS_MODEL_FILE]
            [--enable_v1_converter] [--experimental_new_converter]
toco: error: one of the arguments --saved_model_dir --keras_model_file is required`

Do i have to run the whole training process on google COLAB, this one:
https://colab.research.google.com/drive/1gSd6GpXp1OgJoAFs_kvBbsJVdMNA45os

or can i just use this link of yours to just convert my graph:
https://colab.research.google.com/drive/1MUL7JZDEFYs3A9-xj39UXfx3uJi4aH-x

I just uploaded my tflite_graph.pb file to the root directory of my drive and ran your script but in this 3rd step i am getting this toco error.

Sorry if i am doing something fundamentally wrong here, i am new to all this.
Please Help!

elektronika-ba, 2020-06-23:
The TOCO version must have changed. It is not working anymore with the supplied parameters or my script.
I am currently working on another thing so in case you figure out before me, please post the solution here.

aflat00n, 2020-06-23:
YESS!! It works! Thanks alot!

I was just at the breaking point with trying to build tensorflow from source on windows and having the cl.exe error. I am just glad i just skipped all that. You just saved me! :)

sirvnvu98, 2020-06-27:
> The TOCO version must have changed. It is not working anymore with the supplied parameters or my script.
> 
> Now you have to add the parameter
> `--enable_v1_converter`
> to the end of the **toco** command.
> 
> I just tried, and it works:
> 
> `2020-06-23 10:06:46.111167: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcuda.so.1 2020-06-23 10:06:46.227814: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:981] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero 2020-06-23 10:06:46.228471: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1555] Found device 0 with properties: pciBusID: 0000:00:04.0 name: Tesla T4 computeCapability: 7.5 coreClock: 1.59GHz coreCount: 40 deviceMemorySize: 14.73GiB deviceMemoryBandwidth: 298.08GiB/s 2020-06-23 10:06:46.228775: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudart.so.10.1 2020-06-23 10:06:46.485844: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcublas.so.10 2020-06-23 10:06:46.667184: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcufft.so.10 2020-06-23 10:06:46.694985: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcurand.so.10 2020-06-23 10:06:46.968964: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcusolver.so.10 2020-06-23 10:06:47.014532: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcusparse.so.10 2020-06-23 10:06:47.585462: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudnn.so.7 2020-06-23 10:06:47.585654: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:981] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero 2020-06-23 10:06:47.586353: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:981] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero 2020-06-23 10:06:47.586928: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1697] Adding visible gpu devices: 0 2020-06-23 10:06:47.587406: I tensorflow/core/platform/cpu_feature_guard.cc:142] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX512F 2020-06-23 10:06:47.611675: I tensorflow/core/platform/profile_utils/cpu_utils.cc:94] CPU Frequency: 2000125000 Hz 2020-06-23 10:06:47.611901: I tensorflow/compiler/xla/service/service.cc:168] XLA service 0x55ad173a6bc0 initialized for platform Host (this does not guarantee that XLA will be used). Devices: 2020-06-23 10:06:47.611957: I tensorflow/compiler/xla/service/service.cc:176] StreamExecutor device (0): Host, Default Version 2020-06-23 10:06:47.763890: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:981] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero 2020-06-23 10:06:47.764571: I tensorflow/compiler/xla/service/service.cc:168] XLA service 0x55ad173a6d80 initialized for platform CUDA (this does not guarantee that XLA will be used). Devices: 2020-06-23 10:06:47.764600: I tensorflow/compiler/xla/service/service.cc:176] StreamExecutor device (0): Tesla T4, Compute Capability 7.5 2020-06-23 10:06:47.765860: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:981] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero 2020-06-23 10:06:47.766489: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1555] Found device 0 with properties: pciBusID: 0000:00:04.0 name: Tesla T4 computeCapability: 7.5 coreClock: 1.59GHz coreCount: 40 deviceMemorySize: 14.73GiB deviceMemoryBandwidth: 298.08GiB/s 2020-06-23 10:06:47.766569: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudart.so.10.1 2020-06-23 10:06:47.766593: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcublas.so.10 2020-06-23 10:06:47.766617: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcufft.so.10 2020-06-23 10:06:47.766637: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcurand.so.10 2020-06-23 10:06:47.766658: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcusolver.so.10 2020-06-23 10:06:47.766680: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcusparse.so.10 2020-06-23 10:06:47.766703: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudnn.so.7 2020-06-23 10:06:47.766771: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:981] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero 2020-06-23 10:06:47.767447: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:981] successful NUMA node read from SysFS had negative value (-1), but there must be  [... cut ...]
2020-06-27 08:03:38.466704: I tensorflow/lite/toco/graph_transformations/graph_transformations.cc:39] After Removing unused ops pass 1: 1680 operators, 2835 arrays (0 quantized)
2020-06-27 08:03:38.612698: I tensorflow/lite/toco/graph_transformations/graph_transformations.cc:39] Before general graph transformations: 1680 operators, 2835 arrays (0 quantized)
2020-06-27 08:03:38.701963: F tensorflow/lite/toco/graph_transformations/resolve_constant_slice.cc:59] Check failed: dim_size >= 1 (0 vs. 1)
Aborted (core dumped)

please help me !

elektronika-ba, 2020-06-27:
I can't help much, but the problem is probably in your model file. It has to be quantized SSD model to convert it using the provided parameters.

sirvnvu98, 2020-06-28:
> I can't help much, but the problem is probably in your model file. It has to be quantized SSD model to convert it using the provided parameters.
> Follow Edje's tutorial on creating such model, and then you will be able to convert it to Lite model. Or try twiddling with the paramters for TOCO converter.

I used your Google COLAB notebook to retraining but encounter errors and do not know how to fix

/usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:526: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:527: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:528: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:529: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:530: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:535: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])

WARNING: The TensorFlow contrib module will not be included in TensorFlow 2.0.
For more information, please see:
  * https://github.com/tensorflow/community/blob/master/rfcs/20180907-contrib-sunset.md
  * https://github.com/tensorflow/addons
If you depend on functionality not listed there, please file an issue.

WARNING:tensorflow:From /usr/local/lib/python3.6/dist-packages/tensorflow/python/platform/app.py:125: main (from __main__) is deprecated and will be removed in a future version.
Instructions for updating:
Use object_detection/model_main.py.
Traceback (most recent call last):
  File "train.py", line 184, in <module>
    tf.app.run()
  File "/usr/local/lib/python3.6/dist-packages/tensorflow/python/platform/app.py", line 125, in run
    _sys.exit(main(argv))
  File "/usr/local/lib/python3.6/dist-packages/tensorflow/python/util/deprecation.py", line 324, in new_func
    return func(*args, **kwargs)
  File "train.py", line 93, in main
    FLAGS.pipeline_config_path)
  File "/content/tensorflow1/models/research/object_detection/utils/config_util.py", line 96, in get_configs_from_pipeline_file
    text_format.Merge(proto_str, pipeline_config)
  File "/usr/local/lib/python3.6/dist-packages/google/protobuf/text_format.py", line 693, in Merge
    allow_unknown_field=allow_unknown_field)
  File "/usr/local/lib/python3.6/dist-packages/google/protobuf/text_format.py", line 760, in MergeLines
    return parser.MergeLines(lines, message)
  File "/usr/local/lib/python3.6/dist-packages/google/protobuf/text_format.py", line 785, in MergeLines
    self._ParseOrMerge(lines, message)
  File "/usr/local/lib/python3.6/dist-packages/google/protobuf/text_format.py", line 807, in _ParseOrMerge
    self._MergeField(tokenizer, message)
  File "/usr/local/lib/python3.6/dist-packages/google/protobuf/text_format.py", line 932, in _MergeField
    merger(tokenizer, message, field)
  File "/usr/local/lib/python3.6/dist-packages/google/protobuf/text_format.py", line 1006, in _MergeMessageField
    self._MergeField(tokenizer, sub_message)
  File "/usr/local/lib/python3.6/dist-packages/google/protobuf/text_format.py", line 899, in _MergeField
    (message_descriptor.full_name, name))
google.protobuf.text_format.ParseError: 109:5 : Message type "object_detection.protos.DetectionModel" has no field named "loss".

elektronika-ba, 2020-06-29:
Yea, they changed something on the Colab system again, I am trying to adjust the notebook.

elektronika-ba, 2020-06-29:
I finally managed to get the training working again. I was stuck on compiling the *.proto files, so I used the ones I had on my local PC and copied them to the Colab.

@sirvnvu98 you really need to follow each step carefully. For the proto files, you can find their *.py counterparts here: https://drive.google.com/file/d/1Yr0au-UzKLOD_qQQWCKi7Wq44wvAoUCO/view?usp=sharing

Copy them to the destination /content/tensorflow1/models/research/object_detection/protos/ on your Colab at the step after connecting to Google Drive.

I did it this way:

`%cd /content/tensorflow1/models/research/object_detection/protos/`
`!cp '/content/drive/My Drive/compiled_protos.zip' .`
`!unzip compiled_protos.zip -d ./ # OVERWRITE -> No`

I will update my shared Notebook (when I find the time) for future use.

sirvnvu98, 2020-06-30:
Thanks for the help,
I used your protos file but it still doesn't work

sirvnvu98, 2020-07-03:
Sorry @elektronika-ba , I tried your protos file a lot but still got an error. Can you check it for me again?
I look forward to your help.

WARNING: The TensorFlow contrib module will not be included in TensorFlow 2.0.
For more information, please see:
  * https://github.com/tensorflow/community/blob/master/rfcs/20180907-contrib-sunset.md
  * https://github.com/tensorflow/addons
If you depend on functionality not listed there, please file an issue.

WARNING:tensorflow:From /usr/local/lib/python3.6/dist-packages/tensorflow/python/platform/app.py:125: main (from __main__) is deprecated and will be removed in a future version.
Instructions for updating:
Use object_detection/model_main.py.
Traceback (most recent call last):
  File "train.py", line 184, in <module>
    tf.app.run()
  File "/usr/local/lib/python3.6/dist-packages/tensorflow/python/platform/app.py", line 125, in run
    _sys.exit(main(argv))
  File "/usr/local/lib/python3.6/dist-packages/tensorflow/python/util/deprecation.py", line 324, in new_func
    return func(*args, **kwargs)
  File "train.py", line 93, in main
    FLAGS.pipeline_config_path)
  File "/content/tensorflow1/models/research/object_detection/utils/config_util.py", line 96, in get_configs_from_pipeline_file
    text_format.Merge(proto_str, pipeline_config)
  File "/usr/local/lib/python3.6/dist-packages/google/protobuf/text_format.py", line 693, in Merge
    allow_unknown_field=allow_unknown_field)
  File "/usr/local/lib/python3.6/dist-packages/google/protobuf/text_format.py", line 760, in MergeLines
    return parser.MergeLines(lines, message)
  File "/usr/local/lib/python3.6/dist-packages/google/protobuf/text_format.py", line 785, in MergeLines
    self._ParseOrMerge(lines, message)
  File "/usr/local/lib/python3.6/dist-packages/google/protobuf/text_format.py", line 807, in _ParseOrMerge
    self._MergeField(tokenizer, message)
  File "/usr/local/lib/python3.6/dist-packages/google/protobuf/text_format.py", line 932, in _MergeField
    merger(tokenizer, message, field)
  File "/usr/local/lib/python3.6/dist-packages/google/protobuf/text_format.py", line 1006, in _MergeMessageField
    self._MergeField(tokenizer, sub_message)
  File "/usr/local/lib/python3.6/dist-packages/google/protobuf/text_format.py", line 899, in _MergeField
    (message_descriptor.full_name, name))
google.protobuf.text_format.ParseError: 109:5 : Message type "object_detection.protos.DetectionModel" has no field named "loss".

closed 2022-09-07 by EdjeElectronics