EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #44: Cannot connect to X server Segmentation fault when running

mitchnesteruk, 2020-02-07:


EdjeElectronics, 2020-03-08:
This usually occurs because you're trying to run TFLite detection program over an SSH connection. The program needs a monitor to display results to, so if there is no monitor, the program will crash. Plug a monitor in to your Raspberry Pi and try running it again!

closed 2020-03-08 by EdjeElectronics

Buzzology, 2020-03-17:
Just for anyone not using a monitor, you can use the image_detection script and cv2.imwrite to write the output to a file instead:

**Script to run:**
`python3 TFLite_detection_image.py --modeldir=Sample_TFLite_model --image test1.jpg`

**Section to modify (currently line 178):**
```
#cv2.imshow('Object detector', image)
cv2.imwrite("output`.jpg", image)
```