EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #46: Edge TPU program not working

thepiam, 2020-02-11:
Thanks for the great tutorial. I followed instructions in "Section 1 - How to Set Up and Run TensorFlow Lite Object Detection Models on the Raspberry Pi" and got it working without any problems.

However, in "Section 2 - Run Edge TPU Object Detection Models on the Raspberry Pi Using the Coral USB Accelerator", got the following error when running the demo script:

(tflite1-env) pi@raspberrypi:~/tflite1 $ python3 TFLite_detection_webcam.py --modeldir=Sample_TFLite_model --edgetpu
INFO: Initialized TensorFlow Lite runtime.
/home/pi/tflite1/Sample_TFLite_model/edgetpu.tflite
Traceback (most recent call last):
  File "TFLite_detection_webcam.py", line 140, in <module>
    interpreter.allocate_tensors()
  File "/home/pi/tflite1/tflite1-env/lib/python3.7/site-packages/tensorflow_core/lite/python/interpreter.py", line 244, in allocate_tensors
    return self._interpreter.AllocateTensors()
  File "/home/pi/tflite1/tflite1-env/lib/python3.7/site-packages/tensorflow_core/lite/python/interpreter_wrapper/tensorflow_wrap_interpreter_wrapper.py", line 106, in AllocateTensors
    return _tensorflow_wrap_interpreter_wrapper.InterpreterWrapper_AllocateTensors(self)
RuntimeError: Internal: Unsupported data type in custom op handler: 0Node number 2 (EdgeTpuDelegateForCustomOp) failed to prepare.

Is it possible that this is related to recent update as noted here?
`https://github.com/google-coral/edgetpu/issues/44#issuecomment-579787546`

There appears to be a fix:
`https://github.com/google-coral/edgetpu/issues/44#issuecomment-579905056`
I followed the fix but this did not solve the problem.
Thank you so much again.

**My system info:**
(tflite1-env) pi@raspberrypi:~/tflite1 $ sudo dpkg -l | grep edge
ii  libedgetpu1-std:armhf                13.0                                armhf        Support library for Edge TPU

(tflite1-env) pi@raspberrypi:~/tflite1 $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

MDCR55, 2020-02-13:
Thanks for the amazing tutorial.

I have the same issue when running the command `python3 TFLite_detection_webcam.py --modeldir=Sample_TFLite_model --edgetpu`

then the following error appears:

/home/pi/tflite1/Sample_TFLite_model/edgetpu.tflite
Traceback (most recent call last):
  File "TFLite_detection_webcam.py", line 140, in <module>
    interpreter.allocate_tensors()
  File "/home/pi/tflite1/tflite1-env/lib/python3.7/site-packages/tensorflow_core/lite/python/interpreter.py", line 244, in allocate_tensors
    return self._interpreter.AllocateTensors()
  File "/home/pi/tflite1/tflite1-env/lib/python3.7/site-packages/tensorflow_core/lite/python/interpreter_wrapper/tensorflow_wrap_interpreter_wrapper.py", line 106, in AllocateTensors
    return _tensorflow_wrap_interpreter_wrapper.InterpreterWrapper_AllocateTensors(self)
RuntimeError: Internal: Unsupported data type in custom op handler: 6488064Node number 2 (EdgeTpuDelegateForCustomOp) failed to prepare.

neftaliw, 2020-02-14:
Same issue here, it seems to be a compatibility problem after a recent update to the Coral TPU, has anyone encountered the solution?

neftaliw, 2020-02-14:
UPDATE:
The source of the problem is with the libedgetpu library that updated to version 13 recently and this requires tfensorflow 2.1 which is still not available for raspberry pi (as to why they would release a version of libedgetpu that isn't backwards compatible is beyond me).

Now to make things worse, the people behind libedgetpu's library stupidly only allow you to install the latest version. The way I solved this is by going through a third party repository and downloaded libedgetpu 12.1 from them and installed that instead, and now is working again. Here is the command for you:

`wget https://packages.cloud.google.com/apt/pool/libedgetpu1-std_12-1_armhf_99b783b7a9ec7879e010727c92ee92d96f2e6f16d0999099ea05e61fef23d297.deb
pip3 install libedgetpu1-std_12-1_armhf_99b783b7a9ec7879e010727c92ee92d96f2e6f16d0999099ea05e61fef23d297.deb `

thepiam, 2020-02-15:
@neftaliw Thank you very much for the information!
I was able to get the edgetpu script running now using older version of libedgetpu  from https://coral.ai/software/  link [(https://dl.google.com/coral/edgetpu_api/edgetpu_runtime_20190920.tar.gz)]

(Your script did not work for me for some reason).
Hopefully these incompatibilities will be sorted out soon.
Thanks so much again.

ericlo00, 2020-02-15:
I fixed it after adding 'from tflite_runtime.interpreter import Interpreter' the following code in TFLite_detection_webcam.py .  

This is mentioned in [(https://github.com/EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi/issues/20)](url)

if use_TPU:
    from tflite_runtime.interpreter import Interpreter
    interpreter = Interpreter(model_path=PATH_TO_CKPT,
                              experimental_delegates=[load_delegate('libedgetpu.so.1.0')])
    print(PATH_TO_CKPT)

killinen, 2020-02-17:
These solutions didn't work for me, but I got It working with this with mod from @neftaliw code:

`wget https://packages.cloud.google.com/apt/pool/libedgetpu1-std_12-1_armhf_99b783b7a9ec7879e010727c92ee92d96f2e6f16d0999099ea05e61fef23d297.deb`
`sudo dpkg -i libedgetpu1-std_12-1_armhf_99b783b7a9ec7879e010727c92ee92d96f2e6f16d0999099ea05e61fef23d297.deb`

Then I got the libedgetpu1-max working by these commands (I think it's running on max):

`wget https://packages.cloud.google.com/apt/pool/libedgetpu1-max_12-1_armhf_bc9b8954aadd2c89d579afc1cfdba3bd3842aa88332ae8ce885699a78f41ca26.deb`
`sudo dpkg -r libedgetpu1-std`
`sudo dpkg --purge libedgetpu1-std`
I'm not sure if the above command is necessary.
`sudo dpkg -i libedgetpu1-max_12-1_armhf_bc9b8954aadd2c89d579afc1cfdba3bd3842aa88332ae8ce885699a78f41ca26.deb`

System: RasPi4, 4 GB with Raspbian GNU/Linux 10 (buster) running on latest update.

Hyun5, 2020-02-25:
@killinen Thanks. It works for me.

(running on max)
`wget https://packages.cloud.google.com/apt/pool/libedgetpu1-max_12-1_armhf_bc9b8954aadd2c89d579afc1cfdba3bd3842aa88332ae8ce885699a78f41ca26.deb`

`sudo dpkg -i libedgetpu1-max_12-1_armhf_bc9b8954aadd2c89d579afc1cfdba3bd3842aa88332ae8ce885699a78f41ca26.deb`

System: RasPi4, 4 GB with Raspbian GNU/Linux 10 (buster) running on latest update.

gtudor, 2020-03-02:
@killinen Also thank you. The tips above works for me. @EdjeElectronics thank you for the wonderfully written tutorial. It just needs to be updated to reflect the changes here. Otherwise everyone that tries this now will get the same error above when adding the --edgetpu option.

EdjeElectronics, 2020-03-07:
Thank you everyone for your work on this issue! I will work on testing and implementing the solutions you've discovered.

closed 2020-03-08 by EdjeElectronics

EdjeElectronics, 2020-03-08:
This error can be resolved by using tflite_runtime v2.1 rather than tflite_runtime v2.0. I changed the get_pi_requirements.sh file to automatically download v2.1 rather than v2.0, so people should no longer run in to this error if they're following my guide from scratch.

If you do run in to this error, you can resolve it by uninstalling your existing version of TensorFlow, and installing tflite_runtime v2.1 instead.

If you're running Raspbian Stretch (which is likely if you're using a Raspberry Pi 3), issue:
```
pip3 uninstall tensorflow
pip3 install https://dl.google.com/coral/python/tflite_runtime-2.1.0.post1-cp35-cp35m-linux_armv7l.whl
```

If you're running Raspbian Buster (which is likely if you're using a Raspberry Pi 4), issue:
```
pip3 uninstall tensorflow
pip3 install https://dl.google.com/coral/python/tflite_runtime-2.1.0.post1-cp37-cp37m-linux_armv7l.whl
```