EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #48: Step 3a. Create optimized TensorFlow Lite model

rudikx6, 2020-02-22:
Hi @ all, 

i hope someone maybe can help me... 
I get this error after running my model through the TOCO tool. 

`(tensorflow-build) C:\tensorflow-build>bazel run --config=opt tensorflow/lite/toco:toco -- --input_file=%OUTPUT_DIR%/tflite_graph.pb --output_file=%OUTPUT_DIR%/detect.tflite --input_shapes=1,300,300,3 --input_arrays=normalized_input_image_tensor --output_arrays=TFLite_Detection_PostProcess,TFLite_Detection_PostProcess:1,TFLite_Detection_PostProcess:2,TFLite_Detection_PostProcess:3 --inference_type=QUANTIZED_UINT8 --mean_values=128 --std_values=128 --change_concat_input_ranges=false --allow_custom_ops

WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.google.protobuf.UnsafeUtil (file:/C:/Users/rudik/_bazel_rudik/install/49689753f6e99985995e6295d4436977/_embedded_binaries/A-server.jar) to field java.nio.Buffer.address
WARNING: Please consider reporting this to the maintainers of com.google.protobuf.UnsafeUtil
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
ERROR: The 'run' command is only supported from within a workspace.
WARNING: --batch mode is deprecated. Please instead explicitly shut down your Bazel server using the command "bazel shutdown".
INFO: Invocation ID: 5bfd1709-3dfa-4fcb-9bb5-c10458885e6e`

Greetings, Rudi

rudikx6, 2020-02-23:
Found the problem.... 
in step 3a. need to go one level deeper in the folder

> cd C:\tensorflow-build\tensorflow

closed 2020-02-23 by rudikx6

Petros626, 2021-10-05:
@rudikx6 hey could you help me with this, i get one step deeper in the foldet, but the error seems to be the same for me:
[image]

dont know if you read this, but would be great if you can help me with this

rudikx6, 2021-10-05:
Puh... That's long time ago. Sorry but if it's not the deeper folder I have no Idea.

Petros626, 2021-10-05:
@rudikx6 Maybe I am using the wrong folders. i can show you some folders from where i started the command, would be cool if you can remember :)

1: C:\Users\Petros\models_test in my case tensorflow-build is models_test i think or its the source folder of tensorflow?

2: C:\ProgramData\Anaconda3\envs\object_test\Lib\site-packages\tensorflow_core\lite\toco this was my other option to run it from the toco folder, i read about it in other issues. Or i must run the command one folder before from lite?

rudikx6, 2021-10-09:
Sorry Petros, i have no idea. Finaly i have used another way to create my own TFlite models. 
[https://namburger.medium.com/deploy-your-own-ssdlite-mobiledet-object-detector-on-google-corals-edgetpu-using-tensorflow-s-f41f1e3360c8]

For me it was the best way that time.