EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #50: EdgeTpuDelegateForCustomOp

edgarscr, 2020-03-02:
Hello i have new coral usb accelerator im testing but i go this bug. 

[image]

maybe can you help me whats wrong?

```
2020-03-02 15:46:39.970128: E tensorflow/core/platform/hadoop/hadoop_file_system.cc:132] HadoopFileSystem load error: libhdfs.so: cannot open shared object file: No such file or directory
/home/pi/tflite1/Sample_TFLite_model/edgetpu.tflite
Traceback (most recent call last):
  File "TFLite_detection_image.py", line 119, in <module>
    interpreter.allocate_tensors()
  File "/home/pi/.local/lib/python3.7/site-packages/tensorflow_core/lite/python/interpreter.py", line 245, in allocate_tensors
    return self._interpreter.AllocateTensors()
  File "/home/pi/.local/lib/python3.7/site-packages/tensorflow_core/lite/python/interpreter_wrapper/tensorflow_wrap_interpreter_wrapper.py", line 110, in AllocateTensors
    return _tensorflow_wrap_interpreter_wrapper.InterpreterWrapper_AllocateTensors(self)
RuntimeError: Internal: Unsupported data type in custom op handler: 134350849Node number 2 (EdgeTpuDelegateForCustomOp) failed to prepare.

```

sometimes too

[image]

```
2020-03-02 16:06:25.274480: E tensorflow/core/platform/hadoop/hadoop_file_system.cc:132] HadoopFileSystem load error: libhdfs.so: cannot open shared object file: No such file or directory
Traceback (most recent call last):
  File "parkio-lite.py", line 100, in <module>
    interpreter.allocate_tensors()
  File "/home/pi/.local/lib/python3.7/site-packages/tensorflow_core/lite/python/interpreter.py", line 245, in allocate_tensors
    return self._interpreter.AllocateTensors()
  File "/home/pi/.local/lib/python3.7/site-packages/tensorflow_core/lite/python/interpreter_wrapper/tensorflow_wrap_interpreter_wrapper.py", line 110, in AllocateTensors
    return _tensorflow_wrap_interpreter_wrapper.InterpreterWrapper_AllocateTensors(self)
RuntimeError: Internal: Unsupported data type in custom op handler: 0Node number 2 (EdgeTpuDelegateForCustomOp) failed to prepare.

```

edgarscr, 2020-03-02:
Hey @EdjeElectronics or @Namburger can you help

Namburger, 2020-03-05:
@edgarscr This has definitely been solved. Please take a look at [this](https://github.com/google-coral/edgetpu/issues/44) and upgrade to use tflite_runtime v2 instead.

EdjeElectronics, 2020-03-07:
Hey @Namburger, thank you for your work on getting this issue resolved. I'm going to update my "get_pi_requirements.sh" file to make it download tflite_runtime v2.1. I think I had previously used regular TensorFlow v2.0 because tflite_runtime wasn't available for Python 3.7 when I wrote this guide. I see that it is available for Python 3.7 now!

Future readers: if you're getting this error, you need to update the version of TensorFlow on the Raspberry Pi.

If you're running Raspbian Stretch (which is likely if you're using a Raspberry Pi 3), use:

```
pip3 install https://dl.google.com/coral/python/tflite_runtime-2.1.0.post1-cp35-cp35m-linux_armv7l.whl
```

If you're running Raspbian Buster (which is likely if you're using a Raspberry Pi 4), use:

```
pip3 https://dl.google.com/coral/python/tflite_runtime-2.1.0.post1-cp37-cp37m-linux_armv7l.whl
```

Then, try re-running the detection script.

closed 2020-03-07 by EdjeElectronics