EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #63: ImportError: libgstbase (and other missing dependencies)

devopsleigh, 2020-05-11:
I have followed your guide, but I cannot get the Python scripts to run successfully.

This was a fresh installation of Rasbpian Lite (Buster, 2020-02-13). All steps attempted twice with a re-image between.

Some dependencies were missing from the guide, which I took as assumed knowledge:

`sudo apt install -y git python3 python3-pip python3-venv python3-opencv
pip3 install virtualenv opencv-python`

They only got me so far. I can't seem to get past this one:
`python3 TFLite_detection_image.py --modeldir=Sample_TFLite_model --imagedir=.
Traceback (most recent call last):
  File "TFLite_detection_image.py", line 18, in <module>
    import cv2
  File "/home/pi/tflite1/tflite1-env/lib/python3.7/site-packages/cv2/__init__.py", line 3, in <module>
    from .cv2 import *
ImportError: libgstbase-1.0.so.0: cannot open shared object file: No such file or directory`

I don't know much about Python and nobody else has cited these issues, so I suppose I must be missing something.

prkapur, 2020-10-03:
@devopsleigh I ran into the same issue and for anyone else running into this again, this should resolve it.

https://stackoverflow.com/questions/52378554/problems-importing-open-cv-in-python

pip3 install opencv-python
sudo apt install libqtgui4
sudo apt install libqt4-test

Wifialarm, 2021-09-30:
Hi,
I successfully ran project on Rasp on first try.  All worked
But after restart it gives 
import cv2 error

followed https://github.com/EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi/issues/63#issuecomment-703019851
but appears all sets are available

If run pip3 show opencv-python
Gives
Version: 3.4.6.27

Is there any idea?
Thanks