EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #64: GPIO issue when using tflite detection

talhakhaira, 2020-05-14:
I have successfully trained my custom model and used it for object detection. However, I am facing a slight issue regarding the if-else condition on my custom model. I only have one object/class in my trained model, **when it is detected I want to make a gpio pin to go HIGH and when the object is removed from the webcam feed the pin should go LOW.** When I run the code the pin does go HIGH on a perfect detection but stays HIGH even if I remove the object from webcam feed. I used the following if-else condition:

if (object_name == labels[int(classes[i])]) and (scores[i]) >= 0.95):
GPIO.output(13, GPIO.HIGH)
else:
GPIO.output(13, GPIO.LOW)

Stefan-1998, 2020-06-11:
Hello, i have a similar Problem.  I can't access the GPIO's at all.
It looks like you can access them. 

Are you using :import RPi.GPIO as GPIO and the tflite tutorial from Edje?

If i complile the code with import RPI.GPIO it shows a  error.

dnomyaR23, 2022-08-15:
@talhakhaira Hello, would you like to share the source code to me? Thank you