EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #70: No difference in speed when switch to edge-TPU

zlg9folira, 2020-06-22:
Hi, I am using Pi 4 with a Full HD 30fps webcam. I followed exactly the same instruction given in the repo and everything was done nicely. 

**Without Coral USB accelerator:**
The FPS stamp shows around 4.5 and the actual frame rate looks 4.5.

**With Coral USB accelerator:**
The FPS stamp shows around 24, but the actual frame rate looks the same as previous case 4.5.

Can you help me on how to fix the case with Coral USB accelerator?

contractorwolf, 2020-07-09:
If you are using your own custom model I think I know what the issue is.  I did the same thing on a custom trained model and was only seeing like 3-4FPS.  The reason is that your custom model needs to be quanitized so that it can run in the TPU.  If you follow his video and try using the edge.tflite file you will notice your TPU lights up when it is going and you will get like 20+FPS but when you run your custom tflite file that has not been converted to the edge tflite the TPU will not light up when it is going (just the small steady white, not blinking).  I am trying to figure out how to quantize my model right now too :/

elektronika-ba, 2020-07-09:
When you get to the point where you need to compile your TFLITE model for EDGE TPU, you can use my Colab to perform this task:

https://colab.research.google.com/drive/1q4sS2ZgZfsxeXdoU0cU6gC699bOZkiZf?usp=sharing

I did it couple of days ago when I got my Coral AI accelerator. Works great!

thongvhoang, 2021-06-14:
> When you get to the point where you need to compile your TFLITE model for EDGE TPU, you can use my Colab to perform this task:
> 
> https://colab.research.google.com/drive/1q4sS2ZgZfsxeXdoU0cU6gC699bOZkiZf?usp=sharing
> 
> I did it couple of days ago when I got my Coral AI accelerator. Works great!

Hi Trax @elektronika-ba 

I ask which devices do EdgeTPU deploy on (Android, usb accelerator, something else) ?

Thank you.

Hecto-r, 2023-11-23:
For anyone that runs into this. Either pass the --resolution or change the python file to have a default resolution of 640x480. This makes the camera video look better. I don't think its 100% matching the FPS being displayed though.

Using SSD MobileNet V1 model coral I was able to hit 35 fps.