EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #72: connecting to HikVision cameras

phairplay, 2020-06-28:
Hi,
I've install Tensorflow lite using you very well detailed guide, thank you for that.

it is possible to connect to them my link i use within home assistant is
"rtsp://admin:password@IP:PORT/Streaming/Channels/101"

I would really appreciate  your help

elektronika-ba, 2020-06-29:
I managed to connect to my cameras using this: https://stackoverflow.com/a/52428164

phairplay, 2020-06-29:
Hi, 
thank you for the reply.

can I please confirm.

you used the python-vlc to capture an image second and used the TFLite_detection_image.py?

elektronika-ba, 2020-06-29:
I used OpenCV to access my IP cameras via their IP addresses, and I passed the image to the TF inferencing as per Edje's tutorial.