EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #73: FAILED: Build did NOT complete successfully

contractorwolf, 2020-07-04:
I am following the instructions laid out in this tutorial (starting at Step 2) to convert my model for use by tflite:
https://github.com/EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi#part-1---how-to-train-convert-and-run-custom-tensorflow-lite-object-detection-models-on-windows-10

My model works correctly on my windows machine but when attempting to convert it to a tfliet model I cannon get tensorflow to build, errors every time.  I am following the directions very carefully and even restarted the tensorflow-build process from scratch again just to be sure

**System information**
- OS Platform and Distribution (e.g., Linux Ubuntu 16.04): Windows 10
- TensorFlow installed from (source or binary): Source
- TensorFlow version: 1.13
- Python version: 3.6
- Installed using conda?: Conda
- Bazel version (if compiling from source): 0.21.0
- CUDA/cuDNN version: just building for CPU
- GPU model and memory: GTX 980

everything seems to work find until I try and build using babel with this command:
> bazel build --config=opt //tensorflow/tools/pip_package:build_pip_package

It start off fine and runs for a while but then fails with this output:

INFO: From Linking tensorflow/python/gen_lookup_ops_py_wrappers_cc.exe:
   Creating library bazel-out/x64_windows-opt/bin/tensorflow/python/gen_lookup_ops_py_wrappers_cc.lib and object bazel-out/x64_windows-opt/bin/tensorflow/python/gen_lookup_ops_py_wrappers_cc.exp
INFO: From Linking tensorflow/python/gen_script_ops_py_wrappers_cc.exe:
   Creating library bazel-out/x64_windows-opt/bin/tensorflow/python/gen_script_ops_py_wrappers_cc.lib and object bazel-out/x64_windows-opt/bin/tensorflow/python/gen_script_ops_py_wrappers_cc.exp
INFO: From Linking tensorflow/python/gen_bitwise_ops_py_wrappers_cc.exe:
   Creating library bazel-out/x64_windows-opt/bin/tensorflow/python/gen_bitwise_ops_py_wrappers_cc.lib and object bazel-out/x64_windows-opt/bin/tensorflow/python/gen_bitwise_ops_py_wrappers_cc.exp
INFO: From Linking tensorflow/python/gen_ragged_conversion_ops_py_wrappers_cc.exe:
   Creating library bazel-out/x64_windows-opt/bin/tensorflow/python/gen_ragged_conversion_ops_py_wrappers_cc.lib and object bazel-out/x64_windows-opt/bin/tensorflow/python/gen_ragged_conversion_ops_py_wrappers_cc.exp
INFO: From Linking tensorflow/python/gen_ragged_math_ops_py_wrappers_cc.exe:
   Creating library bazel-out/x64_windows-opt/bin/tensorflow/python/gen_ragged_math_ops_py_wrappers_cc.lib and object bazel-out/x64_windows-opt/bin/tensorflow/python/gen_ragged_math_ops_py_wrappers_cc.exp
**ERROR: C:/tensorflow-build/tensorflow/tensorflow/python/BUILD:293:1: C++ compilation of rule '//tensorflow/python:bfloat16_lib' failed (Exit 2): cl.exe failed: error executing command**
  cd C:/users/james/_bazel_james/j7bi4x5j/execroot/org_tensorflow
  SET INCLUDE=C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE;C:\Program Files (x86)\Windows Kits\10\include\10.0.10240.0\ucrt;C:\Program Files (x86)\Windows Kits\NETFXSDK\4.6\include\um;C:\Program Files (x86)\Windows Kits\8.1\include\shared;C:\Program Files (x86)\Windows Kits\8.1\include\um;C:\Program Files (x86)\Windows Kits\8.1\include\winrt;
    SET PATH=C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\amd64;C:\WINDOWS\Microsoft.NET\Framework64\v4.0.30319;C:\WINDOWS\Microsoft.NET\Framework64\;C:\Program Files (x86)\Windows Kits\8.1\bin\x64;C:\Program Files (x86)\Windows Kits\8.1\bin\x86;C:\Program Files (x86)\Microsoft SDKs\Windows\v10.0A\bin\NETFX 4.6 Tools\x64\;;C:\WINDOWS\system32
    SET PWD=/proc/self/cwd
    SET PYTHON_BIN_PATH=C:/ProgramData/Anaconda3/envs/tensorflow-build/python.exe
    SET PYTHON_LIB_PATH=C:/ProgramData/Anaconda3/envs/tensorflow-build/lib/site-packages
    SET TEMP=C:\Users\james\AppData\Local\Temp
    SET TF_DOWNLOAD_CLANG=0
    SET TF_NEED_CUDA=0
    SET TF_NEED_OPENCL_SYCL=0
    SET TF_NEED_ROCM=0
    SET TMP=C:\Users\james\AppData\Local\Temp
  C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/amd64/cl.exe /nologo /DCOMPILER_MSVC /DNOMINMAX /D_WIN32_WINNT=0x0601 /D_CRT_SECURE_NO_DEPRECATE /D_CRT_SECURE_NO_WARNINGS /bigobj /Zm500 /EHsc /wd4351 /wd4291 /wd4250 /wd4996 /I. /Ibazel-out/x64_windows-opt/genfiles /Ibazel-out/x64_windows-opt/bin /Iexternal/com_google_absl /Ibazel-out/x64_windows-opt/genfiles/external/com_google_absl /Ibazel-out/x64_windows-opt/bin/external/com_google_absl /Iexternal/eigen_archive /Ibazel-out/x64_windows-opt/genfiles/external/eigen_archive /Ibazel-out/x64_windows-opt/bin/external/eigen_archive /Iexternal/local_config_sycl /Ibazel-out/x64_windows-opt/genfiles/external/local_config_sycl /Ibazel-out/x64_windows-opt/bin/external/local_config_sycl /Iexternal/nsync /Ibazel-out/x64_windows-opt/genfiles/external/nsync /Ibazel-out/x64_windows-opt/bin/external/nsync /Iexternal/gif_archive /Ibazel-out/x64_windows-opt/genfiles/external/gif_archive /Ibazel-out/x64_windows-opt/bin/external/gif_archive /Iexternal/jpeg /Ibazel-out/x64_windows-opt/genfiles/external/jpeg /Ibazel-out/x64_windows-opt/bin/external/jpeg /Iexternal/protobuf_archive /Ibazel-out/x64_windows-opt/genfiles/external/protobuf_archive /Ibazel-out/x64_windows-opt/bin/external/protobuf_archive /Iexternal/com_googlesource_code_re2 /Ibazel-out/x64_windows-opt/genfiles/external/com_googlesource_code_re2 /Ibazel-out/x64_windows-opt/bin/external/com_googlesource_code_re2 /Iexternal/farmhash_archive /Ibazel-out/x64_windows-opt/genfiles/external/farmhash_archive /Ibazel-out/x64_windows-opt/bin/external/farmhash_archive /Iexternal/fft2d /Ibazel-out/x64_windows-opt/genfiles/external/fft2d /Ibazel-out/x64_windows-opt/bin/external/fft2d /Iexternal/highwayhash /Ibazel-out/x64_windows-opt/genfiles/external/highwayhash /Ibazel-out/x64_windows-opt/bin/external/highwayhash /Iexternal/zlib_archive /Ibazel-out/x64_windows-opt/genfiles/external/zlib_archive /Ibazel-out/x64_windows-opt/bin/external/zlib_archive /Iexternal/double_conversion /Ibazel-out/x64_windows-opt/genfiles/external/double_conversion /Ibazel-out/x64_windows-opt/bin/external/double_conversion /Iexternal/snappy /Ibazel-out/x64_windows-opt/genfiles/external/snappy /Ibazel-out/x64_windows-opt/bin/external/snappy /Iexternal/local_config_python /Ibazel-out/x64_windows-opt/genfiles/external/local_config_python /Ibazel-out/x64_windows-opt/bin/external/local_config_python /Iexternal/local_config_cuda /Ibazel-out/x64_windows-opt/genfiles/external/local_config_cuda /Ibazel-out/x64_windows-opt/bin/external/local_config_cuda /Iexternal/lmdb /Ibazel-out/x64_windows-opt/genfiles/external/lmdb /Ibazel-out/x64_windows-opt/bin/external/lmdb /Iexternal/org_sqlite /Ibazel-out/x64_windows-opt/genfiles/external/org_sqlite /Ibazel-out/x64_windows-opt/bin/external/org_sqlite /Iexternal/png_archive /Ibazel-out/x64_windows-opt/genfiles/external/png_archive /Ibazel-out/x64_windows-opt/bin/external/png_archive /Iexternal/icu /Ibazel-out/x64_windows-opt/genfiles/external/icu /Ibazel-out/x64_windows-opt/bin/external/icu /Iexternal/grpc /Ibazel-out/x64_windows-opt/genfiles/external/grpc /Ibazel-out/x64_windows-opt/bin/external/grpc /Iexternal/com_github_nanopb_nanopb /Ibazel-out/x64_windows-opt/genfiles/external/com_github_nanopb_nanopb /Ibazel-out/x64_windows-opt/bin/external/com_github_nanopb_nanopb /Iexternal/boringssl /Ibazel-out/x64_windows-opt/genfiles/external/boringssl /Ibazel-out/x64_windows-opt/bin/external/boringssl /Iexternal/eigen_archive /Ibazel-out/x64_windows-opt/genfiles/external/eigen_archive /Ibazel-out/x64_windows-opt/bin/external/eigen_archive /Iexternal/nsync/public /Ibazel-out/x64_windows-opt/genfiles/external/nsync/public /Ibazel-out/x64_windows-opt/bin/external/nsync/public /Iexternal/gif_archive/lib /Ibazel-out/x64_windows-opt/genfiles/external/gif_archive/lib /Ibazel-out/x64_windows-opt/bin/external/gif_archive/lib /Iexternal/gif_archive/windows /Ibazel-out/x64_windows-opt/genfiles/external/gif_archive/windows /Ibazel-out/x64_windows-opt/bin/external/gif_archive/windows /Iexternal/protobuf_archive/src /Ibazel-out/x64_windows-opt/genfiles/external/protobuf_archive/src /Ibazel-out/x64_windows-opt/bin/external/protobuf_archive/src /Iexternal/farmhash_archive/src /Ibazel-out/x64_windows-opt/genfiles/external/farmhash_archive/src /Ibazel-out/x64_windows-opt/bin/external/farmhash_archive/src /Iexternal/zlib_archive /Ibazel-out/x64_windows-opt/genfiles/external/zlib_archive /Ibazel-out/x64_windows-opt/bin/external/zlib_archive /Iexternal/double_conversion /Ibazel-out/x64_windows-opt/genfi [... cut ...]
tensorflow/python/lib/core/bfloat16.cc(653): error C2664: 'bool tensorflow::`anonymous-namespace'::Initialize::<lambda_d0df84676ae54709cf34c880a157d294>::operator ()(const char *,PyUFuncGenericFunction,const std::array<int,3> &) const': cannot convert argument 2 from 'void (__cdecl *)(char **,npy_intp *,npy_intp *,void *)' to 'PyUFuncGenericFunction'
tensorflow/python/lib/core/bfloat16.cc(653): note: None of the functions with this name in scope match the target type
Target //tensorflow/tools/pip_package:build_pip_package failed to build
INFO: Elapsed time: 107.347s, Critical Path: 59.11s
INFO: 206 processes: 206 local.
**FAILED: Build did NOT complete successfully**

JincoElectronics, 2020-07-08:
I have the same problem
when running the code:
bazel build --config=opt //tensorflow/tools/pip_package:build_pip_package
after fetching for a while i receive error that sayes **"FAILED: Build did NOT complete successfully "**
Did you found any solution?

contractorwolf, 2020-07-08:
@JincoElectronics well sort of, I decided to try and do the whole conversion process in Colab instead.  I just made it public if you wanted to try it and let me know if the conversion works for you? you might need to make a copy of my Colab to run it yourself but it is here:
https://colab.research.google.com/drive/1Px7I6PxeeLhCepyA9Dv22pwz66NuT1JR?usp=sharing

I was able to use that to convert the .pb file from the previous steps into a proper .tfilte that I was able to use on my raspberry pi.  It was slow (3fps) but works fine, but I am working on getting it converted to the EdgeTPU version of the file to use with my Coral TPU, so more to come.  Let me know if that works.

JincoElectronics, 2020-07-09:
thanks for sharing your code
i run it with my pb file i got the following error
any idea how to solve it?

Ignore 'tcmalloc: large alloc' warnings.
---------------------------------------------------------------------------
DecodeError                               Traceback (most recent call last)
/usr/local/lib/python3.6/dist-packages/tensorflow/lite/python/lite.py in from_frozen_graph(cls, graph_def_file, input_arrays, output_arrays, input_shapes)
    253           graph_def = _graph_pb2.GraphDef()
--> 254           graph_def.ParseFromString(file_content)
    255         except (_text_format.ParseError, DecodeError):

DecodeError: Error parsing message

During handling of the above exception, another exception occurred:

UnicodeDecodeError                        Traceback (most recent call last)
1 frames
/usr/local/lib/python3.6/dist-packages/tensorflow/lite/python/lite.py in from_frozen_graph(cls, graph_def_file, input_arrays, output_arrays, input_shapes)
    259             if not isinstance(file_content, str):
    260               if PY3:
--> 261                 file_content = file_content.decode("utf-8")
    262               else:
    263                 file_content = file_content.encode("utf-8")

UnicodeDecodeError: 'utf-8' codec can't decode byte 0x80 in position 41: invalid start byte

contractorwolf, 2020-07-09:
Not sure, what version of Tensorflow did you train it under?

JincoElectronics, 2020-07-09:
i have used tensorflow 1.8 to train on ssd_mobilenet_v2_coco.config

> Not sure, what version of Tensorflow did you train it under?

elektronika-ba, 2020-07-09:
I too have my own Colab version of the converter to TFLite, you can give it a go also:

https://colab.research.google.com/drive/1cUUfhw52A-JjjXGCHDO0eBvgN2xLgtho?usp=sharing

contractorwolf, 2020-07-09:
@elektronika-ba thanks for that but I am getting an error on your Google Colab.  It looks like it is needing some min and max values?

Here is the error I am seeing:

```
2020-07-09 13:06:52.791335: I tensorflow/lite/toco/import_tensorflow.cc:659] Converting unsupported operation: TFLite_Detection_PostProcess
2020-07-09 13:06:52.807922: I tensorflow/lite/toco/graph_transformations/graph_transformations.cc:39] Before Removing unused ops: 500 operators, 754 arrays (0 quantized)
2020-07-09 13:06:52.822293: I tensorflow/lite/toco/graph_transformations/graph_transformations.cc:39] Before general graph transformations: 500 operators, 754 arrays (0 quantized)
2020-07-09 13:06:52.863482: I tensorflow/lite/toco/graph_transformations/graph_transformations.cc:39] After general graph transformations pass 1: 65 operators, 177 arrays (1 quantized)
2020-07-09 13:06:52.864946: I tensorflow/lite/toco/graph_transformations/graph_transformations.cc:39] Before pre-quantization graph transformations: 65 operators, 177 arrays (1 quantized)
2020-07-09 13:06:52.865704: I tensorflow/lite/toco/graph_transformations/graph_transformations.cc:39] Before Group bidirectional sequence lstm/rnn: 65 operators, 177 arrays (1 quantized)
2020-07-09 13:06:52.866624: F tensorflow/lite/toco/tooling_util.cc:1728] Array FeatureExtractor/MobilenetV1/MobilenetV1/Conv2d_0/Relu6, which is an input to the DepthwiseConv operator producing the output array FeatureExtractor/MobilenetV1/MobilenetV1/Conv2d_1_depthwise/Relu6, is lacking min/max data, which is necessary for quantization. If accuracy matters, either target a non-quantized output format, or run quantized training with your model from a floating point checkpoint to change the input graph to contain min/max information. If you don't care about accuracy, you can pass --default_ranges_min= and --default_ranges_max= for easy experimentation.
Aborted (core dumped)
```

JincoElectronics, 2020-07-09:
> I too have my own Colab version of the converter to TFLite, you can give it a go also:
> 
> https://colab.research.google.com/drive/1cUUfhw52A-JjjXGCHDO0eBvgN2xLgtho?usp=sharing

thanks for your help but a got this error running it on my own pb file
i have a question about pb file
Are pb file for different configs same or i should make any changes on it?
```
2020-07-09 13:41:22.538717: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcuda.so.1
2020-07-09 13:41:22.557342: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:981] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-07-09 13:41:22.558100: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1555] Found device 0 with properties: 
pciBusID: 0000:00:04.0 name: Tesla K80 computeCapability: 3.7
coreClock: 0.8235GHz coreCount: 13 deviceMemorySize: 11.17GiB deviceMemoryBandwidth: 223.96GiB/s
2020-07-09 13:41:22.558443: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudart.so.10.1
2020-07-09 13:41:22.560290: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcublas.so.10
2020-07-09 13:41:22.562177: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcufft.so.10
2020-07-09 13:41:22.562515: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcurand.so.10
2020-07-09 13:41:22.564402: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcusolver.so.10
2020-07-09 13:41:22.565760: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcusparse.so.10
2020-07-09 13:41:22.570149: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudnn.so.7
2020-07-09 13:41:22.570281: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:981] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-07-09 13:41:22.571116: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:981] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-07-09 13:41:22.571824: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1697] Adding visible gpu devices: 0
2020-07-09 13:41:22.577440: I tensorflow/core/platform/profile_utils/cpu_utils.cc:94] CPU Frequency: 2300000000 Hz
2020-07-09 13:41:22.577730: I tensorflow/compiler/xla/service/service.cc:168] XLA service 0x55a14983ebc0 initialized for platform Host (this does not guarantee that XLA will be used). Devices:
2020-07-09 13:41:22.577767: I tensorflow/compiler/xla/service/service.cc:176]   StreamExecutor device (0): Host, Default Version
2020-07-09 13:41:22.633648: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:981] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-07-09 13:41:22.634507: I tensorflow/compiler/xla/service/service.cc:168] XLA service 0x55a14983ed80 initialized for platform CUDA (this does not guarantee that XLA will be used). Devices:
2020-07-09 13:41:22.634541: I tensorflow/compiler/xla/service/service.cc:176]   StreamExecutor device (0): Tesla K80, Compute Capability 3.7
2020-07-09 13:41:22.634804: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:981] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-07-09 13:41:22.635592: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1555] Found device 0 with properties: 
pciBusID: 0000:00:04.0 name: Tesla K80 computeCapability: 3.7
coreClock: 0.8235GHz coreCount: 13 deviceMemorySize: 11.17GiB deviceMemoryBandwidth: 223.96GiB/s
2020-07-09 13:41:22.635666: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudart.so.10.1
2020-07-09 13:41:22.635702: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcublas.so.10
2020-07-09 13:41:22.635741: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcufft.so.10
2020-07-09 13:41:22.635809: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcurand.so.10
2020-07-09 13:41:22.635849: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcusolver.so.10
2020-07-09 13:41:22.635880: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcusparse.so.10
2020-07-09 13:41:22.635913: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudnn.so.7
2020-07-09 13:41:22.636020: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:981] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node zero
2020-07-09 13:41:22.636788: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:981] successful NUMA node read from SysFS had negative value (-1), but there must be at least one NUMA node, so returning NUMA node [... cut ...]
  File "/usr/local/lib/python2.7/dist-packages/tensorflow_core/lite/python/tflite_convert.py", line 130, in _convert_tf1_model
    converter = _get_toco_converter(flags)
  File "/usr/local/lib/python2.7/dist-packages/tensorflow_core/lite/python/tflite_convert.py", line 117, in _get_toco_converter
    return converter_fn(**converter_kwargs)
  File "/usr/local/lib/python2.7/dist-packages/tensorflow_core/lite/python/lite.py", line 711, in from_frozen_graph
    "Unable to parse input file '{}'.".format(graph_def_file))
IOError: Unable to parse input file './tflite_graph.pb'.
```

contractorwolf, 2020-07-09:
I have a feeling the reason this is not working is because I trained using the non-quanitized original MobileNet model?

elektronika-ba, 2020-07-09:
I know for fact that it works for quantized model. If your model is not quantized - there is your problem.

contractorwolf, 2020-07-09:
yeah I got it, thanks @elektronika-ba, going to try and train it using a quantized MobileNet model now

JincoElectronics, 2020-07-11:
> yeah I got it, thanks @elektronika-ba, going to try and train it using a quantized MobileNet model now

Have you solved the problem?

contractorwolf, 2020-08-21:
I used some other people examples for doing it in Colab and then made my own that is pretty thoroughly documented.  Mine is for custom training a model to detect masks but you could easily modify it to train on anything.  The full thing is here:
https://colab.research.google.com/drive/1uEkP5j7KM9eSkCUtUyauy7-Dyd5XAY6e?usp=sharing

Here is the Mask Detector model I made running on a Raspberry Pi (model trained with my Colab):
https://www.youtube.com/watch?v=34NSZA2Ne5A

EdjeElectronics, 2020-10-21:
Hi @contractorwolf and @elektronika-ba , thanks so much for sharing your Colabs! It's good to see that TFLite conversion can work in a Colab session using either TOCO or the TFLiteConverter module. Building TensorFlow from source just to convert a TFLite model is a pain, and even I can't get it to build correctly every time. Colab is a much better option!

I will likely replace Steps 2 and 3 of my main guide (Building TensorFlow from Source and using TOCO locally to convert the model) with a link to a Colab. I will be sure to cite both your efforts and give you credit for pointing me in this direction.

closed 2020-10-21 by EdjeElectronics

contractorwolf, 2020-10-21:
Thanks @EdjeElectronics, your great work and tutorials helped me figure this all out. Feel free to pull any of the data and examples I wrote that include the code I am running on my RPi:
https://github.com/contractorwolf/rpi-tensorflow-mask-detector

or my video of running the my Covid Mask Detector model on the Raspberry Pi (using the code above):
https://www.youtube.com/watch?v=34NSZA2Ne5A