EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #76: I have a webcam running on the Pi - Is there a way to use that stream as the source for TF

rcpilotp51, 2020-08-03:
I was wondering if you could direct me to using your tutorial with an existing video stream.
I have a webcam running on the Pi - Is there a way to use that stream as the source for TF?

Thanks for your time!

rcpilotp51, 2020-08-03:
i get the following error providing a stream and a model to the stream.py file:

Traceback (most recent call last):
  File "TFLite_detection_stream.py", line 174, in <module>
    frame = frame1.copy()
AttributeError: 'NoneType' object has no attribute 'copy'

but if i go to the stream url - it resolves fine.

thanks

Dasinator21, 2020-08-05:
> i get the following error providing a stream and a model to the stream.py file:
> 
> Traceback (most recent call last):
> File "TFLite_detection_stream.py", line 174, in
> frame = frame1.copy()
> AttributeError: 'NoneType' object has no attribute 'copy'
> 
> but if i go to the stream url - it resolves fine.
> 
> thanks

How did you manage to get a stream running???
I am struggling. It is saying no port found in URI. What did you do to get it working. What port and ipaddress did you use? Please help me.

tCode5, 2024-06-16:
Modifying the webcam.py script at line 32 could work:
Change
```python3
self.stream = cv2.VideoCapture()
```
into
```python3
self.stream = cv2.VideoCapture("/dev/video0")
```
or
```python3
self.stream = cv2.VideoCapture("v4l2:///dev/video0")
```
or something like that.
Hope that helps!