EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #77: Cannot run the TFLite_detection_stream.py code.

Dasinator21, 2020-08-05:
python TFLite_detection_stream.py --modeldir=TFLite_model --streamurl="http://ipaddress:port/stream/video.mjpeg" 

the output I get:
[tcp @ 0xf7ad40] Port missing in uri
Traceback (most recent call last):
  File "TFLite_detection_stream.py", line 174, in <module>
    frame = frame1.copy()
AttributeError: 'NoneType' object has no attribute 'copy'

How can I run this code!!!

uyesad, 2020-10-16:
did you solve the problem

nozma-knows, 2020-10-28:
I'm having the same issue

basnugroho, 2021-11-07:
i'm having the same issue

busqui04, 2023-11-22:
same issue here. Did you find how to fix it?

EdjeElectronics, 2023-11-22:
Please read the [FAQ](https://github.com/EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi/blob/master/deploy_guides/Raspberry_Pi_Guide.md#1-typeerror-int-argument-must-be-a-string-a-bytes-like-object-or-a-number-not-nonetype). It's most likely a problem with your camera.

closed 2023-11-22 by EdjeElectronics