EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #81: Converted models Predict Nothing

hafizgta, 2020-10-22:
hallo, i'm basically doing whatever this repo instructed but with my own training data.
i was using:
91 test images
356 training images

after following every bits of tutorial, it didnt predict anything.
then i thought maybe the tensorflow version (v1.3) is too out of date, i update it to (v1.5.4) both for training and converting
still the same problem.
i've tried using tflite_convert instead of TOCO, based on [this answer on stackoverflow](https://stackoverflow.com/questions/61749548/how-to-convert-tflite-graph-pb-to-detect-tflite-properly/61851433#61851433)
still the same

after some thinkering i was quite curious on the output, its a complete mess
it was like this:
boxes
```[[ 1.78248568e+28  1.50925612e+16  4.07102690e+15 -2.71258968e-33]
 [-1.11853247e-32  1.96768973e+31 -9.03377323e-39  1.24939601e+33]
 [ 2.89407555e+26  1.30062990e+27  1.86140106e+34  2.09478165e+02]
 [ 7.71351299e+31  1.25456728e+33  5.64565312e+24  6.95112678e+22]
 [ 2.77759736e+29  6.34886799e+10  2.25685488e+31  1.06700113e+21]
 [ 7.14370080e+31  1.12233524e+33  7.67196732e+01 -1.88591859e-16]
 [ 4.42466910e+30  4.10378019e+21  4.11511221e+18 -4.26765503e-38]
 [ 4.78335563e+36  3.13694584e+29 -6.88691560e-37  8.12594240e+34]
 [ 2.94257426e+26  5.41499732e+33  4.80534778e+36  8.10763857e-04]
 [ 1.17784752e+24  5.30551264e+36  5.72120457e+30  4.66273108e+30]]
```
classes
```[ 2.8697603e+26  4.2202749e+24  1.6502402e+25 -4.2376135e-35
  1.2184775e+33  5.5829342e+36  7.6236652e+28  4.9925871e+36
  4.9194568e+30 -1.8692555e-34]
```
scores
```[ 5.8955036e+16  1.5896159e+19  4.5978472e+24  1.5493227e+07
  2.9263451e+29  4.1156166e+18  3.7936189e+17 -8.8601589e-16
  2.0792617e+28  2.1410814e+11]
```

i wonder what's the problem?

dhruvsheth-ai, 2020-10-23:
Can you show the command you used to predict this output. I get the same output when I save the output predicting boxes to a txt file

hafizgta, 2020-10-23:
> Can you show the command you used to predict this output. I get the same output when I save the output predicting boxes to a txt file
i was using the same code that this guide instructed:
`python TFLite_detection_image.py --modeldir=TFLite_model --image=squirrel.jpg
`