EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #83: Unsupported data type in custom op handler: 65109260Node number 2 (EdgeTpuDelegateForCustomOp) failed to prepare.*

Moehrenbaum, 2020-11-14:
Hi there. First off, thanks for the great example. I'm new to the device and followed your description to the bit. I'm getting this error. any thoughts

❯ python3 TFLite_detection_webcam.py --modeldir=Sample_TFLite_model --edgetpu
/home/pi/tflite1/Sample_TFLite_model/edgetpu.tflite
Traceback (most recent call last):
  File "TFLite_detection_webcam.py", line 140, in <module>
    interpreter.allocate_tensors()
  File "/home/pi/tflite1/tflite1-env/lib/python3.7/site-packages/tflite_runtime/interpreter.py", line 242, in allocate_tensors
    return self._interpreter.AllocateTensors()
  File "/home/pi/tflite1/tflite1-env/lib/python3.7/site-packages/tflite_runtime/interpreter_wrapper.py", line 115, in AllocateTensors
    return _interpreter_wrapper.InterpreterWrapper_AllocateTensors(self)
**RuntimeError: Internal: Unsupported data type in custom op handler: 65109260Node number 2 (EdgeTpuDelegateForCustomOp) failed to prepare.**

Moehrenbaum, 2020-11-14:
fixed it by re-installing the  TensorFlow Lite lib again. Maybe it helps somebody.

pip3 install https://github.com/google-coral/pycoral/releases/download/release-frogfish/tflite_runtime-2.5.0-cp37-cp37m-linux_armv7l.whl

closed 2020-11-14 by Moehrenbaum

EdjeElectronics, 2020-11-17:
@Moehrenbaum thank you for sharing your solution! I'm glad you were able to figure it out. I have already pointed a couple people at your solution, so thanks again for helping out.

neilyoung, 2020-11-20:
> fixed it by re-installing the TensorFlow Lite lib again. Maybe it helps somebody.
> 
> pip3 install https://github.com/google-coral/pycoral/releases/download/release-frogfish/tflite_runtime-2.5.0-cp37-cp37m-linux_armv7l.whl

This hint really deserves 10000 kudos... Thanks for sharing. I was successfully following this great tutorial (thanks @EdjeElectronics) step by step, achieved 7 fps @320x240 on an overclocked RPI 4 with 2GB RAM. Then I plugged the Coral (came today), followed the install steps and came across exactly the same problem. Followed the hint and I'm now having about 55 fps at 320x240.

Nice. Thanks again.

Moehrenbaum, 2020-11-27:
> @Moehrenbaum thank you for sharing your solution! I'm glad you were able to figure it out. I have already pointed a couple people at your solution, so thanks again for helping out.

Thanks @EdjeElectronics for taking the time to create the great walkthrough.