EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #96: Tensor Flow Lite Object Detection on Android Problem with step 2e

bitcryien, 2021-02-15:
When I run the command: bazel build --config=opt //tensorflow/tools/pip_package:build_pip_package 

I always recieve the following error:
ERROR: error loading package '': Encountered error while reading extension file 'repositories/repositories.bzl': no such package '@io_bazel_rules_docker//repositories': Traceback (most recent call last):
        File "C:/users/tobias diehl/_bazel_tobias diehl/j7bi4x5j/external/bazel_tools/tools/build_defs/repo/git.bzl", line 164
                _clone_or_update(ctx)
        File "C:/users/tobias diehl/_bazel_tobias diehl/j7bi4x5j/external/bazel_tools/tools/build_defs/repo/git.bzl", line 74, in _clone_or_update
                fail(("error cloning %s:\n%s" % (ctx....)))
error cloning io_bazel_rules_docker:
/usr/bin/bash: line 2: cd: too many arguments
+ cd C:/users/tobias diehl/_bazel_tobias diehl/j7bi4x5j/external
/usr/bin/bash: line 4: cd: too many arguments
+ git -C 'C:/users/tobias diehl/_bazel_tobias diehl/j7bi4x5j/external/io_bazel_rules_docker' reset --hard 251f6a68b439744094faff800cd029798edf9faa
fatal: not a git repository (or any of the parent directories): .git
+ git -C 'C:/users/tobias diehl/_bazel_tobias diehl/j7bi4x5j/external/io_bazel_rules_docker' fetch origin 251f6a68b439744094faff800cd029798edf9faa:251f6a68b439744094faff800cd029798edf9faa
fatal: not a git repository (or any of the parent directories): .git
+ git -C 'C:/users/tobias diehl/_bazel_tobias diehl/j7bi4x5j/external/io_bazel_rules_docker' fetch origin 251f6a68b439744094faff800cd029798edf9faa:251f6a68b439744094faff800cd029798edf9faa
fatal: not a git repository (or any of the parent directories): .git
ERROR: error loading package '': Encountered error while reading extension file 'repositories/repositories.bzl': no such package '@io_bazel_rules_docker//repositories': Traceback (most recent call last):
        File "C:/users/tobias diehl/_bazel_tobias diehl/j7bi4x5j/external/bazel_tools/tools/build_defs/repo/git.bzl", line 164
                _clone_or_update(ctx)
        File "C:/users/tobias diehl/_bazel_tobias diehl/j7bi4x5j/external/bazel_tools/tools/build_defs/repo/git.bzl", line 74, in _clone_or_update
                fail(("error cloning %s:\n%s" % (ctx....)))
error cloning io_bazel_rules_docker:
/usr/bin/bash: line 2: cd: too many arguments
+ cd C:/users/tobias diehl/_bazel_tobias diehl/j7bi4x5j/external
/usr/bin/bash: line 4: cd: too many arguments
+ git -C 'C:/users/tobias diehl/_bazel_tobias diehl/j7bi4x5j/external/io_bazel_rules_docker' reset --hard 251f6a68b439744094faff800cd029798edf9faa
fatal: not a git repository (or any of the parent directories): .git
+ git -C 'C:/users/tobias diehl/_bazel_tobias diehl/j7bi4x5j/external/io_bazel_rules_docker' fetch origin 251f6a68b439744094faff800cd029798edf9faa:251f6a68b439744094faff800cd029798edf9faa
fatal: not a git repository (or any of the parent directories): .git
+ git -C 'C:/users/tobias diehl/_bazel_tobias diehl/j7bi4x5j/external/io_bazel_rules_docker' fetch origin 251f6a68b439744094faff800cd029798edf9faa:251f6a68b439744094faff800cd029798edf9faa
fatal: not a git repository (or any of the parent directories): .git
INFO: Elapsed time: 15,510s
INFO: 0 processes.
FAILED: Build did NOT complete successfully (0 packages loaded)

I'm struggeling now for a week at this step and i have no idea to finally build tensorflow.
can you help me?

EdjeElectronics, 2021-02-15:
Hi @bitcryien , I can't get TensorFlow to build on my Windows PC anymore either! You can try using Google Colab to convert your TensorFlow Lite model. Here is an online Colab where you can upload your .pb file and convert it to a .tflite file. Let me know if this works!

https://colab.research.google.com/drive/1Px7I6PxeeLhCepyA9Dv22pwz66NuT1JR

bitcryien, 2021-02-18:
Hey @EdjeElectronics , it seems like the idea of using a mobilnet_v2 with tensorflow v1.15 pretty screwed me up. My input shape looks like [None, None, None, 3]. I don't have any idea how i can bring my project to an end

bitcryien, 2021-02-20:
Do you have a solution for me? I did some research and it looks like its almost impossible to convert mobilenetv2 to tflite @EdjeElectronics