EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #97: bash get_pi_requirements.sh Error for Raspberry PI 4 64 bit OS beta

hackebike, 2021-02-16:
I am installing the TensorFlow and got the error below:
I am on  a Raspberry PI 4 8Gb 64 bit OS beta, Linux 5.10.11-v8+ #1399 SMP PREEMPT Thu Jan 28 12:14:03 GMT 2021 aarch64 GNU/Linux.
Python installed 3.7.3.

"Could not find a version that satisfies the requirement opencv-python==3.4.6.27 (from versions: 3.4.10.37, 3.4.11.39, 3.4.11.41, 3.4.11.43, 3.4.11.45, 3.4.13.47, 4.3.0.38, 4.4.0.40, 4.4.0.42, 4.4.0.44, 4.4.0.46, 4.5.1.48)
No matching distribution found for opencv-python==3.4.6.27
tflite_runtime-2.5.0-cp37-cp37m-linux_armv7l.whl is not a supported wheel on this platform."

Any suggestions on how to fix this?
thx

EdjeElectronics, 2021-02-16:
Hmm... can you try using this command to install OpenCV? 
`pip3 install opencv-python==3.4.13.47`

Then, can you use these commands and let me know what the output is?
`uname -a`
`cat /etc/os-release`

hackebike, 2021-02-16:
ok
**uname -a:** Linux 5.10.11-v8+ #1399 SMP PREEMPT Thu Jan 28 12:14:03 GMT 2021 aarch64 GNU/Linux

**cat /etc/os-release:**
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

```
pip3 install opencv-python==3.4.13.47
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting opencv-python==3.4.13.47
  Downloading https://files.pythonhosted.org/packages/23/d5/231ea37e8b5013f9d14d46b6d86e2f1d6625b6ac624131ea261d17e9cead/opencv-python-3.4.13.47.tar.gz (87.6MB)
    100% |████████████████████████████████| 87.6MB 6.2kB/s 
  Installing build dependencies ... error
  Complete output from command /home/pi/tflite1/tflite1-env/bin/python3 -m pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-e7ss91wn --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- setuptools wheel scikit-build cmake pip "numpy==1.13.3; python_version=='3.6' and platform_machine != 'aarch64'" "numpy==1.19.3; python_version>='3.6' and sys_platform == 'linux' and platform_machine == 'aarch64'" "numpy==1.14.5; python_version=='3.7' and platform_machine != 'aarch64'" "numpy==1.17.3; python_version=='3.8' and platform_machine != 'aarch64'" "numpy==1.19.3; python_version>='3.9' and platform_machine != 'aarch64'":
  Ignoring numpy: markers 'python_version == "3.6" and platform_machine != "aarch64"' don't match your environment
  Ignoring numpy: markers 'python_version == "3.7" and platform_machine != "aarch64"' don't match your environment
  Ignoring numpy: markers 'python_version == "3.8" and platform_machine != "aarch64"' don't match your environment
  Ignoring numpy: markers 'python_version >= "3.9" and platform_machine != "aarch64"' don't match your environment
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://www.piwheels.org/simple
  Collecting setuptools
    Downloading https://files.pythonhosted.org/packages/15/0e/255e3d57965f318973e417d5b7034223f1223de500d91b945ddfaef42a37/setuptools-53.0.0-py3-none-any.whl (784kB)
  Collecting wheel
    Downloading https://files.pythonhosted.org/packages/65/63/39d04c74222770ed1589c0eaba06c05891801219272420b40311cd60c880/wheel-0.36.2-py2.py3-none-any.whl
  Collecting scikit-build
    Downloading https://files.pythonhosted.org/packages/78/c9/7c2c7397ea64e36ebb292446896edcdecbb8c1aa6b9a1a32f6f67984c3df/scikit_build-0.11.1-py2.py3-none-any.whl (72kB)
  Collecting cmake
    Downloading https://files.pythonhosted.org/packages/26/76/b6df98a03343160a27c449c2866d8b94d349be4a3a6632aaf49c8d27916b/cmake-3.18.4.post1.tar.gz
    Installing build dependencies: started
    Installing build dependencies: finished with status 'done'
  Collecting pip
    Downloading https://files.pythonhosted.org/packages/fe/ef/60d7ba03b5c442309ef42e7d69959f73aacccd0d86008362a681c4698e83/pip-21.0.1-py3-none-any.whl (1.5MB)
  Collecting numpy==1.19.3
    Downloading https://files.pythonhosted.org/packages/cb/c0/7b3d69e6ee68bc54c97ba51f8c3c3e43ff1dbc7bd97347cc19a1f944e60a/numpy-1.19.3.zip (7.3MB)
    Installing build dependencies: started
    Installing build dependencies: finished with status 'done'
  Collecting distro (from scikit-build)
    Using cached https://files.pythonhosted.org/packages/25/b7/b3c4270a11414cb22c6352ebc7a83aaa3712043be29daa05018fd5a5c956/distro-1.5.0-py2.py3-none-any.whl
  Collecting packaging (from scikit-build)
    Using cached https://files.pythonhosted.org/packages/3e/89/7ea760b4daa42653ece2380531c90f64788d979110a2ab51049d92f408af/packaging-20.9-py2.py3-none-any.whl
  Collecting pyparsing>=2.0.2 (from packaging->scikit-build)
    Using cached https://files.pythonhosted.org/packages/8a/bb/488841f56197b13700afd5658fc279a2025a39e22449b7cf29864669b15d/pyparsing-2.4.7-py2.py3-none-any.whl
  Building wheels for collected packages: cmake, numpy
    Running setup.py bdist_wheel for cmake: started
    Running setup.py bdist_wheel for cmake: finished with status 'error'
    Complete output from command /home/pi/tflite1/tflite1-env/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-i1gz8zi9/cmake/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-4azyl48p --python-tag cp37:
    Traceback (most recent call last):
      File "/tmp/pip-build-env-xk_c5n0a/lib/python3.7/site-packages/skbuild/setuptools_wrap.py", line 560, in setup
        cmkr = cmaker.CMaker(cmake_executable)
      File "/tmp/pip-build-env-xk_c5n0a/lib/python3.7/site-packages/skbuild/cmaker.py", line 95, in __init__
        self.cmake_version = get_cmake_version(self.cmake_executable)
      File "/tmp/pip-build-env-xk_c5n0a/lib/python3.7/site-packages/skbuild/cmaker.py", line 82, in get_cmake_version
        "Problem with the CMake installation, aborting build. CMake executable is %s" % cmake_executable)
  
    Problem with the CMake installation, aborting build. CMake executable is cmake
  
    ----------------------------------------
    Failed building wheel for cmake
    Running setup.py clean for cmake
    Running setup.py bdist_wheel for numpy: started
    Running setup.py bdist_wheel for numpy: still running...
[... cut ...] ://www.piwheels.org/simple -- setuptools wheel scikit-build cmake pip "numpy==1.13.3; python_version=='3.6' and platform_machine != 'aarch64'" "numpy==1.19.3; python_version>='3.6' and sys_platform == 'linux' and platform_machine == 'aarch64'" "numpy==1.14.5; python_version=='3.7' and platform_machine != 'aarch64'" "numpy==1.17.3; python_version=='3.8' and platform_machine != 'aarch64'" "numpy==1.19.3; python_version>='3.9' and platform_machine != 'aarch64'"" failed with error code 1 in None
```

EdjeElectronics, 2021-02-16:
Alright, for now I am just going to say that you should switch to using the normal 32-bit Raspberry Pi OS. Feel free to tinker around and see if you can get it working on 64-bit! If you are able to get it working, please share how you did so :)

hackebike, 2021-02-16:
Can't switch to 32 !
Will try to see if I can find a solution.

hackebike, 2021-02-16:
Btw in the script needs an update for for Raspberry PI OS 64 bit beta kernel 5.10

pip3 install https://github.com/google-coral/pycoral/releases/download/v1.0.1/tflite_runtime-2.5.0-cp37-cp37m-linux_aarch64.whl

EdjeElectronics, 2021-02-16:
Thanks! Hopefully you will be able to find 64-bit versions of all the other required packages, too.

Out of curiosity (to help me better understand the types of applications that Raspberry Pi users work on), why do you need 64 bit? The 32 bit version has access to all 8GB of RAM, but it has the limitation that a single program cannot use more than 4GB of RAM. Are you working on a program that will use more than 4GB of RAM?

hackebike, 2021-02-16:
I got it to install OpenCV and you need to make sure that pip3 is the latest version since it needs at least v19 for the latest.
So basically to install on the Raspberry PI 4 64 bit OS beta the script needs these lines:

```
pip3 install --upgrade pip
pip3 install opencv-python
pip3 install https://github.com/google-coral/pycoral/releases/download/v1.0.1/tflite_runtime-2.5.0-cp37-cp37m-linux_aarch64.whl
```

I am doing development and will need to use 64 bit since I will use other software too, like speech recognition and bunch of other sensors. I do not have an immediate need, so I have time to got over the pain of been on beta OS ! :)

hackebike, 2021-02-16:
I think is working, run the various script and the only one that failed is the webcam one, since camera on the PI OS 64 bit has issues:

```
python3 TFLite_detection_webcam.py --modeldir=Sample_TFLite_model
Traceback (most recent call last):
  File "TFLite_detection_webcam.py", line 171, in <module>
    frame = frame1.copy()
AttributeError: 'NoneType' object has no attribute 'copy'
```

fnajafi, 2021-03-14:
It didn't solve my problem , for installing tensorflow 2.2 on raspberry pi 4B , 8G . I raised this issue on ur Twitter page . Thanks

Jhchavezr, 2021-04-18:
Hello guys,
After updating python and opencv-python, I was able to run the Object detector. Thanks a lot Hackebike.
I'm using the Raspberry Pi 4 with Raspbian 10.9 32 bit OS.

> I got it to install OpenCV and you need to make sure that pip3 is the latest version since it needs at least v19 for the install.
> So basically to install on the Raspberry PI 4 64 bit OS beta, python 3.7 the script needs these updates:
> 
> ```
> pip3 install --upgrade pip
> pip3 install opencv-python
> pip3 install https://github.com/google-coral/pycoral/releases/download/v1.0.1/tflite_runtime-2.5.0-cp37-cp37m-linux_aarch64.whl
> ```
> 
> I am doing development and will need to use 64 bit since I will use other software too, like speech recognition and bunch of other sensors. I do not have an immediate need, so I have time to got over the pain of been on beta OS ! :)

alexzobi, 2022-10-20:
> I think is working, run the various script and the only one that failed is the webcam one, since camera on the PI OS 64 bit has issues:
> 
> ```
> python3 TFLite_detection_webcam.py --modeldir=Sample_TFLite_model
> Traceback (most recent call last):
>   File "TFLite_detection_webcam.py", line 171, in <module>
>     frame = frame1.copy()
> AttributeError: 'NoneType' object has no attribute 'copy'
> ```
> 
> Btw, if I want to see the obj recognition done on the PI from a mobile application, stream done via WiFi, can I use the stream class to provide the video to a web page which is accessed from the mobile app, would it work? Currently I passed bunch of data via XML from the PI to a mobile app, would like to pass the live video obj detection too.

This is where I'm stuck. The problem here is that if you're using a picamera module (or in my case the Arducam 16MP), you need to use the picamera2 library to create the stream, as opencv does not yet support the libcamera module. Hopefully this will change at some point, but in the meantime, I'm trying to figure out how to adapt the video stream class code to use picamera2. picamera2 has example docs on how to use with open cv and how to stream, but not how to stream to opencv. i'm not a python dev, so if anyone has tips, that'd be appreciated.

andrewvictor, 2023-05-03:
I was able to just modify the "get_pi_requirements.sh" script to install defaults.  opencv and tensorflow seem to be working for me.  I have yet, though, to test generating my own model.  Will update.
[image]