EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #101: Error when training

sleepingbilly, 2021-03-30:
Hi,

I have A project where I am trying to retrain an object detector to detect 3 different object classes within an image taken from a USB camera on a Raspberry Pi 4. I have successfully followed the Edje Electronics youtube tutorial "How To Run TensorFlow Lite on Raspberry Pi for Object Detection", where my Pi can run those scripts with no issues. 

I have since attempted the tutorial: https://github.com/EdjeElectronics/TensorFlow-Object-Detection-API-Tutorial-Train-Multiple-Objects-Windows-10 with hopes to learn how to re train a model to suit my project, and then follow the tutorial https://github.com/EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi to convert my trained model to Tf-Lite and run on my Pi. 

I am up to the step where I will enter the command: python train.py --logtostderr –train_dir=training/ --pipeline_config_path=training/ssd_mobilenet_v2_quantized_300x300_coco.config, but it throws errors. (I have also tried implementing ssd_mobilenet_v1_coco, faster_rcnn_inception_v2_coco, and still get errors)

I have tried many different tutorials online with no success and no replys. This is the furthest I have gotten as I have managed to actually get the Pi working by following "How To Run TensorFlow Lite on Raspberry Pi for Object Detection". 

Please help...This is super demotivating to continually keep failing at trying to succeed in this part of my project.

I will post my error message. Apologies in advance for including (most likely) a whole lot of redundant information as I am uploading my entire error trace, however I don't want to accidentally leave anything out that could point me in the right direction. 

Kind regards :-) 

[image]

[image]

[image]

[image]

[image]

dhruvsheth-ai, 2021-03-30:
I believe that this is a time consuming method to train a Dataset. Why don't you try to train your dataset on www.EdgeImpulse.com
It's really quick, you can upload images, train the model and get a tflite model from the dashboard. If you run into errors, you can raise an issue at forum.edgeimpulse.com 
This would help in training a custom model and use it with EdjeElectronics python script.

sleepingbilly, 2021-03-30:
Thank you for the suggestion mate, I will give it a try. 

When you say use it with EdjeElectronics python script, do you mean the detection script?

Thank you for your time  :-)

dhruvsheth-ai, 2021-03-31:
Yes, the detection script

sleepingbilly, 2021-03-31:
Hi @dhruvsheth-ai  , I have run through the process of training my dataset on www.EdgeImpulse.com, after running into errors and raising issues on the forum like you suggested, I was informed that object detection blocks are not made public yet on this platform, and that I should use a classification block instead. My project requires object detection, not just classification of an image, so this seems unsuitable for my task.

Thank you for your support and replies though @dhruvsheth-ai. Our convo is kind of deviating away from my original issue that I have with running my EdjeElectronics training script. Would you happen to be able to help me with this error mate?

Kind regards

> Yes, the detection script

Petros626, 2021-08-29:
you must use the model, which is compatible with your tensorflow version and notice that faster rcnn does not work with ssd