EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #103: Error while running my model

babmuckja, 2021-04-16:
HI. I'm doing my project in college . In the project, we need a program that recognizes numbers. so we made the model with teachable machine(google). I followed your example video well. But when I try to run my model, I get an error and I want to ask about this problem. 
[image]
This is my model folder. 
[If your model folder has a different name than "Sample_TFLite_model", use that name instead. For example, I would use --modeldir=BirdSquirrelRaccoon_TFLite_model to run my custom bird, squirrel, and raccoon detection model.] <- we tried this but we couldnt do.
[image]
[image]
How can we solve this problem.? Pls help me..

seung-gu, 2021-04-16:
check the file name ...
The path to the file name in the code is "labelmap.txt", but you have "label.txt".
Plus, not "model.tflite", but "detect.tflite" according to your code.

babmuckja, 2021-04-17:
I changed the file name before. labels.txt -> labelmap.txt and model.tflite -> detect.tflite. but It didn't work.

babmuckja, 2021-04-17:
[image]
[image]
[image]
Im new at cording..