EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #110: low FPS (0.2 fps) on windows 10 without EdgeTPU

gyang2020, 2021-06-23:
Live stream from IP Cam  (RTSP) with the stream script works, but the fps is about 5 seconds per frame. Also tested the test.mp4 file, similar results. I narrowed it down to the  .invoke() line. Is this somewhat expected because TFlite is not for windows? Thanks!

gyang2020, 2021-10-06:
> hey nice to hear that youre working on this project.
>
> i immediatelly tested the webcam script and my fps is also under 0.2 fps.
> i dont know whats the problem, my resolution on webcam is 640x480 with
> 30fps.
>
> before i test it on the RPi 4 i wanted to make sure its working on my
> laptop, but its so slow... did you find a solution?

Hi, we ended up using rpi4 without tpu, it's 4fps, the archetcture of the
processor is the problem I guess. May also be the operating system

Petros626, 2021-10-06:
@gyang2020 
i have not tried the script on the raspberry pi yet. i wanted to try detect.tflite on the laptop via the webcam first and i don't understand the FPS is at 1.0 at the beginning and drops to 0.12 fps afterwards.

it's really strange and i have no explanation for it :(