EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #111: Run custom model on RPi 4 with TFLite_detection_webcam.py (list index out of error)

Petros626, 2021-07-22:
Hey guys,

i have worked through the second tutorial from edjeelectronics and would like to use **coco_ssd_mobilenet_v1_quantized** for my own project. I have created a **detect.tflite** file from my trained network and the matching **labelmap.txt** (this was not yet present when converting on the PC)

On the Pi I then wanted to run the above python script, however I always get the same error:

**File "TFLite_detection_webcam.py", line 189, in <module>.
classes = interpreter.get_tensor(output_details[1]['index']) [0]
IndexError: list index out of range.**

In a forum someone said you have to extend the labelmap.txt to 90 entries, because I had less, but that didn't solve the problem.

Thanks for your help

AlexM1989, 2021-07-25:
I have the same issue

Petros626, 2021-09-28:
> I have the same issue

I know the solution if you're interested

Tonycrux, 2021-10-19:
> > I have the same issue
> 
> I know the solution if you're interested

What is the solution now, i ran into the same issue

Petros626, 2021-10-19:
> > > I have the same issue
> > 
> > 
> > I know the solution if you're interested
> 
> What is the solution now, i ran into the same issue

So in my case I used a trained model that I train on pc, but it wasn't the ssd or another model that fits for the pi. We use a model that is trained on image classification not on object detection and so you will get this error

EdjeElectronics, 2021-11-28:
Yep, this usually happens when you're using an image classification model instead of an object detection model.

Here's a good example showing how to use a classification model on the Pi:
https://github.com/tensorflow/examples/tree/master/lite/examples/image_classification/raspberry_pi

closed 2021-11-28 by EdjeElectronics