EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #123: Dear EdjeElectronics  I am having issues running my custom model built from ssd _ fpn _ lite using the TFLite_detection_webcam.py

OJBameyi, 2021-11-23:
Dear EdjeElectronics
I have followed your tutorials and they have been quite helpful. I am having this error when I try to run my own trained model, based on SSD fpn lite.

Please can you help out. I am using a raspberry 3B+ with a raspian buster.

The error message is pated below:

Traceback (most recent call last):
  File "TFLite_detection_webcam.py", line 138, in <module>
    interpreter = Interpreter(model_path=PATH_TO_CKPT)
  File "/home/pi/Projects/Python/tflite/tflite-env/lib/python3.7/site-packages/tflite_runtime/interpreter.py", line 351, in __init__
    experimental_preserve_all_tensors))
ValueError: Didn't find op for builtin opcode 'STRIDED_SLICE' version '6'. An older version of this builtin might be supported. Are you using an old TFLite binary with a newer model?
Registration failed.

_Originally posted by @OJBameyi in https://github.com/EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi/issues/13#issuecomment-977228358_

EdjeElectronics, 2021-11-28:
What TensorFlow version did you train your SSD FPN Lite model?

OJBameyi, 2021-11-29:
> What TensorFlow version did you use to train your SSD FPN Lite model?
I used TensorFlow 2.7