EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #126: TFLite_detection_webcam.py fails with "open VIDEOIO(V4L2:/dev/video0): can't open camera by index" when the libcamera camera stack is used

christianbaun, 2021-12-22:
When switching from the legacy camera stack under Raspberry Pi OS with debian 10 to the new libcamera camera stack (in Raspberry Pi OS with debian 11 it is standard), the **TFLite_detection_webcam.py** script does not work any longer.

You can switch in Raspberry Pi OS (debian 10) between both camera stacks by adding/removing the line 

`camera_auto_detect=1`

in the file `/boot/config.txt`.

After switching to the new libcamera camera stack, **TFLite_detection_webcam.py** results in this error message:

```
$ python3 TFLite_detection_webcam.py --modeldir=/home/pi/model_2021_07_08 --graph=detect.tflite --labels=/home/pi/model_2021_07_08/labelmap.txt 
[ WARN:0] global /tmp/pip-wheel-2c57qphc/opencv-python_86774b87799240fbaa4c11c089d08cc3/opencv/modules/videoio/src/cap_v4l.cpp (890) open VIDEOIO(V4L2:/dev/video0): can't open camera by index
Traceback (most recent call last):
  File "TFLite_detection_webcam.py", line 172, in <module>
    frame = frame1.copy()
AttributeError: 'NoneType' object has no attribute 'copy'
^CException ignored in: <module 'threading' from '/usr/lib/python3.7/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 1281, in _shutdown
    t.join()
  File "/usr/lib/python3.7/threading.py", line 1032, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.7/threading.py", line 1048, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):

```

Often read recommendations like modify `cv2.VideoCapture(0)` into `cv2.VideoCapture(cv2.CAP_V4L2)` or `cv2.VideoCapture(1)` or `cv2.VideoCapture(-1)` did not help.

The camera works well. I tested it with 

`libcamera-vid -t 0`

```
$ v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
	/dev/video10
	/dev/video11
	/dev/video12
	/dev/video18

bcm2835-isp (platform:bcm2835-isp):
	/dev/video13
	/dev/video14
	/dev/video15
	/dev/video16

unicam (platform:fe801000.csi):
	/dev/video0
	/dev/video1
```

I also gave `/dev/video0` and `/dev/video1` full permissions (777) just to ensure that it is not a permission issue.

Has anyone here seen the same issue and solved it?

Expertcoder111111, 2022-01-23:
Also had a similar issue, anyone have a way of fixing this?
[image]
` [WARN:0] VIDE0I0(V4L2:/dev/video0): can't open camera by index
Traceback (most recent call last):
 File "TFLite_detection_webcam.py", line 171, in <module>
    frame = frame1.copy()
AttributeError: 'NoneType' object has no attribute 'copy'`

I'm using picam on Raspberry Pi OS Buster

EdjeElectronics, 2022-09-02:
Thanks for raising this issue @christianbaun! The new libcamera library is a pain. I haven't taken the time yet to figure out how to make it work with OpenCV. For now, I just enabled the legacy camera library (following [these instructions](https://www.raspberrypi.com/news/new-old-functionality-with-raspberry-pi-os-legacy/)).

If you have time, can you look in to how to get the new libcamera library working with OpenCV and let me know if you find a solution?