EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #139: when will be added TensorFlow Lite 1 notebook??

kotran88, 2022-09-21:
Hello :) 
Thank you for sharing nice Colab notebook for Tensorflow v2 !
I'm having problem in less accuracy with quantized tflite. 
and notebook said quantized with tensor flow v1 is better then v2. 
but as far as I know, Colab is not supporting tensor flow v1 anymore....so....how can I use ssd-mobilenet-v1-quantized model  ? 

Thank you

EdjeElectronics, 2022-09-22:
Thanks for trying out the notebook, and good question! I'll work on getting the TFLite v1 notebook added to the repository some time next week. It's been a couple months since I last used it, so there's a chance it won't work anymore. I'll test it out and post an update here when it's ready!

EdjeElectronics, 2022-10-12:
Hi @kotran88 , I just uploaded the TensorFlow 1 notebook. You can train the `ssd-mobilenet-v2-quantized` model, and it should have better performance than the equivalent version from TensorFlow 2. I still haven't fully fleshed out the comments and text, but all the code is there.

Try it out and let me know if it works for you!
https://colab.research.google.com/github/EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi/blob/master/Train_TFLite1_Object_Detection_Model.ipynb

lakpa-tamang9, 2023-01-31:
Google colab has python starting from 3.8. So tf==1.15 is not supported. I alternatively installed python 3.7 in the colab and the tensorflow is installed successfully, but the pycocotools cannot be installed. Can you please help?

Petros626, 2023-07-19:
> Google colab has python starting from 3.8. So tf==1.15 is not supported. I alternatively installed python 3.7 in the colab and the tensorflow is installed successfully, but the pycocotools cannot be installed. Can you please help?

using this command `!pip install pycocotools`?