EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #141: NotFoundError: /content/labelmap.pbtxt; No such file or directory

rssurya19, 2022-10-09:
[image]

not able to generate labelmap.pbtxt. pls help its urgent

EdjeElectronics, 2022-10-10:
The labelmap.pbtxt file should be automatically generated when you run the code in the "Create TFRecords" section. Did you run all the code blocks in that section?

rssurya19, 2022-10-10:
yes i run all the code . but labelmap.pbtxt is not form.
in"CREATE TFRecord " description u mention their are three code but their are only two are present. which are "create_csv.py" and 
"create_tfrecord.py"

rssurya19, 2022-10-12:
i created manually by referring others labelmap.pbtxt file . add my classes in it

arrrrny, 2023-03-14:
@rssurya19 Can you share the  labelmap.pbtxt file please?

Junaid0411, 2023-03-22:
@rssurya19 could you show us the changes you made in order to get it to work?

muhd360, 2024-04-06:
just rename the file from .txt to .pbtxt

tCode5, 2024-07-04:
> just rename the file from .txt to .pbtxt

This won't work, because .pbtxt-files are structured differently.
If you want to create one yourself, you could do something like this:

labelmap.txt
```text
house
car
person
cat
```

labelmap.pbtxt
```text
item {
  id: 1,
  name: house
}
item {
  id: 2,
  name: car
}
item {
  id: 3,
  name: person
}
item {
  id: 4,
  name: cat
}
```

muhd360, 2024-08-31:
ohh sounds good thx