EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #147: Training Custom Model: 3.2 Split images into train, validation, and test folders not splitting

Junaid0411, 2022-11-17:
Hi Edje,

Your repo has been very helpful and I'm grateful that you've been keeping everything updated.

I've been experiencing issues with train_val_test_split.py , it doesn't split my images and XML files into the 3 folders. 
Here's a screenshot:

[image]

Any advice would be greatly appreciated!

Junaid0411, 2022-11-17:
Issue resolved; just an issue with folder management.

closed 2022-11-17 by Junaid0411

EdjeElectronics, 2022-11-17:
Hi @Junaid0411 , thanks for closing this issue after you figured out how to resolve it. Looking at the screencap you shared, it's pretty common for people to use a similar folder structure as you did. I've got it on my TO DO list to figure out a good way to handle the images.zip file if the images are nested inside an additional folder within the compressed folder.