EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #151: Error in Colab sheet when creating TFRecords

joshfox10, 2022-12-03:
[image]

I am able to run now and completed all steps without errors.

I tried with my own dataset and bumped into this error. Please advise.

_Originally posted by @elvenkim1 in https://github.com/EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi/issues/135#issuecomment-1238984539_
 
I've gotten the same error.  It works when I use the coin dataset.  It works when I use a smaller dataset that I have with nice names (IMG_123), however when I use a larger dataset with names from roboflow export long strings of numbers, letters, symbols, etc I get the error shown here.  I've tried numerous things but can't crack it.  Any suggestions?  Thank you!

EdjeElectronics, 2022-12-11:
I'm guessing it has something to do with the `.` in the filenames, I don't think the `create_tfrecord.py` script handles those well. I'll take a look!

EdjeElectronics, 2022-12-11:
Ahh, shoot. It might be an issue with how Roboflow generates Pascal VOC files. Can you please share a few images and XML files that you're trying to use?

joshfox10, 2022-12-11:
I found it to be the create_csv.  I used the attached script in place of it.  Then the create tfrecord worked well.  
Also, I was unable to get through the tensorflow 1 sheet due to errors since the TF2 quantized result was quite poor.  If I dig in again I'll forward the errors.
[xml_to_csv (1).zip](https://github.com/EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi/files/10202985/xml_to_csv.1.zip)

closed 2022-12-26 by joshfox10