EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #154: Is headless raspberry possible?

beckmx, 2023-01-24:
Hello, I was testing the code and I setup a headless raspberry with only ssh access, and whenever i tried the sample for still images it is givingme a GTK warning, no display was found, I was wondering if you can point me in the right direction to create a new screenshot with the details of the recognized object.

Greetings!

beckmx, 2023-01-24:
nevermind, I just saw the parameters in the file

closed 2023-01-24 by beckmx

EdjeElectronics, 2023-01-25:
Hey, thanks for closing the issue! If you're in headless mode, you can't display any images with cv2.imshow, or you will get an error. Instead, you can add something like `print(object_name)` after line 207 to make it print each object that's detected.