EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #156: Tensorflow 1 tutorial does not create labelmap.pbtxt file

arrrrny, 2023-03-14:
I ran the code twice, I added bird, squirrel, raccoon at step 10 but no luck
[image]
Getting this error 
[image]

Please help

Junaid0411, 2023-03-22:
I have the same issue, have you found a fix by any chance?

Junaid0411, 2023-03-26:
Is there any other way I can train a detection model? I've tried doing several different things but have not succeeded