EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #158: Picamera v3 and libcamera

diogodh, 2023-04-15:
The new picamera v3 doesn't work with legacy anyome and needs libcamera. This way, the script needs to be modified to work with libcamera.
The error it outputs is
` frame = frame1.copy()
AttributeError: 'NoneType' object has no attribute 'copy'`

If anyone can help porting the script to use libcam would be awesome

Petros626, 2023-04-18:
I only know PiCamera2 (Python). You mean a camera model or software? Either you have wrong ordered code commands or your camera is really not detected.

diogodh, 2023-04-20:
Sorry for don't explain me correctly. I mean the Raspberry Camera V3 (hardware) is showing an error. Nothing related to the library picamera2.

closed 2023-04-20 by diogodh

diogodh, 2023-04-20:
> so it's solved for you, I guess.
> <EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi/issues/158/1516173464
> @github.com>

Not solved but i need to frame the issue better to i can explain myself
better
I can use it on my windows laptop perfectly but in my Raspberry pi with the
new raspberry camera V3 i can't.

I am not sure if it is related with the fact that with raspberry camera V3,
the old legacy camera libraries were depreciated.

Petros626, 2023-04-20:
this issue is better placed here https://github.com/orgs/raspberrypi/repositories