EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #159: TypeError: object of type 'numpy.float32' has no len()

Darrenll, 2023-05-03:
Line 177 has a type error. The example in your 19 year video ran smoothly, but I trained a new one to encounter this error.
pi@raspberrypi:~ $ cd tflite1/
pi@raspberrypi:~/tflite1 $ source tflite1-env/bin/activate
(tflite1-env) pi@raspberrypi:~/tflite1 $ python3 TFLite_detection_webcam.py --modeldir=custom_model_lite
Traceback (most recent call last):
  File "TFLite_detection_webcam.py", line 177, in <module>
    for i in range(len(scores)):
TypeError: object of type 'numpy.float32' has no len()
^CException ignored in: <module 'threading' from '/usr/lib/python3.7/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 1281, in _shutdown
    t.join()
  File "/usr/lib/python3.7/threading.py", line 1032, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.7/threading.py", line 1048, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt
May I ask if there is a problem with my training set or with the version? Please help

Darrenll, 2023-07-05:
> 您使用哪种型号？我之前在尝试训练与 ssd-mobilenet-v2 不同的模型时见过该错误。

It is indeed a version issue, and I have solved the problem using your code from last year's version of TF. Thank you very much!

Darrenll, 2023-07-05:
> Which model are you using? I've seen that error before when trying to train a different model than ssd-mobilenet-v2.

It is indeed a version issue, and I have solved the problem using your code from last year's version of TF. Thank you very much!

Gottmituns2022, 2023-07-21:
Hello everyone. I used the ssd-mobilnet-v2 version, teaching her on this collab notebook https://colab.research.google.com/github/EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi/blob/master/Train_TFLite2_Object_Detction_Model.ipynb?authuser=2#scrollTo=6t8CMarqBqP9 and I got the same error as you. Can you tell me how to fix it?