EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #164: Empty .tfrecord files

evanjohnstone06, 2023-07-12:
I have followed all steps and found workarounds for all bugs, but whenever I try and finish setting up the model to train, the train.tfrecord and val.tfrecord files are both empty and then I get stuck on "Use 'tf.cast' instead" during training. I have used two different labelling softwares for labelling my images, RectLabel and Labelimg, and have manually edited my xml files to look exactly as the xml files in the coin dataset, but for some reason its just not working. Any help is appreciated.

OriLevi026, 2024-01-08:
**replace inside the create_csv.py file or just run this instead:(Section 3.3)**
 ```
import os
 import glob
 import pandas as pd
 import xml.etree.ElementTree as ET

 def xml_to_csv(path):
     xml_list = []
     for xml_file in glob.glob(path + '/*.xml'):
         tree = ET.parse(xml_file)
         root = tree.getroot()
         filename = root.find('filename').text
         width = int(root.find('size/width').text)
         height = int(root.find('size/height').text)

         for obj in root.findall('object'):
             name = obj.find('name').text
             xmin = int(obj.find('bndbox/xmin').text)
             ymin = int(obj.find('bndbox/ymin').text)
             xmax = int(obj.find('bndbox/xmax').text)
             ymax = int(obj.find('bndbox/ymax').text)

             xml_list.append((filename, width, height, name, xmin, ymin, xmax, ymax))

    column_name = ['filename', 'width', 'height', 'class', 'xmin', 'ymin', 'xmax', 'ymax']
    xml_df = pd.DataFrame(xml_list, columns=column_name)
    return xml_df

def main():
    for folder in ['train', 'validation']:
        image_path = os.path.join(os.getcwd(), ('images/' + folder))
        xml_df = xml_to_csv(image_path)
        xml_df.to_csv(('images/' + folder + '_labels.csv'), index=None)
        print('Successfully converted xml to csv.')

main()
```

zdkr4ii, 2024-07-28:
Have you ever solved this?

OriLevi026, 2024-07-28:
Yes, it just the xml format you use is integer and not float, the code I have added, match both types