EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #170: Error in step 7.1 - Inference test images

vankhoa01, 2023-09-11:
Hi @EdjeElectronics and my friends,
Thank you so much for your brilliant library.
I'm trying to custom model with your document.
But I'm facing issue at step 7.1, pls check image below for more detail:
[image]

Here's xml file:
[image]

Do you have any ideas to resolve it.
Thank you so much

KimYu54, 2023-12-11:
hi @vankhoa01, I am currently facing the same issue, have you sorted it out?

vankhoa01, 2023-12-13:
@KimYu54 : Please refer tutorial here https://www.youtube.com/watch?v=amURyS6CAaY
It's really helpful