EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi #171: ModuleNotFoundError: No module named 'object_detection'

SpiderElectronics, 2023-09-12:
Hello Evan:
!pip install pyyaml==5.3 installation error showing error: subprocess-exited-with-error

Displayed when the next module is run:

# Run Model Bulider Test file, just to verify everything's working properly
!python /content/models/research/object_detection/builders/model_builder_tf2_test.py

Traceback (most recent call last):
  File "/content/models/research/object_detection/builders/model_builder_tf2_test.py", line 24, in <module>
    from object_detection.builders import model_builder
ModuleNotFoundError: No module named 'object_detection'

5amleach, 2023-09-12:
Hi, yes I am experiencing the exact same issue.

Jimbwlah, 2023-09-12:
I'm getting the same issue today - this colab was working last week

EdjeElectronics, 2023-09-12:
Thanks all! I'll look into it.

stenglx, 2023-09-13:
https://github.com/tensorflow/models/issues/11085#issuecomment-1716889016

Solves the issue for me

5amleach, 2023-09-13:
> [tensorflow/models#11085 (comment)](https://github.com/tensorflow/models/issues/11085#issuecomment-1716889016)
> 
> Solves the issue for me

thank you that solved it for me too

Jimbwlah, 2023-09-13:
> [tensorflow/models#11085 (comment)](https://github.com/tensorflow/models/issues/11085#issuecomment-1716889016)
> 
> Solves the issue for me

Thank you, that also solved it for me!

EdjeElectronics, 2023-09-14:
Thanks @stenglx , you rock! I just updated the Colab to implement that change. I tested the full notebook and everything worked. Please re-open this issue if the error still crops up.

closed 2023-09-14 by EdjeElectronics

skupperr, 2025-07-12:
This issue, "ModuleNotFoundError: No module named 'object_detection'", is happening again. 😢